kriswiner/MPU9250 #160: error compiling MPU9250BasicAHRS_t3.ino

seamaster101, 2017-07-11:
Hi Kris,
Sorry but I'm way to green to deal with this stuff, but regardless I decided to give it a try. How one learns otherwise :) ?
I'm using Teensy3.2 and have the sparkfun mini breakout board: https://www.sparkfun.com/products/13762?_ga=2.163313852.1677630905.1499749857-1121855153.1486198878
When i download and try to compile the MPU9250BasicAHRS_t3.ino I get:
 error: 'MahonyQuaternionUpdate' was not declared in this scope
  MahonyQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f, my, mx, mz);
I'm I missing a library or something? 
if I rem this line out it compiles and runs but does not give me changing values for roll, pitch and yaw
My goal is to create a rate compass that is tilt compensated for a boat autopilot reference. Any hints for that use?

thanks for the help

seamaster101, 2017-07-11:
or I get error: "MadgwickQuaternionUpdate" was not declared in this scope, depending which line in the code I have enabled.

kriswiner, 2017-07-11:
Looks like you forgot to include the quaternionFilters.ino file in your
sketch

seamaster101, 2017-07-12:
thank you! this resolved that issue of mine. Sorry for the stupid question....

closed 2017-07-12 by seamaster101