kriswiner/MPU9250 #161: no values coming from the MPU9250

seamaster101, 2017-07-12:
Hi Kris,
I did compile successfully MPU9250BasicAHRS_t3.ino after your help this morning.
It works and I get proper pitch and roll  but yaw is not quite right. It is all over the place when I rotate the unit to try to point to the North.
Then I tried to compile MPU9250_MS5637_AHRS_t3.ino so I can perform the calibration for the magnetometer, that is built into this sketch and everything appears to work, but all data from the devices except the temperature is 0.
*****************
Scanning...
I2C device found at address 0x0C  !  /// why are 2m devices here?
I2C device found at address 0x68  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM 71 I should be 71

MPU9250 is online...
x-axis self test: acceleration trim within : 1.4% of factory value
y-axis self test: acceleration trim within : -0.1% of factory value
z-axis self test: acceleration trim within : 5.4% of factory value
x-axis self test: gyration trim within : -3.7% of factory value
y-axis self test: gyration trim within : -0.4% of factory value
z-axis self test: gyration trim within : 0.5% of factory value
 Calibrate gyro and accel
accel biases (mg)
-108.64
14.71
24.17
gyro biases (dps)
-1.27
3.40
-0.55
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
19.78
136.18
469.67
AK8963 mag scale (mG)
0.80
0.70
3.09
X-Axis sensitivity adjustment value 1.20
Y-Axis sensitivity adjustment value 1.21
Z-Axis sensitivity adjustment value 1.16
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Gyro temperature is 33.3 degrees C    //this reacts when i put my finger on the MPU9250
Digital temperature value = 0.00 C
Digital temperature value = 32.00 F
Yaw, Pitch, Roll: 13.80, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 0.03 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
************************************
Attached here is the code that I ended up with after I modified MPU9250_MS5637_AHRS_t3.ino to remove MS5637 references and the display references. The rest of the code is untouched. I'm trying to simplify the code so I can read it and understand it better:

so far I figured the issue maybe is in the data that is populated in the array readMPU9250Data. 

> `

 void loop()
{  
  // If intPin goes high, all data registers have new data
   if(newData == true) {  // On interrupt, read data
     newData = false;  // reset newData flag
     readMPU9250Data(MPU9250Data); // INT cleared on any read
 //   readAccelData(accelCount);  // Read the x/y/z adc values

> `

but I'm not really sure...I might be completely wrong here. Yet with the 
Just for the record, I was getting the exact same result when i compiled the original file without my simplifications.
my MPU is [https://www.sparkfun.com/products/13762](url) and I'm using teensy 3.2.
My project is "tilt compensated rate compass" for boat autopilot reference.
it is little different than aviation, as boat compass needs quite a bit of dampening. 
I hope you can help me figure it out one step at the time...

kriswiner, 2017-07-12:
> my MPU is https://www.sparkfun.com/products/13762 <http://url> and I'm

The sketch you are using relies on a data ready interrupt, likely you
forgot to connect it.

seamaster101, 2017-07-12:
Hi Kris,
Thanks for the immediate replay. 
How do i do that. some direction please?
Jordan

kriswiner, 2017-07-12:
connect the interrupt

seamaster101, 2017-07-12:
Kris, thanks again,
I realize that’s my second time asking stupid question :)  I got that. I did read some references of how to connect and there was no reference of connecting the interrupt for I2C communication on sparkfun website. Now I know better.
Soldering as I type this email...
Jordan

seamaster101, 2017-07-13:
Hi Kris,
Partial success. I get values now, but they are not stable. Even if the board is static the values change quite a bit. What's could that be? Would it be useful to provide serial terminal log? Where do I start?

Jordan

kriswiner, 2017-07-13:
No output please.

What values are unstable?

accel, gyro, and mag? Crappy sensor...

yaw, pitch and roll? Crappy calibration...

Start there.

seamaster101, 2017-07-13:
Hi Kris,
the breakout board is from reputable vendor - Sparkfun
I did some more observation:
If I leave the board alone for 15-20 seconds it settles. 
if I move it around the Z axis it changes the yaw value, rather fast, but after I stop and leave it motionless, it continues to change yaw for a while see the plots on these links:
blue is yaw, red is pitch and green is roll
https://www.dropbox.com/s/wu3bvv6rgs0bkzr/Screenshot%202017-07-12%2022.20.13.png?dl=0
https://www.dropbox.com/s/ihq6ij3yik99yy5/Screenshot%202017-07-12%2022.38.14.png?dl=0
https://www.dropbox.com/s/bxu4755mksjrpdl/Screenshot%202017-07-12%2022.36.30.png?dl=0

When I change the roll or pitch even few degrees and try to maintain heading, the change affects the heading.
Im following the calibration request to wave the board in figure 8 on start up.
thanks,
Jordan

kriswiner, 2017-07-13:
> 202017-07-12%2022.20.13.png?dl=0
> 202017-07-12%2022.38.14.png?dl=0
> 202017-07-12%2022.36.30.png?dl=0

The Sparkfun board is incorrectly designed. Still it shouldn;t behave this
badly. It sounds like the sensoris not properly calibrated.

seamaster101, 2017-07-13:
Thanks again Kris,
is there any other secrets in the calibration? please tell…

 I can give it a shot with calibration, but if you are saying it is incorrectly designed, then I would not waste anymore of your and my time. What board you recommend for best results? I will just order new one.
I just need reliable heading with up to 30-35deg roll and pitch. do you think this is possible?

Jordan

monstah, 2017-07-13:
Hi, why is the sparkfun IMU incorrectly designed? I am facing similar problems with the MPU-9250 from sparkfun, even after the magnetic calibration and a promising plot using an arduino Nano. What fluctuations could I expect from this IMU (in degrees)?

kriswiner, 2017-07-13:
Among other things, they routed power traces underneath the package, not
good practice.

With properly designed boards with proper calibration you should expect 4
degree heading accuracy and 1 degree pitch and roll accuracy using open
source madgwick or mahony fusion filters when using an mcu able to run the
fusion at >1 kHz rates.

monstah, 2017-07-13:
Wow, okay, only 4 to 1 degree.. this would be fairly enough... According to your [wiki-entry-about-sensor-fusion](https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion) I would need to use for example a Teensy with at least 48Mhz to be safe, right? My arduino nano will obviously not fit the requirements with 16Mhz frequency. Sigh.. 

Another (hopefully) simple question, I just want get things right: 

[image]

When I look onto other IMU chips, they show, that the magnetometer and gyro/accel are oriented in the same way, but on the sparkfun IMU they have different orientations in body-frame, as you can see in the picture above. Does that mean I have to make changes in the filter inputs?

kriswiner, 2017-07-13:
Yes, you need to input the sensor data as NED where you choose which axis
is N, so often for the MPU9250 it would look like:

 MadgwickQuaternionUpdate(-ax, ay, az, gx*pi/180.0f, -gy*pi/180.0f,
-gz*pi/180.0f,  my,  -mx, mz);

for x-accel axis == North.

Teensy will work but this is better:

https://www.tindie.com/products/TleraCorp/butterfly-stm32l433-development-board/?pt=full_prod_search

and a nice way to pay me back for this "advice".

seamaster101, 2017-07-13:
Kris,
I don’t mind getting the stm32l433 board, but what IMU should i combine it with? 4 degree accuracy is what bothers me more. commercial compasses claim to provide 2 degree dynamic accuracy with some claiming 1 deg or less.  here ar couple of links:
https://www.maretron.com/products/ssc300.php
http://www.simrad-yachting.com/en-US/Products/Compasses/Simrad-Precision-9-Compass-en-us.aspx

I am pretty sure that they use similar sensor chips. Are they being just too optimistic in their marketing or they really have the accuracy? The prices of these compasses are around $450-$650 US. 

It is not about spending the money, as by the time i’m done I would’ve most likely spend similar amount :). For me is more about the change making it myself, putting the time in and learning new things.
Is there a way to reach 1 deg accuracy within 30-35deg pitch and roll change or this is too tall order to fulfill?
what is the best

kriswiner, 2017-07-13:
> Simrad-Precision-9-Compass-en-us.aspx
> stm32l433-development-board/?pt=full_prod_search

If you want the bestthat can be done at the $35 price range then you can
get 2 degree heading accuracy and 1 degree pitch and roll accuracy using
this:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=full_prod_search

in this case, it doesn't matter what your host MCU is since the fusion is
done on the EM7180 motion co-processor.

There's a nice wiki that describes what it is and what it does:

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki

I would have recommended this sooner but I thought you wanted to use
open-source fusion.

seamaster101, 2017-07-13:
Hi Kris,
This looks promising. I see the graphs you plotted. The heading appears solid. How do you calibrate if attached to a 25 tons boat. I can't possible wave it in figure 8 or flip to each axis. 
Is there a way to compensate for soft and hard magnetic field?

Jordan

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=full_prod_search

https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki

kriswiner, 2017-07-13:
> sensor-fusion-solution/?pt=full_prod_search

The USFS has dynamic calibration for the mag and gyro. You can calibrate
the accel once and store all calibrations in the onboard EEPROM so
calibration in use isn't much of an issue.

This is one of many reasons we use this solution.

seamaster101, 2017-07-17:
Today I was successfully of outputting the yaw roll and pitch to NMEA2000 bus and Simrad NSS Evo3 multifunction display. Yeee! Thank you for the code and the help. 
I'm ready  to order Ultimate Sensor Fusion Solution. I really like that solution.  I'm not sure where I can find simple sample examples.  I read the web page, but the link to the user's guide didn't quite get me to a manual, but rather to another forum. I'm interested in examples for Ultimate Sensor Fusion Solution that will give me yaw, pitch and roll output as well as method adjust the sensitivity/damp the sensitivity for boat use. Is there any such examples?
BTW, do you ship to Canada?

kriswiner, 2017-07-17:
Plent of sample code here:

https://github.com/kriswiner/EM7180_SENtral_sensor_hub

and we can advise on some tweaking once you get some results.

Yes, we ship all over the world.

seamaster101, 2017-07-18:
Hi Kris,
You've got one new customer :). I guess my board is on its way to me. Thank you. I hope for better results as you mentioned.  One thing I forgot to ask: is there a rate of turn (ROT) calculation and output of any of the examples?

Jordan

kriswiner, 2017-07-18:
The output is in degrees for heading, pitch, and roll, so you can construct
a rate of turn measure from these. I have never done this though, not being
a seaman.

seamaster101, 2017-07-18:
Hi Kris, 
Sorry taking too much of your time. I hope that is the last question I have for a while... 
I'm looking at the explanation that i found below, and I was thinking that I need the speed of angle change form the gyro sensor. Based on your experience, is the output I need as "rate of turn" is just the output from the Gyro sensor in deg/sec? 
or I'm simplifying that too much?

kriswiner, 2017-07-18:
Gyro outputs degrees per second, probably good enough for what you need...

seamaster101, 2017-07-18:
Thanks Kris for all your help!

closed 2017-07-18 by seamaster101

kriswiner, 2017-07-18:
Let me know if you have any trouble with the EM7180.