kriswiner/MPU9250 #162: Unable to configure sleep mode

SreeCA, 2017-07-14:
Hi,
I am using MPU9250-SPI for my project, and i want to put it on sleep mode for a while during the execution , in order to reduce the power draw. The issue is i am not able to configure power management register 1 to configure it to sleep mode. I even tried to disable both gyro and accel on pwr_mgmt_2 , no progress. by the way i am not using magnetometer and temperature sensor. I am able to configure MPU9250 initially as per the requirements to read and update data, further configuration to enable sleep mode is unsuccessful. I tried reading back the pwr_mgmt registers, right after the configuration of sleep mode, which always read..
PWR_MGMT_1 = 0x01 and PWR_MGMT_2 = 0x00.

        {MPUREG_PWR_MGMT_1,         0x80, 100}, // reset
        {MPUREG_PWR_MGMT_1,         0x48, 100}, // sleep mode and temp disable
        {MPUREG_PWR_MGMT_2,         0x3F, 100}, // disable both gyro and accel

SreeCA, 2017-07-14:
I will appreciate your help, Thanks in advance

kriswiner, 2017-07-14:
Not sure what your problem is.

You cannot write to the register?

Or when you set sleep bit to 1 nothing happens?

If you want to save power just disable the gyro and set the sample rate of
the accel to a low value. The accel takes very little power. But do not set
bit 0 in power management 1 to 0, the PLL clock source needs to stay on and
active.

SreeCA, 2017-07-17:
Hi kris,
Thank you so much for the quick reply!!!!
I guess i am not able to write to the register, i tried just disabling temp sensor with no other configuration on PWR_MGMT_1 , but when i read it back it is always 0x01. i tried setting of low sample rate for accel and disable gyro as you suggested still it draws same current as before (4 mA). My goal is to make it less than 1mA, since the current drawn during sleep mode is just 8uA as per the document.

Full Chip Idle Mode Supply Current         8          μA

Thanks
Sree

kriswiner, 2017-07-17:
Not sure what to tell you. If you can;t write to the register there is
something wrong with your I2C bus, your MCU or your sensor.

I can put my MPU9250 to sleep just by setting the sleep bit to 1, or I can
disable the gyro and drop the power by several mA, etc.

Maybe you need a better breakout? MCU? Don;t know...

SreeCA, 2017-07-17:
I forgot to mention that i am using SPI interface not I2C. i am able to write other registers like accel config, gyro config etc.., and successfully configure to enable and read both accel and gyro. The problems are with two power management register configuration.  may be bad chip????

Thank you
Sree

kriswiner, 2017-07-17:
No idea...

peter64, 2017-10-13:
Hey SreeCA,

Not to drag up an old issue (that isn't really an issue with this code anyways) but I'm running into the same problem.
Did you figure it out? I've tried everything and the sensor just appears to keep cycling away even in sleep mode. It seems to cycle at different speeds depending on whether I use sleep mode or not so I know calling sleep mode is doing something. But it doesn't appear to be fully putting the sensor to sleep.

When I don't call sleep 
The current is 1.6mA
The duty cycle of the 2.4mA currents is 1851.85 Hz

When I do call sleep 
The current is 0.144 mA
The duty cycle of the 2.4mA currents is 263.15 Hz

Does anyone know why even when I set the SLEEP mode bit in PWR_MGMT_1 it is still cycling away like this. Any idea which part of the chip is doing this.

In the datasheet on page 21 it shows an LDO, could this be the source?
https://www.invensense.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf

I'm having difficulty understanding how I could ever get this chip down to the 8.4 uA described on page 11 of the datasheet under 1 Hz Low Power Accelerometer, when even in sleep mode it's using 144 uA. I need to look at my breakout board more but I already removed the voltage regulator which wasn't necessary as I'm running 3.3V and the large 10uF capacitor. It should now be basically connected in a pretty minimal fashion.

http://www.ebay.ca/itm/SPI-IIC-MPU-9250-9-Axis-Attitude-Gyro-Accelerator-Magnetometer-Sensor-Module-/131185485892?

Any idea why it might be like this. I have 6 boards and they all do the exact same thing, so I don't think it's a flaky chip. I also tried turning off the compass by sending it it's own sleep mode command separately after enabling I2C slave bypass. It didn't have any effect.

[image]
[image]
[image]
[image]

kriswiner, 2017-10-13:
> MPU-9250A-01-v1.1.pdf
> Gyro-Accelerator-Magnetometer-Sensor-Module-/131185485892?

You likely need to disable the gyro as well as the mag. Sleep just affects
the analog circuitry but the sensors will keep using power (especially the
gyro) unless you disable it.

peter64, 2017-10-13:
Hey Kris,
Thanks for the suggestion. I ran some more tests with different orderings of the commands to disable the sensors themselves.

    {MPUREG_PWR_MGMT_1,         0x80, 100}, // reset
delay(100);
    {MPUREG_PWR_MGMT_2,         0x3F, 100}, // disable both gyro and accel  
delay(100);
    {MPUREG_PWR_MGMT_1,         0x48, 100}, // sleep mode and temp disable

Nothing seemed to make any difference it seems to still draw around 144uA to 200uA. I'll keep digging.

kriswiner, 2017-10-13:
The 200 uA might just be your MCU, etc . In other words, the other
components besides the MPU9250. We manage to get the total sleep current
(MPU9250, STM32L432, BMA280, SPI flash, LDO) to 55 uA this same way.

What else is on the circuit?

peter64, 2017-10-13:
Yeah I'm using a whispernode avr lora, and I've measured the sleep current of the rest of the system down at 4.4uA.
https://bitbucket.org/talk2/whisper-node-avr-lora
I'll keep fiddling and see if I can figure out whats causing it. 
Just to make sure, what voltage and logic levels are you running your STM32L432 and MPU9250 at?

kriswiner, 2017-10-13:
3V3

peter64, 2017-10-14:
Thanks again Kris,

Just to correct myself.
I just realized I was measuring using uCurrent range but the pulses from the battery were 15mA in height so I wasn't getting the full average power right. What I'm actually getting when I take into account the 15mA pulses instead of 2.4 mA is as follows. I'll try to figure out why.

with sensor awake 1.666 mA
with sensor asleep (and gyro's and accelerometers off) 0.45685 mA
with no sensor 0.00676 mA

kriswiner, 2017-10-14:
Definitely something wrong then. These are my "sleep" functions:

void MPU9250::gyromagSleep()
{
  uint8_t temp = 0;
  temp = readByte(AK8963_ADDRESS, AK8963_CNTL);
  writeByte(AK8963_ADDRESS, AK8963_CNTL, temp & ~(0x0F) ); // Clear bits 0
- 3 to power down magnetometer
  temp = readByte(MPU9250_ADDRESS, PWR_MGMT_1);
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, temp | 0x10);     // Write bit 4
to enable gyro standby
  delay(10); // Wait for all registers to reset
}

void MPU9250::gyromagWake()
{
  uint8_t temp = 0;
  temp = readByte(AK8963_ADDRESS, AK8963_CNTL);
  writeByte(AK8963_ADDRESS, AK8963_CNTL, temp | _Mmode ); // Reset normal
mode for  magnetometer
  temp = readByte(MPU9250_ADDRESS, PWR_MGMT_1);
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x01);   // return gyro and accel
normal mode
  delay(10); // Wait for all registers to reset
}

and

void MPU9250::MPU9250wake()
{
  // Clear sleep bit to wake MPU9250
  uint8_t c = readByte(MPU9250_ADDRESS, PWR_MGMT_1);
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, c & ~(0x40) ); // Clear sleep mode
bit (6), enable all sensors
  delay(100); // Wait for all registers to reset
}

void MPU9250::MPU9250sleep()
{
  // Put the device to sleep
  uint8_t c = readByte(MPU9250_ADDRESS, PWR_MGMT_1);
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, c | 0x40); // Set sleep mode bit
(6), disable all sensors
  delay(100); // Wait for all registers to reset
}

when I invoke these:

 MPU9250.MPU9250sleep();  // start with MPU9250 asleep
 MPU9250.gyromagSleep();  // start with MPU9250 gyro and mag off

in the main code my current usage in << 100 uA.

peter64, 2017-10-14:
Hey Kris,

Thanks again I think I've tracked down the issue after fully mapping out the breakout board. I'm pretty sure the issue is the breakout board pull-ups. I missed one pull-up related to ADO. Which I need to set to VCC to get the second I2C address. That's sinking 330 mA which probably makes up the difference. I'm going to de-solder it tomorrow and see if that fixes things. I should have checked the simple stuff before diving into the more complicated stuff. Thanks again for all your tips.

kriswiner, 2017-10-14:
I will be surprisedif this is the problem, usually pullups on gpios don;t
sink a lot of current. Let me know if this fixes the problem.

peter64, 2017-10-15:
Yeah it fixed the problem. It was actually a pull-down on ADO and the issue was I needed to have ADO set to vcc to change the I2C address. I didn't have any io pins left to just toggle ADO when I needed to do I2C so I had to hook it to 3.3V this was constantly sinking 330 mA. I hadn't realized they didn't leave ADO floating on the breakout board. Anyways the whole system is now running at about ~15 uA when sleeping with wake on motion and 4 hz ODR. 

I also tested the sensor independently of the MCU and it seemed to perform with the following currents

on reset				h6b h80		3061 uA
then power down gyro and accel		h6c h3f		196.8 uA
then power down ptat			h6b h08		186.5 uA
then power down clock			h6b h07		36.8 uA
then sleep 				h6b h40		10 uA

on reset				h6b h80		3061 uA
after disabling gyros			h6c h07		418.2 uA
after config accel lowpassfilt		h1d h08		418.2 uA
after config int latch			h37 h22		418.2 uA
after config int wom			h38 h40		418.2 uA
after config motion detection		h69 hc0		418.6 uA
after config wom threshold		h1f h08		418.5 uA
after config detect freq 3.91hz		h1e h04		418.5  uA
after config power cycle		h6b h20		~11.6 uA (peaks to 15.7 every 2 seconds)
after config detect freq 0.24hz		h1e h00		~11.5 uA (peaks to 18.7 every 8 seconds)
after config detect freq 0.49hz		h1e h01		~11.5 uA (peaks to 18.7 every 6 seconds)
after config detect freq 0.98hz		h1e h02		~11.5 uA (peaks to 15.4 every 4 seconds)
after config detect freq 1.95hz		h1e h03		~11.6 uA (peaks to 15.4 every 3 seconds)
after config detect freq 3.91hz		h1e h04		~11.6 uA (peaks to 15.7 every 2 seconds)
after config detect freq 7.81hz		h1e h05		~12.5 uA (peaks to 15.3 every second)
after config detect freq 15.63hz	h1e h06		~16.2 uA (peaks to 18.1 every second)
after config detect freq 31.25hz	h1e h07		22.2 uA
after config detect freq 62.5hz		h1e h08		32.4 uA

Also you were correct in your original assertion that calling sleep is sufficient to get the current usage to 10uA. Based on my testing you didn't need to power down the sensors separately. Infact in one test calling sleep 0x40 after turning off the clock actually turned it back on instead of going to sleep :).

Thanks for all your help.

ezechiele2517, 2020-07-13:
Hi everyone, I want to know the data rate and BW when the MPU9250 is configurated in SLEEP MODE by PWR_MGMT_1 register.
In the previous comments of this post I read: 
When I do call sleep
The current is 0.456 mA
The duty cycle of the pulses is 263.15 Hz

Kingwulin, 2022-12-01:
> Definitely something wrong then. These are my "sleep" functions: void MPU9250::gyromagSleep() { uint8_t temp = 0; temp = readByte(AK8963_ADDRESS, AK8963_CNTL); writeByte(AK8963_ADDRESS, AK8963_CNTL, [...]

Hi,Kris Winer,I found it is fail to read the gyro, accel and mag data from reg after I set the module to sleep mode and then call the wake up as you suggested

kriswiner, 2022-12-01:
"it don't work for me" is not enough information, sorry.

I have no idea what MCU you are using, whether the MPU9250 otherwise works,
and how you are sleeping/waking the MPU9250 so I can't offer any advice
beyond "look at the sketch where this has been successfully implemented...."

Can you get the MPU9250 to work at all?

Can you get the MPU9250 to enter sleep mode with only the accel working?

How are you putting the MPU9250 to sleep? How waking up? How do you know it
is asleep/not waking up?