kriswiner/MPU9250 #163: Detect full 360º rotations

hyshuke, 2017-07-16:
Hello, first of all, thanks for sharing your code! It really helped me start up my work.

I'm doing a thesis about hand tracking and i've decided to use the MPU9250 (Ended up getting a MPU9255 from China instead) on an Arduino Micro to detect the hand's rotation.

I'm using a modified version of your code that includes the magnetometer calibration on each run (I found it on: https://create.arduino.cc/projecthub/30503/using-the-mpu9250-to-get-real-time-motion-data-08f011). I have also tweaked it to match the latest version of MPU9250BasicAHRS.ino minus the displays and changed the WHO_AM_I to 0x73 (because of my mpu version).

I got the yaw, pitch and roll values showing, but there are some problems that i can't find a way to solve:
**1** - Roll works from -180 to 180, but whenever it gets bigger than 45 or lower than -70, Yaw starts getting some really messy values.
**2** - Pitch only works from almost -90 to almost 90, the values start falling back whenever i get near them. They also mess up Roll and Yaw values.
**3** - If i turn the board upside down, the values get kinda lost.

I want to achieve something like this: http://x-io.co.uk/open-source-ahrs-with-x-imu/
Any ideas on what I can do to achieve this level of detection? Am i missing something on the code?

kriswiner, 2017-07-16:
Assuming proper calibration, next item is proper order of sensor data into
the madgwick fusion filter, then fast enough fusion rate...

hyshuke, 2017-07-16:
I think calibration is somewhat ok, your code calibrates it every startup, and by adding up the magnetometer calibration, all of the values seems ok.
I did 3 tests to see if the values are close to each other:

**TEST 1**
MPU9250 I AM 73 I should be 73
MPU9250 is online...
x-axis self test: acceleration trim within : 100.9% of factory value
y-axis self test: acceleration trim within : 99.2% of factory value
z-axis self test: acceleration trim within : 100.6% of factory value
x-axis self test: gyration trim within : 98.4% of factory value
y-axis self test: gyration trim within : 99.2% of factory value
z-axis self test: gyration trim within : 99.6% of factory value
MPU9250 bias
 x   y   z  
19 11 54mg
3.5 1.2 0.9o/s
MPU9250 initialized for active data mode....
Mag Calibration done!
X-Axis sensitivity adjustment value 1.17
Y-Axis sensitivity adjustment value 1.18
Z-Axis sensitivity adjustment value 1.13
AK8963
ASAX 
1.17
ASAY 
1.18
ASAZ 
1.13

**TEST 2**
MPU9250 I AM 73 I should be 73
MPU9250 is online...
x-axis self test: acceleration trim within : 101.0% of factory value
y-axis self test: acceleration trim within : 99.3% of factory value
z-axis self test: acceleration trim within : 100.4% of factory value
x-axis self test: gyration trim within : 98.4% of factory value
y-axis self test: gyration trim within : 99.2% of factory value
z-axis self test: gyration trim within : 99.6% of factory value
MPU9250 bias
 x   y   z  
14 12 56mg
3.5 1.2 0.9o/s
Mag Calibration done!
X-Axis sensitivity adjustment value 1.17
Y-Axis sensitivity adjustment value 1.18
Z-Axis sensitivity adjustment value 1.13
AK8963
ASAX 
1.17
ASAY 
1.18
ASAZ 
1.13

**TEST 3**
MPU9250 I AM 73 I should be 73
MPU9250 is online...
x-axis self test: acceleration trim within : 100.9% of factory value
y-axis self test: acceleration trim within : 99.3% of factory value
z-axis self test: acceleration trim within : 100.4% of factory value
x-axis self test: gyration trim within : 98.4% of factory value
y-axis self test: gyration trim within : 99.2% of factory value
z-axis self test: gyration trim within : 99.6% of factory value
MPU9250 bias
 x   y   z  
17 12 56mg
3.5 1.1 0.9o/s
Mag Calibration done!
X-Axis sensitivity adjustment value 1.17
Y-Axis sensitivity adjustment value 1.18
Z-Axis sensitivity adjustment value 1.13
AK8963
ASAX 
1.17
ASAY 
1.18
ASAZ 
1.13

On those tests, the magnetometer and gyro were the same, only accelerometer had a slight difference.

One thing that seems weird is the factory values trimming. Weren't they supossed to be lower? Should i just make the value -100% to find the difference?

Now i'll try to find out the order of sensor data and how to change the fusion rate to change all of them to -180 to 180. Thanks for the help!

kriswiner, 2017-07-16:
The accel and gyro results look reasonable. The magnetometer is showing
just the Fuse ROM corrections from the factory. Where are the bias
measurements?

On the self test, there is an error in most of these sketches. The test is
intended to use only the least significant 8 bits of the measurement
results (LSbyte). For most MPU9250 this error doesn't change the test
results, for some it is off by 1, which is why you are seeing ~100%. The
self test is just a convenience, not critical to proper functioning.

To check mag offset bias calibration, record mag data while moving the
sensor all around, then plot Mx vs My, My vs Mz, and Mx vs Mz. Are the
clouds of points centered on the origin and reasonably circular?

hyshuke, 2017-07-16:
I printed the mx, my and mz after calibration and left it running for 2 minutes.

I used Excel to plot the data and got the following results (I drew the red circle to see if it was reasonably circular):
[image]

What do you think? I checked your wiki page about magnetometer calibration, this seems calibrated doesn't it?

kriswiner, 2017-07-16:
Was this moving the sensor all around for two minutes? If so, it looks good
enough...

hyshuke, 2017-07-16:
Yes, i moved it all around my desk as much as the cable allowed me to.

Now that calibration seems ok, i'll go look for the other two items you said: Proper order of sensor data into the madgwick/mahony fusion filter and a faster fusion rate.

Thanks for all the help so far! I'll come back here later to show my results :+1:

FN0005, 2019-10-30:
> 我在校准後打印了mx，my和mz，並使其運行了2分鐘。
> 
> 我使用Excel繪製數據並獲得以下結果（我畫了紅色圓圈以查看它是否合理地呈圓形）：
> [image]
> 
> 你怎麼看？我檢查了有關磁力計校準的Wiki頁面，這似乎已經校準了，不是嗎？

How to print the Mxy Mxz Myz point picture with excel?

kriswiner, 2019-10-30:
Is this not what you are showing here?