kriswiner/MPU9250 #165: About magnetometer calibration

ijunglee, 2017-07-31:
Hello, I would like to ask that  how to check if the mag calibration is finished and the data after calibrating is correct and can be used? Are there any parameters which I can check for the issue? I want to write the checking process into the program to check if the mag calibration is finished; however, I have no idea how to check it with certain parameters......Thanks in advance!

kriswiner, 2017-08-01:
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

Sanjanawinchester, 2018-07-01:
thanks, @kriswiner, great job with the library. Can you please tell how and where should I add above code in your library.

kriswiner, 2018-07-01:
What code?

There are plenty of examples of how to use the mag calibration in the
github repository. Here's one:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250