kriswiner/MPU9250 #168: Strange Data from GY-80

afnan, 2017-08-03:
Hi, 
I have been reading your libraries and envied your skills. I have managed to use your GY-80 code but data is strange erratic behavior with no sense of rotation. After search on your answers to others i came to conclusion that 

1. I have not calibrated Magnetometer 
2. I have not calculated Declination for where i live.

I came across this sweet article https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration by you and tried to implement the routine for my program. 

So far the missing variable is `MPU9250magCalibration `which you have calculated in your `initAK8963` in MPU9250 code. 

**Questions:**

1. Is there any workaround for calibration ? 
2. Following link states that my Magnetometer has capability of self calibration i wonder if its worth https://github.com/helscream/HMC5883L_Header_Arduino_Auto_calibration/blob/master/Core/Compass_header_example_ver_0_2/Compass_header_example_ver_0_2.ino
3. How to Calculate Declination ?

Thanks

kriswiner, 2017-08-03:
> MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration by you and
>    HMC5883L_Header_Arduino_Auto_calibration/blob/master/Core/
>    Compass_header_example_ver_0_2/Compass_header_example_ver_0_2.ino

No substitute for mag calibration, absolutely required.

Look up your declination in an atlas, it is not calculated.

afnan, 2017-08-03:
What about missing last variable ` dest1[0] = (float) mag_bias[0]*MPU9250mRes*MPU9250magCalibration[0];`? which is **MPU9250magCalibration**

kriswiner, 2017-08-03:
Are you using an MPU9250?

afnan, 2017-08-04:
No I am using GY-80 (HMC5883L) with NodeMCU running at 80MHz

kriswiner, 2017-08-04:
Then set mpu9250 calibration to 1

afnan, 2017-08-04:
So setting that variable to 1 and performing calibration and recording value while making 8 figure with device gives following images. 
**After Calibration** 
[image]

**Before Calibration** 
[image]

I am not sure if routine was success but problem is

1. Rotate against YAW other two remains constant. Yaw does not show 180 to -180 
2. Rotate against any other all get changed. 

What should i look next into?