kriswiner/MPU9250 #169: NED angles into RPY boat convention

leforban, 2017-08-09:
I spend a hard time in order to get RPY angles for boat attitude and didn't succeed to find a solution yet. MPU9250 returns RPY in NED frame and this seems not to be what I want to obtain. May you give me some advice on how to achieve this?

kriswiner, 2017-08-09:
If not NED, what do you want.

leforban, 2017-08-09:
May be I am wrong but it seems to me that roll and pitch are dependant when yaw is not at zero in NED. I am working on a SEN-14001 module from sparkfun and Euler angles return by the DMP does not give roll pitch and yaw as commonly used on a boat as described here: https://en.wikipedia.org/wiki/Ship_motions.

Sorry I am really lost with coordinate system and transformations

kriswiner, 2017-08-09:
If you are using the DMP, you need to be talking with Invensense not me.
The DMP cannot do 9 DoF fusion.

This would be a better solution:

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/?pt=full_prod_search

leforban, 2017-08-11:
Hello Kris and thanks for your advice. However  datasheet indicates that MPU-9250 directly provides complete 9-axis MotionFusion™output without describing it. I don't know how this DMP can compete against Mahony or Madgiwk algorithm. I am extracting Euler angles directly from the DMP and print a 3d solid using processing and the resulting orientation is really reliable. However As I told, this is not exactly what I expected. Indeed examining the angle that is supposed to be Roll, I can increase this angle by rolling the board (expected) but also by pitching it. This seems to me that therefore rolling the board involves something like a trigonometric projection but I have not enough math skills to extract pure heel , trim and heading from roll pitch yaw...  However I have recognized a Tait-Bryan transposed rotation matrix. I will try to extract from that. If you have any idea, I would appreciate :-)

kriswiner, 2017-08-11:
DMP cannot do 9 DoF fusion, this must be done on the host.