kriswiner/MPU9250 #170: How to use MPU9250 with a 5V 16Mhz Arduino Pro mini

AbhiBjee, 2017-08-10:
Hi Kris;

I am presently working for a biomedical device project where I need to integrate a Force sensor load cell which operates in 5V and an MPU9250 which operates in 3.3V. 

However I have used your codes and tried the same with a 3.3V 8Mhz arduino pro mini and It works fine. But in order to improve the accuracy of our system we do need to use a 5V arduino pro mini with a 5V FTDI connector.

However I connected the MPU 9250 to a stepped down 3.3V power using a voltage divider ckt of 1K7 and 3K3 resistors and I connected the SDA and SCL to A4 and A5 of the ARDUINO without pull up resistors.

The Serial Monitor says:
MPU9250 i am FF i should be 71 
Could not connect to mpu9250: 0xFF.

I also tried to test the I2C scan : 
It says : 
No I2C devices found. 

 How should I solve this any Idea ?

kriswiner, 2017-08-10:
pullup resistors.

AbhiBjee, 2017-08-10:
I was probably expecting this answer from you.

I tried it with two different pull up resisters 
1) 10K 
2) 4.7K 

but there is no display on the serial monitor at both the instances
also when I tried the I2C scan it says 

no I2C devices found for resistors 4.7K

error at address 0x0F for resistors 10K and then again no I2C devices found.

Is there any way out to solve this issue ??

kriswiner, 2017-08-10:
Are you sure you have the right pins for SDA/SCL? Any other I2C sensors
that work? etc

AbhiBjee, 2017-08-10:
I am definitely sure that the SDA/SCL works because When I plug the 3V3 FTDI to the same board with all connections same the MPU9250 works fine with your code, but when I use the 5V FTDI to the same board and step down the voltage to 3V3 using the voltage divider, it shows the above mentioned errors. 
However the MPU9250 works fine in Arduino UNO without any pull up resistors. 
I have space limitations so I am bound to use Pro mini