kriswiner/MPU9250 #171: Scope of the function problem

shashwathsb1998, 2017-08-11:
MPU9250_MS5637_AHRS_t3:312: error: 'I2C_MASTER' was not declared in this scope

   Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

              ^

MPU9250_MS5637_AHRS_t3:312: error: 'I2C_PINS_16_17' was not declared in this scope

   Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

                                ^

MPU9250_MS5637_AHRS_t3:312: error: 'I2C_PULLUP_EXT' was not declared in this scope

   Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

                                                ^

MPU9250_MS5637_AHRS_t3:312: error: 'I2C_RATE_400' was not declared in this scope

   Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

                                                                ^

D:\IDM Downloads\Compressed\MPU9250-master\MPU9250_MS5637_AHRS_t3\MPU9250_MS5637_AHRS_t3.ino: In function 'void loop()':

MPU9250_MS5637_AHRS_t3:521: error: 'MadgwickQuaternionUpdate' was not declared in this scope

     MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);

                                                                                                    ^

D:\IDM Downloads\Compressed\MPU9250-master\MPU9250_MS5637_AHRS_t3\MPU9250_MS5637_AHRS_t3.ino: In function 'void MS5637PromRead(uint16_t*)':

MPU9250_MS5637_AHRS_t3:1219: error: 'I2C_NOSTOP' was not declared in this scope

           Wire.endTransmission(I2C_NOSTOP);        // Send the Tx buffer, but send a restart to keep connection alive

                                ^

D:\IDM Downloads\Compressed\MPU9250-master\MPU9250_MS5637_AHRS_t3\MPU9250_MS5637_AHRS_t3.ino: In function 'uint32_t MS5637Read(uint8_t, uint8_t)':

MPU9250_MS5637_AHRS_t3:1233: error: 'I2C_NOSTOP' was not declared in this scope

         Wire.endTransmission(I2C_NOSTOP);        // Send the Tx buffer, but send a restart to keep connection alive

                              ^

D:\IDM Downloads\Compressed\MPU9250-master\MPU9250_MS5637_AHRS_t3\MPU9250_MS5637_AHRS_t3.ino: In function 'uint8_t readByte(uint8_t, uint8_t)':

MPU9250_MS5637_AHRS_t3:1340: error: 'I2C_NOSTOP' was not declared in this scope

  Wire.endTransmission(I2C_NOSTOP);        // Send the Tx buffer, but send a restart to keep connection alive

                       ^

D:\IDM Downloads\Compressed\MPU9250-master\MPU9250_MS5637_AHRS_t3\MPU9250_MS5637_AHRS_t3.ino: In function 'void readBytes(uint8_t, uint8_t, uint8_t, uint8_t*)':

MPU9250_MS5637_AHRS_t3:1352: error: 'I2C_NOSTOP' was not declared in this scope

  Wire.endTransmission(I2C_NOSTOP);  // Send the Tx buffer, but send a restart to keep connection alive

I dont know how to solve these errors.

Please do help me out

kriswiner, 2017-08-11:
Are you using a Teensy? The i2c_t3.h wire library is for Teensy only.

shashwathsb1998, 2017-08-11:
No im using an arduino UNO . Any idea where i can get the code for arduino ? :(

shashwathsb1998, 2017-08-11:
Im kinda new at this. What do you mean ?

kriswiner, 2017-08-11:
Ask at the Arduino forum...

anchals02, 2018-09-11:
@shashwathsb1998  Were you able to find any solution to this problem?
I am facing the same issue with Arduino Nano board.

kriswiner, 2018-09-12:
This sketch is specifically for the Teensy. Make sure you are using an
appropriate sketch for the MCU you are using or change the sketch.

kriswiner, 2018-09-12:
If you are using an Arduino try this sketch:

https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_BasicAHRS2_ESP8266.ino

pocolocos, 2018-09-26:
Hi,
i'm facing the same problem using a Teensy 3.5
in detail:
`MadgwickQuaternionUpdate(-ax, ay, az, gx*pi/180.0f, -gy*pi/180.0f, -gz*pi/180.0f,  my,  -mx, mz);`

any ideas?

pocolocos, 2018-09-26:
rehi,

just found the missing code here:
https://github.com/kriswiner/MPU9250/blob/master/quaternionFilters.ino
already working - great stuff!