kriswiner/MPU9250 #172: Invensense Application Notes for MPU9250

TediumRemedy, 2017-08-13:
Hello Kris,
Do you happen to know where I could download Application Notes for the MPU9250 and AK8963? 
It looks like they are no longer available at https://www.invensense.com/developers/ (even though I am registered. Or I just can't find them)
Thanks!

kriswiner, 2017-08-13:
Which application notes do you need? e-mail me at tleracorp@gmail.com.

TediumRemedy, 2017-08-14:
> Which application notes do you need? e-mail me at tleracorp@gmail.com.

I was actually wondering what ANs they have, hoping they have some tutorials on the correct FIFO usage, DMP/hardware sensor fusion and other "best practices" (as opposed to the dry description of device registers and capabilities as they did in the data sheet)
The thing is I am trying to understand how things work, instead of simply copying/porting of your code, and I hope those ANs could help me.

Currently I am trying to reimplement your sensor reading/fusion algorithm using the FIFO buffer.