kriswiner/MPU9250 #175: About the parameters of MPU9250 and Madgwick algorithm

ijunglee, 2017-08-15:
First, I have to say "Thank you so much" to Kris again. I have been here for 2-3 years. I got so much help from your codes. =)
I take a look at the latest version of MPU9250_MS5637_AHRS_t3. I would like to ask something about the parameters setting.

This is my system environment:
MCU: TI MSP430F5438A ([This one](http://www.mouser.tw/ProductDetail/Texas-Instruments/MSP430F5438AIPZR/?qs=sGAEpiMZZMsp%252bcahb6g%252bWz%252bwskpj4jWSYSSbURGebC4%3d))
IMU: MPU9250
Interface: SPI

And this is my initial settings of MPU9250
[image]

I found that when I send the raw data into Madgwick algorithm which runs on the software, it will got drift when I move it to a certain direction. The value will gradually increase or decrease in a certain direction. But in other directions, it works fine.

The order of sensor data I sent to Madgwick is (ax, ay, az, gx, gy, gz, my, mx, -mz)

Each set of data is sent to the program every 20 ms.

I am not sure if the problem is caused by the initial setting or algorithm.  I would like to ask some questions about the parameter setting in MPU9250 and Madgwick algorithm:

1. How do I know the correct sampling rate of acc, gyro and mag? Where can I get the Internal_Sample_Rate? I refer to the register map of MPU9250 but I have no idea how to check it:
[image]

And what's the meaning of Bandwidth, Delay and Fs?
[image]

2. How to set the parameters like GyroMeasError, GyroMeasDrift and beta in Madgwick algorithm? I am also not sure if it cause the problems.

Thanks in advance

kriswiner, 2017-08-15:
>    1. How to set the parameters like GyroMeasError, GyroMeasDrift and

I am not sure what you are doing here. If you enable the low pass filter on
the accel/gyro, the sample rate is set to 1 kHz. Then it is decimated by
1/(1 + SMPLRTDIV), so in your case you are running the accel and gyro at
1000 Hz sample rates and 92 Hz low pass filter, all fine. Now to get
reasonable results with the Madgwick filter, you need to iterate 5 - 10
times for every data sample, this means you need to be running the fusion
filter at > 5 kHz, otherwise the filter can't keep up with the data and
results will lag and drift. I assume you have properly calibrated the
sensors, especially the mag. But if not, you will get terrible results no
matter what you do.

ijunglee, 2017-08-15:
Thank you for helping to check the sensor initializing process and giving the suggestion of Madgwick filter. I would like to ask how to change the speed of the fusion filter? Does it mean I should send the data to the filter at a higher frequency or I can just change the deltat while running the algorithm. I got really confused about this part. Thank you very much!

kriswiner, 2017-08-15:
The strategy is to run the filter as many times as possible for your MCU
every time you receive new data. On a 3V3 AVR Pro Mini running at 16 MHz
you can achieve maybe 100 Hz this way, too slow. With an STM32L4 running at
80 MHz with FPU you can get 60 kHz, really to fast. So it depends on your
MCU. You have to calculate what rate it can achieve. If it is too slow, you
will need a different MCU. If it is too fast, you can simple set the
iterations to, say, 10 or 20, and then control the rate directly.

ijunglee, 2017-08-15:
Thanks for your reply. 
My system is: The MCU check the data of MPU9250 and send it to the program on a notebook via Bluetooth every 20 ms. The fusion filter is running on the program. So I can only get the data every 20 ms.......about 50 Hz. 
Even if I run the fusion algorithm on the MCU, the MCU can only run at 25 MHz......I am not sure how to calculate the data rate it can achieve. But I think it's not enough......

kriswiner, 2017-08-15:
Just iterate on the computer 10 or 20 times for each data set you send it,
this should be enough. If I understand that you are doing fusion on the
computer, is this right?

ijunglee, 2017-08-15:
Yes, you are right. But now I just iterate each data set for just one time.
Do you mean when I receive each data set, I just iterate the algorithm by keeping calling the function MadgwickQuaternionUpdate (like the one in your code) 10 or 20 times?
Thanks in advance.

kriswiner, 2017-08-15:
Yes

ijunglee, 2017-08-16:
Hello, I have tried to iterate the fusion filter 10 times. The parameters I used are same as the settings in your code "MPU9250_MS5637_AHRS_t3.ino"
[image]

When the fusion filter iterates only one time, it converges slowly and has a slight drift.
[1 iteration](https://youtu.be/WRlrCnMo9QE)

However, when fusion filter iterates for 10 times, it gets "serious" drift......
[10 iterations](https://youtu.be/r-AQMw7iUFM)

Do you have any ideas to solve the problems? =(
I think the parameter "beta" is important in this issue but I am not sure how to set it. Thanks in advance.

kriswiner, 2017-08-16:
Assuming you have calibrated the sensors properly, the only other concern
would be with the way you are sending the sensor data to the madgwick
filter.