kriswiner/MPU9250 #176: Yet another Yaw / magnetometer calibration issue

arilotter, 2017-08-15:
Hello! I've followed your https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration guide, but it results in a hard iron bias of {0, 0, 0}.  I can see the magnetometer output, and the soft iron bias is a non-zero number, but the Yaw axis is still totally wrong (almost sinusoidal). The code's running at ~140MHz on an adafruit feather 32u4

kriswiner, 2017-08-15:
> MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration guide, but it

offset bias of 0, 0, 0 is highly unlikely. Looks like you have an error in
implementation.

henzim, 2017-09-08:
From where did you call magcalMPU9250() ?
I had the same problem when I called the function from the setup(), without waiting for     
   if (readByte(MPU9250_ADDRESS, INT_STATUS) & 0x01) {  // On interrupt, check if data ready interrupt
try calling it later in the loop()