kriswiner/MPU9250 #179: Pitch, Roll, Yaw not working correctly in Unity

FreakyMex, 2017-08-28:
Hello,

I am struggling with using the MPU9250. My goal is rotation tracking for a unity-based app. So far I was successful in communicating between the arduino micro and unity, and with using an MPU6050 with this code (using the MPU's DMP and quaternion output) : 

`/* ============================================
              Uduino IMU Project.
  Original MP-6050 library by Jeff Rowberg
  =============================================== */

#include "Uduino.h"
Uduino uduino("IMU");

#include "I2Cdev.h"
#include "MPU6050_6Axis_MotionApps20.h"

#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
#include "Wire.h"
#endif

#define INTERRUPT_PIN 2

MPU6050 mpu;

bool dmpReady = false;  // set true if DMP init was successful
uint8_t mpuIntStatus;   // holds actual interrupt status byte from MPU
uint8_t devStatus;      // return status after each device operation (0 = success, !0 = error)
uint16_t packetSize;    // expected DMP packet size (default is 42 bytes)
uint16_t fifoCount;     // count of all bytes currently in FIFO
uint8_t fifoBuffer[64]; // FIFO storage buffer
volatile bool mpuInterrupt = false;     // indicates whether MPU interrupt pin has gone high

Quaternion q;           // [w, x, y, z]         quaternion container
float euler[3];         // [psi, theta, phi]    Euler angle container
float ypr[3];           // [yaw, pitch, roll]   yaw/pitch/roll container and gravity vector

#define OUTPUT_READABLE_QUATERNION
//#define OUTPUT_READABLE_EULER
//#define OUTPUT_READABLE_YAWPITCHROLL

void setup() {

#if I2CDEV_IMPLEMENTATION == I2CDEV_ARDUINO_WIRE
  Wire.begin();
#elif I2CDEV_IMPLEMENTATION == I2CDEV_BUILTIN_FASTWIRE
  Fastwire::setup(400, true);
#endif

  Serial.begin(38400);
  while (!Serial);

  mpu.initialize();
  pinMode(INTERRUPT_PIN, INPUT);

  devStatus = mpu.dmpInitialize();
  if (devStatus == 0) {
    mpu.setDMPEnabled(true);
    attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), dmpDataReady, RISING);
    mpuIntStatus = mpu.getIntStatus();
    dmpReady = true;
    packetSize = mpu.dmpGetFIFOPacketSize();
    Serial.print("DMP initialzed");
  }
}

void loop() {
  int sensorReading = analogRead(A0);
  uduino.readSerial();

  if (uduino.isInit() && dmpReady) {

[... 540 lines cut ...]
        Serial.print("/"); Serial.print(*(getQ() + 3));
        Serial.print("/");  Serial.println(sensorReading);`

But it didn't work either.

The Unity Code is the following one : 
`  string[] values = data.Split('/');
        float w = float.Parse(values[0]); // ROLL
        float x = float.Parse(values[1]); // PITCH
        float y = float.Parse(values[2]); // YAW
        float z = float.Parse(values[3]); // SENSORREADING
        float translation = float.Parse(values[4]);
        rotation.x = y; // YAW
        rotation.y = x; // PITCH
        rotation.z = w; // ROLL
        this.transform.rotation = Quaternion.Euler(rotation);`

I also tried with Quaternion with (w, x, y, z) and Quaternion.Inverse(w, x, y, z) but it was worse than using euler Angles, it didn't correspond to the movements.

Sorry if my question seems stupid or else, I am a newbie in electronics and Arduino and I am really struggling with this... I googled a read a lot of topics but I can't really understand the problem here. Do I need a filter ?

Here you can see the youtube video of me trying the sensor : 
https://www.youtube.com/watch?v=rs1C5n4_RBc

Thank you very much for you time

kriswiner, 2017-08-28:
> quaternions_and_Euler_angles

First of all you are using Sparkfun's code which is cribbed from me, and I
don;t know what they did to it.

Secondly, try something like the attached instead. You'll have to make some
changes in the wire calls and maybe some other things. It uses an interrupt
shceme but this is easily changed to a pollling scheme for data ready.

What you want is to calculate yaw, pitch, and roll from the quaternions
(done in the code) and this is your input to unity.

FreakyMex, 2017-08-28:
Wow, thank you for your really fast answer. I can't see the attached code...

FreakyMex, 2017-08-28:
I tried to use  	MPU9250BasicAHRS.ino 
from your .zip to use the original code (sorry, didn't see it was modified by SparkFUN, it was the only one that compiled directly). My Arduino IDE says it is not a valid library when I try to import the .zip.
I copy/pasted it in the libraries folder, installed adafruit gfx and the nokia display libraries, then it compiled, tried to upload it on my arduino micro but it was too big, put /* */ around the displays (only this) functions, then it was ok, but my monitor screen in Arduino stays blank...

I'm starting to be desperate ! Sorry for bothering you

FreakyMex, 2017-08-28:
Well, I tried it again with the script I had as you suggested, modified little things and it works like a charm.
I didn't think the 6050 library would work, actually. 

Thank you very much.

closed 2017-08-28 by FreakyMex

kosletr, 2021-05-04:
FreakyMex, could you please share your code because I have a similar issue?

FreakyMex, 2021-05-05:
Hi Kris,

I did not manage this issue, and I don't have this code anymore... Sorry.

Good luck,