kriswiner/MPU9250 #180: Magnetometer and Madgwick or Mahony does not work in case of tilting

mindchanger82, 2017-08-30:
Dear Chris.
First of all thanks a lot for your great work for all in this esp and 9250 things. Great Job.

I'm facing a probem right now. I successfully get the imu to run with your example codes but it seems, that the filter functions have no influence to the mag data when i turn the device over some axes.
EG I mountet the imu to a robotic lawn mower. Everytime the mower moves over a small hill it moves to somewhere but not to the should direction.
I tested it on a breakout board inside with another imu and esp32 device with the same result.
Here is my function to print out the heading degrees.
I placed it in the loop, directly under the serial prints of mx,my,mz
`
float heading = atan2(my, mx);
    float headingDegrees=0;
    float declinationAngle = 30.45/1000;
    heading += declinationAngle;
  
    if(heading < 0)
      heading += 2*PI;
  
    if(heading > 2*PI)
      heading -= 2*PI;
  
    // Convert radians to degrees for readability.
    //float headingDegrees = 360-(heading * 180/M_PI);
    headingDegrees = heading * 180/M_PI;
    headingDegrees = 360.00 - headingDegrees;
    Serial.print("Heading = "); Serial.print( (int)headingDegrees ); Serial.println(" Degrees!");`

So everytime i softly move the device (yaw or pitch) the mag data changes a lot to unsensful values.

Have you got any idea? Is my code wrong for this case?

PS: I used your latest "MPU9250_MS5637_AHRS_t3.ino" and removed all the display stuff.

Thanks in advance and with best regards

Marius

kriswiner, 2017-08-30:
"float heading = atan2(my, mx);", funny!

Why aren't you using the sensor fusion to get heading?

And did you calibrate your sensors to remove offset bias?

mindchanger82, 2017-08-30:
Hi Chris.

Thanks for your reply. That is exactly my problem. Either I don't know how to use the sensor fusion, that's my goal to achive, nor I don't exactly know how to calibrate the sensors. I just used your mag calibration 8.

TediumRemedy, 2017-08-30:
@mindchanger82 the idea is to feed accelerometer, magnetometer and gyroscope measurements into the madgwickQuaternionUpdate, which, when run, fills out the q[0]...q[3] array (quaternion), which you then can convert into 3 angles - heading (yaw), tilt around 1 axis (pitch), tilt around another axis (roll) using simple formulas https://en.wikipedia.org/wiki/Conversion_between_quaternions_and_Euler_angles
madgwickQuaternionUpdate is the sensor fusion algorithm which turns raw sensor measurements into more or less reliable estimate of the Euler angles.

mindchanger82, 2017-08-30:
Dear TediumRemedy.
That means:
/*from wiki*/
   // yaw (z-axis rotation)
   double siny = +2.0 * (q.w() * q.z() + q.x() * q.y());
   double cosy = +1.0 - 2.0 * (q.y() * q.y() + q.z() * q.z());  
    yaw = atan2(siny, cosy);

but from where do I know which arry pos (q[0] or q[xyz] stands for q.w or q.z ?
Sorry but I'm not really seeing through this.

TediumRemedy, 2017-08-30:
@mindchanger82 you can use pieces of code from: https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino

` //   yaw   = atan2f(2.0f * (q[1] * q[2] + q[0] * q[3]), q[0] * q[0] + q[1] * q[1] - q[2] * q[2] - q[3] * q[3]);   
 //   pitch = -asinf(2.0f * (q[1] * q[3] - q[0] * q[2]));
 //   roll  = atan2f(2.0f * (q[0] * q[1] + q[2] * q[3]), q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3]);
 //   pitch *= 180.0f / PI;
 //   yaw   *= 180.0f / PI; 
 //   yaw   += 13.8f; // Declination at Danville, California is 13 degrees 48 minutes and 47 seconds on 2014-04-04
 //   if(yaw < 0) yaw   += 360.0f; // Ensure yaw stays between 0 and 360
 //   roll  *= 180.0f / PI;`

The original code uses another conversion: 
`a12 =   2.0f * (q[1] * q[2] + q[0] * q[3]);
    a22 =   q[0] * q[0] + q[1] * q[1] - q[2] * q[2] - q[3] * q[3];
    a31 =   2.0f * (q[0] * q[1] + q[2] * q[3]);
    a32 =   2.0f * (q[1] * q[3] - q[0] * q[2]);
    a33 =   q[0] * q[0] - q[1] * q[1] - q[2] * q[2] + q[3] * q[3];
    pitch = -asinf(a32);
    roll  = atan2f(a31, a33);
    yaw   = atan2f(a12, a22);
    pitch *= 180.0f / PI;
    yaw   *= 180.0f / PI; 
    yaw   += 13.8f; // Declination at Danville, California is 13 degrees 48 minutes and 47 seconds on 2014-04-04
    if(yaw < 0) yaw   += 360.0f; // Ensure yaw stays between 0 and 360
    roll  *= 180.0f / PI;
    lin_ax = ax + a31;
    lin_ay = ay + a32;
    lin_az = az - a33;`

@kriswiner could you explain why do you prefer this form?

mindchanger82, 2017-08-30:
woaaaaah. Thank you so much.

I used, as you said, the following lines:
`
yaw = atan2f(2.0f * (q[1] * q[2] + q[0] * q[3]), q[0] * q[0] + q[1] * q[1] - q[2] * q[2] - q[3] * q[3]);
yaw *= 180.0f / PI;
yaw += 1.44f; // Declination at Düsseldorf Germany 
if(yaw < 0) yaw += 360.0f; // Ensure yaw stays between 0 and 360
`
The data is now very stable, but false :-) haha, real south is 279 degrees with the code... mhhhhhh where is the error.

TediumRemedy, 2017-08-30:
That's a good question.
You could "calibrate" it by just adding a number of degrees to yaw that make yaw 0º at the location where a real compass points to north, instead of adding the declination to yaw.
I am getting a wrong geographical heading value too (I don't need absolute heading though).
Maybe Kris might know why this is happening?

kriswiner, 2017-08-30:
The other formulation is to calculate members of the rotation matrix from
which gravity and linear acceleration are easily estimated.

kriswiner, 2017-08-30:
The Madgwick filter is NED, so choose which accel axis you want for North,
and then, make sure you feed in sensor data into the Madwick filter
accordingly. I usually choose accel x as North and also negate the accel
data, this gives me a true North reading along the accel x-axis modulo the
declination.

My colleague says Madgwick is ENU, so maybe this is why negating the accel
data works better.

But the point is, it needs to be consistent.

TediumRemedy, 2017-08-30:
Thanks Kris

mindchanger82, 2017-09-08:
Thanks Kris and TediumRemedy.
Everything works fine now.

closed 2017-09-08 by mindchanger82

henzim, 2017-09-09:
@mindchanger82 glad you got your setup working! I am also faced with "yaw" problems. Could you please have a look at [https://github.com/kriswiner/MPU9250/issues/182](my issue) and confirm you don't have this problem in your setup. In brief, the problem is that the yaw angle "misses" a couple of degrees when rotating the setup, e.g., I rotate the MPU by 90° and the yaw changes by 80°... Thanks alot!

mindchanger82, 2017-09-12:
Dear henzim.
Sorry for my late reply. I'm on a biz trip and I'll be back on sunday. I will try to find out if I have the same prob too.
Thanks

mindchanger82, 2017-10-17:
Hi Henzim. Sorry for my late replay.
I tested it the last weekend and and I don't get these kind of errors. But keep in mind that I use a esp32 and it seems to be a "bit faster" than the arduino mini pro.

Best regards

henzim, 2017-10-17:
Hey, no problem, thanks a lot for testing!It really seems to depend a lot on the CPU performance.Although I am still not quite convinced... I did try sending the values to the PC and doing the integration there but the results were still far from perfect.Now I'll try my luck with this powerful beauty: https://www.tindie.com/products/onehorse/nrf52832-development-board/

Kaijudo2, 2022-11-01:
I am having issue with 'TwoWire i2c'. Program says "no matching function for call to 'TwoWire::TwoWire()'". Would really appreciate if anyone could help me!

Kaijudo2, 2022-11-01:
> What MCU are you using?

I am using adafruit esp32 feather

kriswiner, 2022-11-01:
See https://github.com/kriswiner/ESP32