kriswiner/MPU9250 #181: acceleration in world frame

infamous123, 2017-08-31:
Hi and first of all thanks for all your nice work.

I am using MPU9250_MS5637_AHRS_t3.ino and i am trying to get the acceleration in the world frame of reference regardless of how the sensor is tilted in space.

In order to get acceleration without gravity I had to modify your code a little bit:

using:
```
lin_ax = ax + a32;
lin_ay = ay - a31;
```
instead of:
```
//lin_ax = ax + a31;
//lin_ay = ay + a32;
```
With this modification, the lin_ax/ay/az are always around 0 when they are at rest, regardles of how i 
position my sensor.

I am not sure how to interpret this and i thought that since i can see the presumably intended effect with the above modification it is probably due to set up differences (i am using the sparkfun breakout board).

...
Following that, as i said i am trying to graph the acceleration in the world frame of reference. For that, the way i am finding the acceleration in the x axis is a bit naive, and i tried to combine the accelerations in the sensor frame with the angles in the world frame (Y/P/R); this is what i do for the x axis (perpendicular to Yaw=0): 

```
float x = lin_ax * cos(pitch) * cos(roll) * sin(yaw) 
        + lin_ax * cos(pitch) * sin(roll) * sin(yaw)
        + lin_ay * cos(pitch) * cos(roll) * cos(yaw)
        + lin_ay * sin(pitch) * cos(roll) * cos(yaw)
        + lin_az * cos(pitch) * sin(roll) * cos(yaw)
        + lin_az * sin(pitch) * cos(roll) * sin(yaw);
```
By first sight this seems to be working alright but i was wondering if there is a cleaner way to do it. What i would like to ask is if there is a more straightforward way to do it by adding up some quaternion products similar to a31, a32 etc.. - or if you have any advice in calculating  the acceleration the above way.

kriswiner, 2017-08-31:
Not exactly sure what you are trying to do, but the linear accelerations
have gravity removed; isn't this what you want?

infamous123, 2017-08-31:
I want to be able to swing the sensor to the right and left, with myself facing north (where yaw is 0), and regardless of the sensor's angular position, lin_ax to update. 

(for example, right now, if i swing the sensor along the y axis, facing north, lin_ay increases)

kriswiner, 2017-08-31:
If you choose Ax to lie along North, the swinging the sensor side to side
in y direction you will show the correct linear acceleration along the
y-axis and zero along the x-axis.

Still not sure what the problem is.

If you are using Madgwick's algorithm, choose Ax to be North, then the East
and Down components are defines for all sensors and the algorithm expects
the sensor data to be fed in as NED, that is (AN, AE, AD, GN, GE, GD, MN,
ME, MD). then the  Euler angles should work correctly.

infamous123, 2017-08-31:
First of all thanks a lot for your replies and sorry if there is something i am missing here.

yes it works fine in that sense.

The scenario you describe above is what indeed happens: where Ax lies along north, and the swinging happens side to side in north direction, i am getting the linear acceleration along the y-axis and zero along x.
-> but then, if i rotate the sensor and where was y-axis now it is x-axis and perform the exact same swing i want to consider this again as y acceleration and zero at x.

Meaning that no matter along which x,y,z accelerometer axis the force is applied to, if it happens along North an imaginary "real_lin_ay" axis to show the acceleration.

kriswiner, 2017-08-31:
Yes, for that, you need to feed the sensor data into the Madwick filter in
the proper way.

infamous123, 2017-08-31:
ok i see.. 

thanks alot for the replies, i will look into the filter more closely then.

kriswiner, 2017-08-31:
I use NED and then negate the accel, like this:

MadgwickQuaternionUpdate(-ax, ay, az, gx*pi/180.0f, -gy*pi/180.0f,
-gz*pi/180.0f,  my,  -mx, mz);

for MPU9250 and choosing North == Ax;

don;t forget the delination correction if you want True North == zero Yaw.

infamous123, 2017-09-04:
Just for closure i wanted to say that i didn't manage to get the behavior we describe in situations above. The strange thing is that Y/P/R are ok and they seem pretty accurate; also the gravity is successfully removed with the lin_ax, lin_ay changes i mention in my first post.

What i did was:
1. c/p the two files MPU9250_MS5637_AHRS_t3.ino and quaternionFilters.ino into one, 
2. removed the display code,
3. change attachInterrupt to: attachInterrupt(digitalPinToInterrupt(intPin), myinthandler, RISING);
4. modified Wire code since i am using arduino board,
5. changed some serial output text,
\+ what i mention in my first post

MadgwickQuaternionUpdate(-ax, ay, az, gx*pi/180.0f, -gy*pi/180.0f, -gz*pi/180.0f,  my,  -mx, mz); was indeed the function i found in MPU9250_MS5637_AHRS_t3.ino.

In any case i think i will approach my problem in a different way, avoiding IMUs , and i would like to thank you again for replying and providing such resources.

closed 2017-09-04 by infamous123