kriswiner/MPU9250 #182: Wierd yaw offset when rotating

henzim, 2017-09-08:
Hello Kris!

Thank you for excellent MPUxxxx libraries and documentation!
I have been using your MPU6050 code successfully and now wanted to remove the yaw drift by switching to MPU9255. Initially I wanted to post yet another question about the magnetometer calibration, but your documentation helped me fabulously with that
[image]
 and I have now successfully decoupled Yaw from Pitch and Roll.

Unfortunately, I have a problem with the Yaw angle itself (during rotations) and I was hoping you might have an idea. 

**Setup**:
Arduino Pro Mini 8MHz
MPU9255 (0x73)
Breadboard

**Problem**:
Starting at 0° performing a 90° rotation of the breadboard leads to a 97° change in Yaw. After another 90° rotation the Yaw is correctly +/- 180°.

I am using 100Hz magnetometer sampling and tried both Mahony and Madgwick.
(Interestingly the Yaw resulting from the two algorithms is also about 7° apart. I.e, if I leave the breadboard at a position where Madgwick yielded Yaw=97° and restart with Mahony the Yaw is coincidentally now about 90°.)

Here are my Yaw readings with both approaches starting from Yaw +/-180°

**Mahony**
Rotation  Yaw
0°    ->  180°
-90°  ->  +90°
-180° ->  -**13**°
-270° ->  -**101**°
Here the -90° rotation seems fine, also the rotation from -180 to -270, but between -90° and -180° something happens.

**Magdwick**
Rotation  Yaw
0°    ->  180°
-90°  ->  -**104**°
-180° ->  +/-0°
270°  ->  +**105**°
Here a 180° rotation is correctly represented by Yaw, but 90° rotation are not.

I might be missing something obvious, but since I cannot figure it out, I was wondering if you have any idea?

By the way, I just noticed a strange behaviour: When fixating the breadboard where Yaw=180°, the yaw angle varies very little (+/-0.5), however, turning the breadboard 180°, where Yaw should be about 0°, there is a much higher **variance (+/-2.5)**
Again, any idea?

Thanks in advance!
//Henrik

kriswiner, 2017-09-08:
Hi Henrik,

I am assuming you also correctly calibrated the accel and gyro. If not i
recommend you do so and repeat these tests.

The MPU9250 will be sensitive to external magnetic fields from laptops and
USB cables, etc. Likely this is the source of these excursions.

On the other hand, you shouldn't expect better than ~+/- 4 degrees rms
accuracy from these algorithms. That is just as good as they can do, and
your results aren;t too far from this.

Kris

henzim, 2017-09-08:
Hi Kris,

thanks for you speedy reply! It is reassuring that my errors are not too far from the truth.
(Shortly after posting my question I also noticed that [you had similar accuracy problems](https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion) around Yaw 0°. I still do not understand why the variance is generally smaller at other angles.)

I do have a USB cable connected (for Serial and power), but will replace it by a bluetooth module HC-05 and a LiPo and try again.

About calibrating accel and gyro. I use the 
`void calibrateMPU9250(float * dest1, float * dest2) {
from your sketch. This calibration works by computing offsets (if I understood it correctly) from reading accel/gyro data while the MPU is at rest. Are there any other calibration options for accel/gyro?

//Henrik

henzim, 2017-09-08:
Just tried with bluetooth at the kitchen table, for the a computer or usb-cable... Did not help. Partly the values where alot worse, missing yaw 20° or so from one 90° rotation...

kriswiner, 2017-09-08:
Yes, did you also recalibrate the mag with the new configuration?

henzim, 2017-09-08:
No, true, I did not do that. Will try tomorrow. Thanks

henzim, 2017-09-09:
I now perform the mag calibration at every start up (as opposed to doing it once and hard code the biases). Now again with my simple setup consisting of Arduino Pro Mini, MPU9250, HC-05 and LiPo, I went away from the computer only to find no improvement.

I am still missing 10-15° per 90° rotation of the setup. Starting at Yaw=180° and turning the board slowly about 90° then Yaw becomes 259° (missing about 11°)

Is anyone else experiencing this? 

Are there absolute attitude sensors that do not suffer from this? Can it be attributed to the 2.5$ MPU?

> On the other hand, you shouldn't expect better than ~+/- 4 degrees rms accuracy from these algorithms. 
@kriswiner Your answer seems to imply, that it might (just?) be an algorithmic/computational problem, correct? Meaning it is not primarily a limitation of the cheap MPU?

If so, something like ["BWT901 / BJY901"](http://a.co/hWsaKWc), which supposedly has an onboard Kalman-filter, should do the trick, right? Incidentally I have a BJY901 in my possession (alas I don't have a Windows computer to run the manufacturer calibration software), but "even" in the factory/uncalibrated state I get the same problem: The yaw angle changes seemingly smoothly but does not add up in the end, say after a 90° rotation...
(By the way, the amazon item confusingly has "MPU6050" in its name, it should be MPU9250 since it is 9-axis).

henzim, 2017-09-10:
It seems my problems come from the insufficient performance of the Arduino Pro Mini 8MHz. 

I will now implement Magdwick "offline" on the PC to verify that my problems disappear :)

@kriswiner Thank you very much for the fruitful discussions and your quick replies. I will stay in touch with you regarding EM7180.

closed 2017-09-10 by henzim

ErikBob, 2019-09-02:
Anyone stumbling upon this might be interested to know that the MPU6050 does indeed have an intern fusion filter as well. Check out Jeff Rowbergs Github for working with it.

kriswiner, 2019-09-02:
Yes, the internal DMP was a remarkable fusion engine for its time, but it
can only manage 6 DoF fusion at 100 Hz maximum, requires loading of a
~5KByte binary blob, and still suffers from rather large heading drift. So
not really useful for heading estimation and not needed for roll and pitch.
Simple open-source Magdwick or Mahony filtering does as well. The DMP is
nice for step counting and pedometry, etc but the still obsolete MPU9250 is
even better.