kriswiner/MPU9250 #185: yet again: two MPU9250 on I2C

ClaireBol, 2017-09-18:
Hello,

I am trying to connect my MPU9250 to an Arduino Nano with I2C. Since I want to connect two sensors I want to use with the address 0x69. I managed to obtain the 0x69 address (when scanning the I2C devices, I see the 0x69 address), however when I try to read the sensor, I get unusable values. I am using the MPU9250BasicAHRS_I2C code, an example provided with the library. However I do not understand how do I have to change the code to read the sensor when using this address. When I upload the code, in the self test I obtain:

x-axis self test: acceleration trim within : -100.0% of factory value
y-axis self test: acceleration trim within : -100.0% of factory value
z-axis self test: acceleration trim within : -100.0% of factory value
x-axis self test: gyration trim within : -100.0% of factory value
y-axis self test: gyration trim within : -100.0% of factory value
z-axis self test: gyration trim within : -100.0% of factory value

So indeed I obtain completely wrong data... Any idea why?

Please help a poor student with her thesis!
Thanks a lot!

kriswiner, 2017-09-18:
Change the address in the sketch to 0x69?

ClaireBol, 2017-09-19:
I already tried that. I changed the address with:
#define MPU9250_ADDRESS 0x69
 and yet it is not working. Do I have to change something else? Either in the library or in some other parts?

Thanks a lot for your help!

kriswiner, 2017-09-19:
Replace this:

#define ADO 1
#if ADO
#define MPU9250_ADDRESS 0x69  // Device address when ADO = 1
#else
#define MPU9250_ADDRESS 0x68  // Device address when ADO = 0
#define AK8963_ADDRESS 0x0C   //  Address of magnetometer
#endif 

with this:
#define MPU9250_ADDRESS 0x69
#define AK8963_ADDRESS 0x0C

MitalPattani, 2018-03-31:
i tried doing that, but to no avail!

kriswiner, 2018-03-31:
Tried doing what?

MitalPattani, 2018-03-31:
I tried this in the library as well as in the sketch!
Replace this:

#define ADO 1
#if ADO
#define MPU9250_ADDRESS 0x69 // Device address when ADO = 1
#else
#define MPU9250_ADDRESS 0x68 // Device address when ADO = 0
#define AK8963_ADDRESS 0x0C // Address of magnetometer
#endif

with this:
#define MPU9250_ADDRESS 0x69
#define AK8963_ADDRESS 0x0C

MitalPattani, 2018-03-31:
It still detects the 9250 as 0×68, also i get some weird values for yaw pitch roll, i suppose that's happening because it can't detect the mag sensor properly but don't know what to do about the address!

kriswiner, 2018-03-31:
I thought I explained this.

You have to toggle the pass through mode enable for each MPU9250, one after
the other, then read from the mag, one after the other.

MitalPattani, 2018-04-01:
okay so after a bit of research i figured that this is the code to enable pass through mode 

 writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecond
pulse and any read to clear
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit
0) interrupt

but i need reading from both the sensors simultaneously, how do you suggest i toggle between the two without putting this in the main loop!

kriswiner, 2018-04-01:
" i need reading from both the sensors simultaneously, "

Cannot, so use an I2C multiplexer. You can only read one after another, not
both at the same time. Even with a multiplexer or using two different I2C
busses on one micrcontroller this is true. You will have to read one after
another no matter what, I think.

First try to use two MPU9250 on the same bus and see if this can work for
you. The latency (time for toggle) will be a few tens of microseconds.
Maybe this is good enough.

You will have to put this:

writeByte(MPU9250_ADDRESS1, INT_PIN_CFG, 0x12); // enabl;e
writeByte(MPU9250_ADDRESS2, INT_PIN_CFG, 0x10); // disable

in the main loop, of course, to enable one and disable the other, then
reverse, then repeat.

MitalPattani, 2018-04-01:
i am getting an error saying writeByte is not defined in void loop

kriswiner, 2018-04-01:
You have to create a function to parse this command. Like this one maybe?

// Wire.h read and write protocols
void writeByte(uint8_t address, uint8_t subAddress, uint8_t data)
{
Wire.beginTransmission(address); // Initialize the Tx buffer
Wire.write(subAddress); // Put slave register address in Tx buffer
Wire.write(data); // Put data in Tx buffer
Wire.endTransmission(); // Send the Tx buffer
}

MitalPattani, 2018-04-01:
for some reason, writeByte was yellow colored, hence the editor must read it as a defined function but still there was the error.....i guess I'll just create the function again.