kriswiner/MPU9250 #188: weird behavior of magnetometer

mahmoudelgendy1995, 2017-09-23:
hi kris,
i get yaw reading just 180 degree range from north to south counterclockwise and from north clockwise doesn't give any response ! 
also when i flip the sensor the yaw stop giving any response and advise ?

kriswiner, 2017-09-23:
Did you calibrate your sensors? How?

mahmoudelgendy1995, 2017-09-24:
i calibrate it like you mention in this article https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

mahmoudelgendy1995, 2017-09-25:
now the yaw reading is fine but it is affect by the other reading i.e when rotate the sensor in other direction the yaw changes why is that ?

kriswiner, 2017-09-25:
You are not feeding the Madgwick fusion algorithm data in the proper order,
or your sensors are not calibrated properly, or you have crappy sensors.