kriswiner/MPU9250 #189: weird behavior of roll

MahmoudNagah, 2017-09-25:
hi kris, 
in my sensor the roll reading is good but when i move in (pitch) direction it increases as well , this happens when the pitch reading reaches 75 , the roll is suddenly goes to 150 . what do you think is the problem ?

kriswiner, 2017-09-25:
You are not feeding the Madgwick fusion algorithm data in the proper order,
or your sensors are not calibrated properly, or you have crappy sensors.

MahmoudNagah, 2017-09-25:
Ok i read your article and made the  calibration as you said in this article and use the code your shared .
How could i know if is my calibration is right or not ?!

kriswiner, 2017-09-25:
Plot Mx vs My, Mx vs Mz, and My vs Mz from data recorded while moving the
sensor all around.

Are the plots circles with centers on the origin?

MahmoudNagah, 2017-09-26:
I made the calibration and this the the value of Max and the Min 
and use Matlab to draw it M,z,x,y i think this not good but i made the calibration like ypur articale used by your code so whais the problem
[image]

[image]

kriswiner, 2017-09-26:
The problem is these are not circles centered on the origin.

MahmoudNagah, 2017-09-26:
How could I solve this problem  and overcome this

kriswiner, 2017-09-26:
Calibrate the magnetometer properly?

MahmoudNagah, 2017-09-26:
I use your sketch so Could you except what i do wrong in my calibration so i could check ?