kriswiner/MPU9250 #190: Cannot get data from MPU9250

superviruz, 2017-10-02:
Hi Kris,

Thank you very much for your effort and sharing code. Before explain my issue, I may apologize for my English. If you cannot get it, please feel free to ask me again.

In the past, I connected MPU9250 + OLED + mini pro    into my custom board with some modified of your code. Normally there're yaw, pitch, row shown on OLED screen. Everything works good with calibration bias in each set (around 30 set has done before).

I bought MPU from Ebay and aliexpress from many seller for share any risk/loss. I found first problem is each seller have vary MPU address and not match at "if" condition. Then I skip "if (c == 0x73) // WHO_AM_I should always be 0x68" and it solved.

But today I get MPU from new seller(for me) and found new issue that I cannot get any value from MPU. It just show 0 for all axis. I checked by upload I2C scanner into mini pro and it shows proper address both of my OLED and MPU (0x68) on serial monitor. I tried upload modified code again and swap SCL and SDA but still show zero. Then swap back and then tried with my old lot of MPU, it is good. Then tried again with new lot of MPU (same mini pro, OLED), its bad (for whole 12 pcs).

Here is where I get [MPU](https://www.aliexpress.com/item/MPU-9250-GY-9250-9-axis-sensor-module-I2C-SPI-Communications-Thriaxis-gyroscope-triaxial-accelerometer-triaxial/1885410798.html?spm=a2g0s.9042311.0.0.kHjmo1). He sold for many orders (121 orders as shown in his page) but no one complain or found issue same as me. I'm not sure where is the problem. I feel stupid now that I didn't order from previous seller. Then I will really appreciate for your help. I don't want to trow this MPU lot in the trash. Thank you.

kriswiner, 2017-10-02:
Did you connect 3v3 to NCS?

superviruz, 2017-10-02:
No, I using mini pro 5 v. Just connect VCC GND SCL SDA.

kriswiner, 2017-10-02:
No I2S then.

kriswiner, 2017-10-02:
I mean I2C

superviruz, 2017-10-02:
I ever connect pullup(or pulldown cannot remember) 10K to SDA&SCL but no different result. I suppose there's internal pullup resistor already. Then I leave it no connection. I also tried connect gnd to NCS but no solved. I may try it again ASAP.

kriswiner, 2017-10-02:
3 V to NCS is what you need.

superviruz, 2017-10-02:
I seem to get hope. Tried to connect NCS with 10K to GND. It shows some strange value (not zero now).
But tried to connect 3.7V to NCS (or 3.7V >> 10K resistor >> 3.7V) still show zero. What I doing wrong? It need exactly 3V?

kriswiner, 2017-10-02:
3.3V to NCS, no pll up required.

Maybe the boards you bought are junk.

superviruz, 2017-10-02:
Tried connect 3V (2x AA batteries) to NCS-GND. But still show zero. Really sad to hear it's all junk. But really thank you for your help and very fast respond.

PS. one more question. MPU9250 yaw is absolute value but pitch and roll are relative value right? sometime when I lay it on flat table then power up . pitch and roll would begin with +/- 10 deg. Not always nearly 0. Can I make it always nearly 0?

kriswiner, 2017-10-02:
They are all absolute if you calibrate your sensors properly and feed the
fusion filter with the data in the correct order.

superviruz, 2017-10-03:
Hi,
Thank you. I will try as your advice. ^^

closed 2017-10-03 by superviruz

bkdeepak95, 2018-12-28:
Code works fine but sometimes MPU9250 (Accelerometer and Gyrometer) throws garbage values.

bkdeepak95, 2018-12-28:
[main.pdf](https://github.com/kriswiner/MPU9250/files/2715205/main.pdf)

kriswiner, 2018-12-28:
> I followed your code for MPU9250 but sometimes MPU9250 (Accelerometer and
> Gyrometer) throws garbage values. Let me more clarify it when my sensor is
> stationary, for first 30 seconds(time varies in every run) it
> (Accelerometer & Gyrometer) gives readings close to zero, at 31st second it
> gives garbage reading of 250,300... I have checked my code so many times
> but did not find any mistake. Sometimes code works fine with zero error in
> MPU9250 Readings. Kindly help me.

In what units?

bkdeepak95, 2018-12-29:
Accelerometer readings are in g and gyro readings are in deg/sec.

kriswiner, 2018-12-29:
If the data go from ~0 dps to 250 dps then either 1) you are converting the
bytes incorrectly or 2) your sensor is busted.

bkdeepak95, 2018-12-29:
I have gone through my code so many times but it looks error free. I have enclosed my code in above chat, could you please go through it.? I have checked this on number of sensor (MPU9250) i believe that trouble is in code only. Sometimes it works with zero error for entire day but sometimes it throws within a minute.
Please help me sir, this error has exhausted me. I am in need.

kriswiner, 2018-12-29:
"... could you please go through it.?"  No

bkdeepak95, 2018-12-29:
Ok fine. Could you give me more clarification in conversion of bytes..? you mean 8 bit sensor values into 16 bit ..? I will again give my best, kindly guide me.

kriswiner, 2018-12-29:
Look at the raw byte stream and identify if the bytes are changing when the
dps value is changing...

bkdeepak95, 2018-12-31:
My sensor is OK, but my reading is not proper. I am afraid for my reading.

void read_mpu9250(void)
{		
	HAL_StatusTypeDef sts=HAL_OK;
	uint8_t buf[6];
	buf[0] = ACCEL_XOUT_H;
	
	readBytes(INT_STATUS, 1, &MPU_Reg_Read[0]);
	if (*MPU_Reg_Read & 0x01)
	{
	HAL_I2C_Master_Transmit(&hi2c1,MPU9250_I2C_SLAVE_ADDRESS,buf,1,I2C_TIMEOUT);	
	sts =	HAL_I2C_Master_Receive(&hi2c1,MPU9250_I2C_SLAVE_ADDRESS,MPU_Reg_Read,14,I2C_TIMEOUT);
	}

kriswiner, 2018-12-31:
> Yes it is 14 byte array.

Not sure what to tell you. Try Arduino with your sensor and see if it works
there to find out if it is the sensor or the software. if the latter I
can't really help you. Maybe an ST forum will help.

bkdeepak95, 2019-01-03:
Is there any command to reset only MPU9250.? clearing of EEPROM will help..? Kindly give your suggestions.

kriswiner, 2019-01-03:
There is a soft reset for the mpu9250.