kriswiner/MPU9250 #191: Design Questions

afnan, 2017-10-03:
Hi

I am trying to make a small board of preferably USB size which has 9DOF that can tell me accurate Yaw,Pitch and Roll and should be able to communicate via wifi.

Current Setup is MPU9250 + NodeMCU(Esp8266). Issue is I get 922 Hz Frequency with Mahony and not much accurate results if pitch is rotated to 90 causes other angles to be senstive. I was going to give it another try so i thought I should ask you a few questions before I start new hardware

1. Mpu9250 works decent but is there any better one ? LSM9DSO? 
2. Would adding EM7180 Sensor Hub is good idea rather than going for Teensy3.1?
3. Is it good idea to integrate ESP8266, MPU9250 and EM7180 on single chip board to shrink size?

kriswiner, 2017-10-03:
First of all, which direction is South when you are standing on the North
pole?

No IMU can work properly at 90 degree pitch.

MPU9250 is better than the LSM9DS0.

Yes, EM7180 is a better solution than using a Teensy 3.1.

I have designs with MPU9250, EM7180 and ESP8266 on a 17 mm x 30 mm
0.5-mm-thick board we use for flight controllers. We need this size for the
motor control and boost circuitry as well as breaking out pins to PTH at
the edges. Works well. I could sell you some for $50 if you want a ready
made solution.

It could be made a bit smaller, I have another design without the motor
control circuitry that is 17 x 25 mm. Unlikely you will get the size down
much smaller.

Kris

afnan, 2017-10-03:
Sounds great. What would it cost me for board without motor control? And
this price includes shipping to Australia?

kriswiner, 2017-10-03:
Don't have any of those at the moment.

afnan, 2017-10-03:
> issuecomment-333888743

Oh, would you be back in production with this small board and the other
board you have what is the cost after shipping  to Australia.

Thanks

kriswiner, 2017-10-03:
> issuecomment-333979888

Motor control boards are $50 plus $14.50 shipping USD.

afnan, 2017-10-03:
> auth/AGY1qrd8CLHBo-
> issuecomment-333983048

Any picture possible? And mode of payment ?

Thanks

kriswiner, 2017-10-03:
> issuecomment-333983761

https://www.tindie.com/products/onehorse/esp8266-flight-controller/

paypal to tleracorp@gmail.com

afnan, 2017-10-03:
> issuecomment-333984356

Great, I'll work on it and one more question. What do you use to program
this board that has USB port?

kriswiner, 2017-10-03:
> issuecomment-333984697

No, FTDI connector as usual.

kriswiner, 2017-10-04:
Alternative would be to design something specifically for your use but this would be more expenive.

afnan, 2017-10-05:
My need would be fulfilled by ESP8266 + Sentral + MPU9250 + USB Port. I
wonder the board you mentioned without motor controller suits and I do not
want to solder headers and was thinking other way arounds to get FDTI
connected.
1. Also when do you think that board will be available again?
2. How much would you take for PCB Files if board is not available? i can
get it printed here.

kriswiner, 2017-10-05:
I have such a board designed; we use it for a flying robot controller. I
can;t give you the design, but i could assemble one or two for $100 each
plus shipping if you want to use them. They have the FTDI chip already on
the board so the ESP8266 can be programmed via the UDB.

kriswiner, 2017-10-05:
That's USB, of course.

afnan, 2017-10-15:
> issuecomment-334201261

Hi,
I am grateful to you for your help. is there any way I can get pcb designs
as they are mandatory requirements for my submission. plus can you share
link to that board which does not have motor controller and just esp8266 +
Sentral+mpu9250+usb?
https://www.tindie.com/products/onehorse/esp8266-flight-controller/ this
one has motor controller on it rite?

kriswiner, 2017-10-15:
> issuecomment-334332687

I cannot give you any designs not already made open source. There is this
one, which you may use as you like:

https://www.tindie.com/products/onehorse/esp8285-development-board/?pt=full_prod_search

afnan, 2017-10-15:
> development-board/?pt=full_prod_search
> issuecomment-334336823

I see. Well this does not have MPU etc on it. I think i will go with your
previous option that says 4M memory + ESP8266 + Senteral + MPU9250 + USB
built in and MPU. I do not want motor controllers rather prefer small size
over them. I think two boards. i will finalize coming week.

afnan, 2017-10-15:
Also as you mentioned if you go for custom solution what would you charge
for board plus pcb files?

kriswiner, 2017-10-15:
> issuecomment-336686020

Typical charge for well-specified custom work is $2K plus component
expenses. Please e-mail me at tleracorp@gmail.com if interested.