kriswiner/MPU9250 #192: MPU9250, Quaternion Update Question

lyravega, 2017-10-06:
Hello Kris, thank you so much for the 9250 stuff. I've been trying to build a DIY head-tracker, and I wouldn't even tried it without your code :+1:

I have a question though. I've seen a few example DIY stuff based on your code, and in all of those, including your MPU9250 only code, Madgwick is used as below:

`MadgwickQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f, my, mx, mz);`

However, passing the values like this causes heading (yaw) to be heavily affected by the attitude&bank (pitch&roll). While the device is on flat surface, heading is correct but tilt it slightly and it goes nuts.

So I've been reading on issues for all of your code (pretty much anything 9250 related that you've shared with the world, so to speak), and in one of them you've replied to a similar issue with this:

`MadgwickQuaternionUpdate(-ay, -ax, az, gy*PI/180.0f, gx*PI/180.0f, -gz*PI/180.0f, mx, my, mz);`

This one, while provides a stable heading when the "compass" is tilted, swaps bank with attitude. I can swap it again, but haven't been able to test it that much as I've tried another, from "MPU9250_MS5637_AHRS_t3.ino", which passes this:

`MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f, my, -mx, mz);`

Now this one is very good, seems to be the best one for my case, quaternion update works as expected and tilting the compass doesn't affect the heading that much; it's still affected (about 1-2 degrees), but I blame myself for it. 

So my question is, which one should I use? The one that gives me the accurate readings, of course... However I am very confused right now as different 9250s seem to have different parameters to pass. 

Oh almost forgot. The last 3 parameters, why are the passed ones are y-x-z and not x-y-z? What I'm trying to ask is, the function is `void MadgwickQuaternionUpdate(float ax, float ay, float az, float gx, float gy, float gz, float **mx**, float **my**, float **mz**)`, but we pass `..., **my**, -**mx**, **mz**);` is it because of some quaternion related stuff? Apologizes for stupid questions/inexperience/etc. Keep up the great work!

kriswiner, 2017-10-06:
It is really quite simple. The Madgwick filter requires the data in one of
the standard conventions for AHRS. I usually use NED (North, East, Down)
but ENU will also work. The proceudre is to choose which axis you want to
poit int he North direction. Let's say Ax (x-axis acceleration). Then
depending on the sensor axes orientation of your system, you can choose
which axes are pointed in the North direction, which the East, and which
the Down. For the MPU9250 with Ax pointing in the nNorth direction, then Gx
is also North, My is North, so the madwich filter should receive data as:

M(Ax,..,.., Gx,..,.., My,..,..);

East is -Ay, -Gy and -Mx. Down is -Az, -Gz, and Mz.

So the full function ought to be:

M(Ax, -Ay, -Az, Gx, -Gy, -Gz, My, - Mx, Mz);

I usually negate the A components since gravity == +1 is up by convention.

This process works for any sensor combination.

lyravega, 2017-10-06:
Thanks for the answer! I understand it now; it comes down to my choice, my setup's orientation. 

One more question, if you don't mind. Is there a way to get the Madgwick to converge faster without adjusting the beta gain? Anything short of getting a better MCU that is. Sudden, sharp movements cause an overshoot, which stabilizes over time, and on another topic you've said that 

> you need to be running the fusion filter at > 5 kHz

and iterate multiple times on the same data. But achieving that is impossible with a Pro Micro (which I'm using to learn, later on I'll move over to a Teensy 3.2). I was wondering if I can do anything more for Pro Micro so to speak :) I've been optimizing everything I can to squeeze out every bit of Hz that I can for Madgwick. 

Again, sorry for stupid questions.

kriswiner, 2017-10-06:
No alternative to getting a faster MCU. I use this one:

https://www.tindie.com/products/TleraCorp/butterfly-stm32l433-development-board/

lyravega, 2017-10-07:
I see. I'll try to write a program and use my PC to do the computing while using MCU and sensor for data collection only, I guess it'll work too. But in any case, this has been a nice learning experience so far. Thank you for the recommendation, next time I'll skip Teensy and see if I can find something like you recommended Thank you for your answers!

closed 2017-10-07 by lyravega