kriswiner/MPU9250 #194: roll and Pitch Limitation

Muhammad-Khodeer, 2017-10-15:
Hello kris,
thank you for this great project, I used your code to calibrate and test my MPU9250.
I'm using Sparkfun board IMU with Arduino Zero.
However I have a couple of questions

I succeeded to get Yaw, Pitch and Roll reading with ranges Yaw (0 to 360), Roll (-180 to 180) and pitch (-90 to 90), I am not sure yet if I got the best accuracy yet but my question is what if I need the range of Roll and Pitch (from 0 to 360)? how to overcome this limitation?
as I want my unity simulation box to rotate in all directions 360 degree

I read that using matrices (Euler Angles classic or Tait-Bryan ) we can't get 9 degrees of freedom  (0° to 360°) in all Roll, Pitch and Yaw (which what I need) And I have to use quaternions to overcome "Gimbal Lock" and get (0° to 360°) range at the three angels Roll, Pitch and Yaw. 
Is that true ? and have you any code that help me on this?
I am thinking about to change to BNO055 IMU as the simulation video show the object rotate in 360 degree all directions.

Also I want my sensor to be very sensitive so it changes output if change by one degree occurred for example if it is fixed in a servo motor I want to read the one degree of the motor change is that possible? did you measure the sensitivity before? how?

kriswiner, 2017-10-15:
For AHRS, pitch is -90 to +90 and roll is -180 to +180. You can always
remap to anything you want.

With sparkfun's MPU9250 I think you might be able to get +/- 5 degrees
accuracy. The BNO055 is worse.

Muhammad-Khodeer, 2017-10-18:
Thank you A lot Kris
I got good results.
I also have some questions

1- I want to take the best Magnetometer calibration values that worked with me and fix it in my code instead doing calibration every time. Is that right?
what is the area range that same calibration values keep working fine ?

2- What is the best way to measure the accuracy? something like servo motor ?

kriswiner, 2017-10-18:
You can store mag calibration values in your sketch or on EEPROM, etc and
avoid having to calibrate each time. Just make sure you check periodically
that the mag is still calibrated. And if you change environments where the
Earth's magnetic field or temperature vary a lot, you will likely have to
calibrate again.

For accuracy, yes, you just need to test the estimated orientation against
a known orientation. Just a rotating table with degree markings is good
enough to test heading, for example.

inderjeetkharay, 2019-08-22:
Hello Muhammad,

Are you able to get angles in the range of 0 to 360 degrees?

Thanks

kriswiner, 2019-08-22:
You cannot get roll and pitch to go from 0 to 360 since they are defined
from -90 to 90 and -180 to 180 for pitch and roll, respectively.

inderjeetkharay, 2019-08-22:
Dear Kris,

Thanks for your reply. 
Actually I ordered teensy 3.2 and Ladybug.
So at first I tried your code with teensy 3.2. I am getting pitch from -90 to 90 and roll -180 to 18.. That's fine. But at roll 90 or -90, pitch doesn't changes, it remain somehow constant. 
What problem should I have?
For filter I used (North, East, Down) notation. Also At fort worth declination is 3° 22' E  ± 0° 20', so I need to substract 3.36 from yaw. Is that correct?

Also I will try to do connections for Ladybug tomorrow to see the difference.

Thanks 
Inderjeet

inderjeetkharay, 2019-08-22:
Also, to add, my readings at still position at table are:
ax = -0.18 ay = -0.37 az = 1000.49 mg
gx = -0.01 gy = 0.11 gz = 0.08 deg/s
mx = 84 my = -89 mz = 380 mG

kriswiner, 2019-08-22:
That all seems correct. The trouble is that at certain positions the angles
are degenerate. it is the same problem as standing on the North pole and
asking, Which direction is South?

inderjeetkharay, 2019-08-26:
Dear Kris,

I understood the case you are talking about.
But the problem I am facing is:
When my sensor is at table then all three angles (yaw, pitch, roll) works fine.
Bur when I change roll angle, let us say I change roll angle by 20 degrees. Now I am trying to change pitch, but pitch is not changing. At the same position when I change yaw, it change the Pitch angle.
It means yaw becomes the pitch?
Sorry I am not able to understand the problem. Please let me know where can be the problem.
Maybe I need transformation from global to sensor body frame.

Thank You
Inderjeet

kriswiner, 2019-08-26:
Two likely possibilities: you are not feeding the sensor data in the
correct NED orientation or the Madgwick/Mahony filter or the sensors are
not properly calibrated.

kriswiner, 2019-08-27:
> Dear Kris,
>
> Thanks a lot for the comments. Yes, now I am getting good angles. Now I am
> extending your code for 6 MPU9250 sensors using multiplexer. I want to ask
> that the libraries will remain same? (Right now I am using your algorithm
> of Teensy)
> #include <i2c_t3.h>
> #include <SPI.h>
> #include <Adafruit_GFX.h>
> #include <Adafruit_PCD8544.h>
>
> or I need to add any other library for multiplexer?
>
> Thank You
> Inderjeet Singh

Not sure. If you are running multiple MPU9250s on a single I2C bus this
<https://github.com/kriswiner/MPU9250/tree/master/Dual_MPU9250> might be
useful.