kriswiner/MPU9250 #195: Magnetometer signal looking strange

monstah, 2017-10-24:
Hey Kris... I am just wondering if this signal from the magnetometer looks strange to you. 
I expected way more smooth lines.. is this due to the low frequency of the magnetometer or did I something wrong? I used one of your sketches as a basis 

[image]

kriswiner, 2017-10-24:
Not sure I understand this. At 100 Hz, the lines should be constant for 10
ms; instead they seem to vary every 0.15 ms or ~6000 Hz. How is this
possible?

monstah, 2017-10-24:
so basically this is wrong? the accelerometer and gyro look fine to me. Maybe I am asking too much, but could you post an image of your magnetometer output?

kriswiner, 2017-10-25:
Should be new data every 10 ms if using 100 Hz for the mag.

monstah, 2017-10-25:
not sure, if I am using 100 Hz for the mag. I just used one of your sketches and tweaked the I2C for the 
arduino to 400kHz. Maybe I can find out whats "wrong". Thanks for your reply!

kriswiner, 2017-10-25:
The AK8963C can only run at either 8 Hz or 100 Hz, so you should expect to
see new mag data every 125 ms at 8 Hz or 10 ms at 100 Hz, not 0.25 ms.
Maybe something wrong with the way you are reading from the mag? You can
poll the status ready bit and only read the mag data when this is 1.

monstah, 2017-10-26:
I will try to find it out, thanks for your fast replies as always!