kriswiner/MPU9250 #196: STM32F401RE - MPU9250 randomly stops sending data

Thalaivar, 2017-10-26:
The IMU behaves in a very strange way when I try to use it with STM32F401RE. Randomly it will stop reading values (and hence give out a constant reading), upon which resetting the board will not work and you have to plug and unplug the IMU. This is very unreliable because the IMU is mounted on a quadcopter and random failures cannot occur. Any ideas why this might be happening???

Thalaivar, 2017-10-26:
Unplug and plug*

Thalaivar, 2017-10-26:
Perhaps it's because I'm trying to read from magnetometer faster than 100Hz?

kriswiner, 2017-10-26:
You can try to read from the mag as fast as you want but you won't get any
data faster than 100 Hz.

Sounds like a buffer full problem or something. Try slowing down the I2C
bus to 100 kHz and the mag data rate to 8 Hz just to see if this changes
the behvior.

closed 2017-11-03 by Thalaivar

tygamvrelis, 2018-06-24:
I know this is late, but I'll leave this here in case others encounter this problem in the future: if you are using I2C it is possible you are encountering a well-known silicon bug with a well-known workaround. Basically, the I2C busy flag can get stuck due to a STM32 hardware issue; I remember this affected several F-series boards. This is something that I was pulling my hair out about for a week. The following links helped me recover from the issue during runtime.

https://electronics.stackexchange.com/questions/267972/i2c-busy-flag-strange-behaviour/281046#281046

https://community.st.com/thread/35884-cant-reset-i2c-in-stm32f407-to-release-i2c-lines

https://electronics.stackexchange.com/questions/272427/stm32-busy-flag-is-set-after-i2c-initialization
	 http://www.st.com/content/ccc/resource/technical/document/errata_sheet/f5/50/c9/46/56/db/4a/f6/CD00197763.pdf/files/CD00197763.pdf/jcr:content/translations/en.CD00197763.pdf

The following is the function I would call at runtime to help me recover. I would call generateClocks(1, 1) if a call to one of the HAL_I2C functions failed.

void generateClocks(uint8_t numClocks, uint8_t sendStopBits){
	/* This function big-bangs the I2C master clock
	 *
	 * https://electronics.stackexchange.com/questions/267972/i2c-busy-flag-strange-behaviour/281046#281046
	 * https://community.st.com/thread/35884-cant-reset-i2c-in-stm32f407-to-release-i2c-lines
	 * https://electronics.stackexchange.com/questions/272427/stm32-busy-flag-is-set-after-i2c-initialization
	 * http://www.st.com/content/ccc/resource/technical/document/errata_sheet/f5/50/c9/46/56/db/4a/f6/CD00197763.pdf/files/CD00197763.pdf/jcr:content/translations/en.CD00197763.pdf
	 *
	 *
	 * Arguments: numClocks, the number of times to cycle the I2C master clock
	 * 			  sendStopBits, 1 if stop bits are to be sent on SDA
	 *
	 * Returns: none
	 */

	static struct I2C_Module{
		I2C_HandleTypeDef*   instance;
		uint16_t            sdaPin;
		GPIO_TypeDef*       sdaPort;
		uint16_t            sclPin;
		GPIO_TypeDef*       sclPort;
	}i2cmodule = {&hi2c3, MPU_SDA_Pin, MPU_SDA_GPIO_Port, MPU_SCL_Pin, MPU_SCL_GPIO_Port};
	static struct I2C_Module* i2c = &i2cmodule;
	static uint8_t timeout = 1;

	GPIO_InitTypeDef GPIO_InitStructure;

	I2C_HandleTypeDef* handler = NULL;

	handler = i2c->instance;

	// 1. Clear PE bit.
	CLEAR_BIT(handler->Instance->CR1, I2C_CR1_PE);

	GPIO_InitStructure.Mode = GPIO_MODE_OUTPUT_OD;
	GPIO_InitStructure.Pull = GPIO_NOPULL;
	GPIO_InitStructure.Speed = GPIO_SPEED_FREQ_VERY_HIGH;

	GPIO_InitStructure.Pin = i2c->sclPin;
	HAL_GPIO_Init(i2c->sclPort, &GPIO_InitStructure);

	GPIO_InitStructure.Pin = i2c->sdaPin;
	HAL_GPIO_Init(i2c->sdaPort, &GPIO_InitStructure);

	for(uint8_t i = 0; i < numClocks; i++){
		// 3. Check SCL and SDA High level in GPIOx_IDR.
		if(sendStopBits){HAL_GPIO_WritePin(i2c->sdaPort, i2c->sdaPin, GPIO_PIN_SET);}
		HAL_GPIO_WritePin(i2c->sclPort, i2c->sclPin, GPIO_PIN_SET);

		wait_for_gpio_state_timeout(i2c->sclPort, i2c->sclPin, GPIO_PIN_SET, timeout);
		if(sendStopBits){wait_for_gpio_state_timeout(i2c->sdaPort, i2c->sdaPin, GPIO_PIN_SET, timeout);}

		// 4. Configure the SDA I/O as General Purpose Output Open-Drain, Low level (Write 0 to GPIOx_ODR).
		if(sendStopBits){
			HAL_GPIO_WritePin(i2c->sdaPort, i2c->sdaPin, GPIO_PIN_RESET);
			wait_for_gpio_state_timeout(i2c->sdaPort, i2c->sdaPin, GPIO_PIN_RESET, timeout); // 5. Check SDA Low level in GPIOx_IDR
		}

		// 6. Configure the SCL I/O as General Purpose Output Open-Drain, Low level (Write 0 to GPIOx_ODR).
[... 39 lines cut ...]
static uint8_t wait_for_gpio_state_timeout(GPIO_TypeDef *port, uint16_t pin, GPIO_PinState state, uint8_t timeout){
	/* Helper function for generateClocks.
	 *
	 * Arguments:    port, points to the SDA port
	 * 			  pin, the SDA pin number
	 * 			  state, the state to compare SDA to (i.e. desired SDA state)
	 * 			  timeout, the number of ms to wait before leaving function
	 *
	 * Returns: 1 if SDA pin is read to be state, 0 if timeout
	 */

    uint32_t Tickstart = HAL_GetTick();
    uint8_t ret = 1;
    /* Wait until flag is set */
    while((state != HAL_GPIO_ReadPin(port, pin)) && (1 == ret)){
        /* Check for the timeout */
		if ((timeout == 0U) || (HAL_GetTick() - Tickstart >= timeout)){
			ret = 0;
		}
        asm("nop");
    }
    return ret;
}

I did not come up with this code on my own; rather, I owe it to those people who took the time to discuss it on the links I posted before. Of course I still had to tweak it a bit to my specific microcontroller and application, as would anyone else using the workaround.

kriswiner, 2018-06-24:
> 267972/i2c-busy-flag-strange-behaviour/281046#281046
> in-stm32f407-to-release-i2c-lines
> 272427/stm32-busy-flag-is-set-after-i2c-initialization
> errata_sheet/f5/50/c9/46/56/db/4a/f6/CD00197763.pdf/files/
> CD00197763.pdf/jcr:content/translations/en.CD00197763.pdf

Maybe you need to switch to the L4 ;>