kriswiner/MPU9250 #198: ADC Values

Kimkimx, 2017-11-02:
Hi,

I'm trying to get just the adc values, but somehow the values for the accelerometer and the gyroscope are the same, and I have zeros for the magnetometer. Here is the code:

#include "MPU9250.h"

MPU9250 myIMU; 

void setup() 
{
  Serial.begin(9600);
 
}

void loop() {

   if (myIMU.readByte(MPU9250_ADDRESS, INT_STATUS) & 0x01)
  {  
    myIMU.readAccelData(myIMU.accelCount);  
   
    Serial.println(myIMU.accelCount[0]);
    Serial.println(myIMU.accelCount[1]);
    Serial.println(myIMU.accelCount[2]);
    delay(500);

    myIMU.readGyroData(myIMU.gyroCount);  
    
    Serial.println(myIMU.gyroCount[0]);
    Serial.println(myIMU.gyroCount[1]);
    Serial.println(myIMU.gyroCount[2]);
    delay(500);

    myIMU.readMagData(myIMU.magCount);  
  
    Serial.println(myIMU.magCount[0]);
    Serial.println(myIMU.magCount[1]);
    Serial.println(myIMU.magCount[2]);
    delay(500);
  
  }
}

kriswiner, 2017-11-02:
Maybe a little moe info, what MCU, pullups on I2C?, which code/library?

Can;t tell anything with above info...

Kimkimx, 2017-11-02:
It's arduino uno. Sorry, I'm kind of new to arduino. I'm not sure what's the pullups. I got the code from MPU9250BasicAHRS

kriswiner, 2017-11-02:
Can you see the MPU9250 on the I2C bus? Can you read the WHO_AM_I register?

Kimkimx, 2017-11-02:
It said 71

Kimkimx, 2017-11-02:
I think I figure it out now. Thank you for your help 👍

closed 2017-11-03 by Kimkimx