kriswiner/MPU9250 #202: where to buy MPU9250 with breakout?

scheng98, 2017-11-11:
Hi Kris,

I got one MPU9250 working fine. 

Later my co-worker bough 4 more MPU9250s (actually MPU9255, WHOAMI returns 0x73) from internet and it's a complete mess. All of 4 new ones looks like same, but three of them failed in i2c scan() routine you provided. I also verified the result on a scope and a logic analyzer. Clearly these MPU9250  The last one is able to do I2C communication but its performance is also bad compared with my original MPU9250. I tried all suggestions I can find on your posts and on internet with pull up reigster/power supply/int pin/reset routine. But none of them works. So I simply stopped wasting time there. 

Would you help to recommend a good vendor of MPU9250 with breakout that you tested before?

thanks again for your help and I own you as many beers as you want. :-)

cheers,

Steve

kriswiner, 2017-11-11:
I make two varieties, one for breadboard and one for mounting on the back
of Teensies and Dragonflies. You can find them both here:

https://www.tindie.com/products/onehorse/mpu9250-teensy-3x-add-on-shields/?pt=ac_prod_search

There is also an addon for my Ladybug STM32L4 board here:

https://www.tindie.com/products/TleraCorp/mpu9250-add-ons-for-ladybug/?pt=ac_prod_search

and lastly, I can make custom boards but this is more expensive, etc.

scheng98, 2017-11-11:
Well, that's a little bit different from we have here. We are looking for Arduino UNO to connect with MPU9250 board. Will either one of them work?

thanks,

Steve

kriswiner, 2017-11-11:
You will need to use 5V to 3V3 voltage translation but yes, Arduino UNO
will work. I recommend the Ladybug board as a much smaller, more convenient
3V3 MCU that works as well. But if all you have is Arduino UNO, you can at
least read the sensor data. Getting any kind of useful fusion of the sensor
data will be quite impossible with the Arduino UNO because of the AVR
architecture and low clock speed that will limit the fusion rate to ~100 Hz
or so, really to low.

scheng98, 2017-11-11:
Cool, we need raw sensor data for now. Just placed an order, have a good weekend,

scheng98, 2017-11-21:
Hi Kris, I got two Ladybugs. It connected to my Arduino board with just 4 pins, VCC/GND/SDA/SCL, and I2C communication works fine. I didn't use fusion and only used gyro,
void loop()
{  
    static float angle = 0;
   if (readByte(MPU9250_ADDRESS, INT_STATUS) & 0x01) { 
      getAres();
    
      // Now we'll calculate the accleration value into actual g's
      ax = (float)accelCount[0]*aRes; // - accelBias[0];  // get actual g value, this depends on scale being set
      ay = (float)accelCount[1]*aRes; // - accelBias[1];   
      az = (float)accelCount[2]*aRes; // - accelBias[2];  
     
      readGyroData(gyroCount);  // Read the x/y/z adc values
      getGres();
   
      // Calculate the gyro value into actual degrees per second
      gx = (float)gyroCount[0]*gRes;  // get actual gyro value, this depends on scale being set
      gy = (float)gyroCount[1]*gRes;  
      gz = (float)gyroCount[2]*gRes;   
  
      static unsigned long old_time = 0;
      unsigned long new_time;
      new_time = millis();
      angle += gz/1000.0 * (new_time - old_time);
      }
      ...
}

I placed it statically on a table. I saw gyro reading (angle) drifting much faster than the original MPU9250 board which we bought from a different vendor long time ago. As I clocked it, new ladybug MPU9250 gyro drifted 5.5 degrees per minute. 

I guess calibrateMPU9250(gyroBias, accelBias) will do auto-calibration on gyro/accel. What could be wrong?

thanks,

Steve
[image]

kriswiner, 2017-11-21:
You are confusing me a bit here. Ladybug is an STM32L4 development board.
Are you talking about the MPU9250 breakout board for Ladybug?

If the breakout is flat and motionless you should be seeing 0, 0, 0 for
gyro output. If not, you need to calibrate by rmoving this offset bias.

kriswiner, 2017-11-21:
> Yes, I'm talking about MPU9250 breakout board for Ladybug. I connected an
> Arduino UNO with Ladybug.
>
> How to calibrate gyro/accel? As right now I can skip mag. Ideally I shall
> calibrate all of them. Also a few dummy questions,
>
>    1. if I calibrate them here in San Jose for one MPU9250, do I need
>    recalibrate it when using it in Italy?
>    2. do I have to calibrate for each MPU9250? Or I can same calibration
>    data for all MPU9250 breakout boards from you? I believe the calibration
>    has to be done for each MPU9250, but like to confirm with you.
>
> thanks!
>
> Steve
>
> [image: image]
> <https://user-images.githubusercontent.com/15881047/33091589-64d76108-ceac-11e7-9b00-c238234cf7d1.png>

Generally best to calibrate for each environment, certainly for each
sensor. There is a calibration function here
<https://github.com/kriswiner/Ladybug/blob/master/MPU9250_MS5637_BasicAHRS2_Ladybug.ino>

scheng98, 2017-12-22:
I found out the issue and fixed them after being busy with other stuffs. The problem is gyro calibration during initialization can only hold 41 samples in FIFO which is 512 bytes. This is too short. So I make it 20 times. Then I normalized the gyro value and programmed it in GYRO OFFSET Registers. Now it worked much better.

I'd like to order 5 more MPU9250s but it's out of stock on Tindie. Could you help? Happy holiday,

Steve
[image]

kriswiner, 2017-12-22:
Which specific one do you want? I show 5 of the micro for Teensy 3.2 and 9
of the Mini for breadboard.

Is it the one for T3.6?

kriswiner, 2017-12-22:
OK, looks like you mean the MPU9250 breakout for Ladybug, is this right?

scheng98, 2017-12-22:
yes, MPU9250 Add-Ons for Ladybug: https://www.tindie.com/products/TleraCorp/mpu9250-add-ons-for-ladybug/?pt=ac_prod_search
I bought 2 already. Need 5 more at least.

kriswiner, 2017-12-22:
> yes, MPU9250 Add-Ons for Ladybug: https://www.tindie.com/
> products/TleraCorp/mpu9250-add-ons-for-ladybug/?pt=ac_prod_search
> I bought 2 already. Need 5 more at least.

I have two pcbs left and I just ordered 30 more. If you can wait a week or
so I will have them ready for you.

scheng98, 2017-12-22:
May I get 2 first? I can wait for the rest.

kriswiner, 2017-12-22:
OK, you can order two now.

scheng98, 2018-01-17:
Hi Kris, do you have more? I need 5 more. :-)

kriswiner, 2018-01-17:
Yes, forgot to update the quantity...

scheng98, 2018-02-26:
Hi Kris, could you help to prepare 20 more? cheers,

Steve
[image]

kriswiner, 2018-02-26:
Go ahead and order...

scheng98, 2018-02-27:
hmm... Couldn't place an order of 20. Got error that no sufficient quantity, even the webpage displays 25 are available...

kriswiner, 2018-02-27:
OK, try again please.

kriswiner, 2018-02-27:
Thanks for the order.

I am putting these together now and I am finding the MPU9250 chips I got
from aliexpress are really MPU9255, so they have the WHO_AM_I of 0x73,
otherwise they seem to work the same as MPU9250 and test out fine. I hope
this is OK.

Kris