kriswiner/MPU9250 #206: Could not connect to MPU9250: 0xFF

Sandaruwanims, 2017-11-14:
Hi,
I brought a MPU9250 - https://www.addicore.com/mpu-9250-p/ad280.htm  and use your library with arduino mega 2560. I upload https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino arduino program but it gave below output :
     MPU9250 I AM FF I should be 71
     Could not connect to MPU9250: 0xFF
I also check with I2C scan and it gave below lines :
     Scanning...
     I2C device found at address 0x68  !
     done

Arduino connection with mpu9250 like below:
mpu9250                    arduino mega2560
 VDD ---------------------- 3.3V
 SDA ----------------------- SDA
 SCL ----------------------- SCL
 GND ---------------------- GND

Could you help me to get this work?

kriswiner, 2017-11-14:
Pull NCS HIGH.

Sandaruwanims, 2017-12-13:
Hi kriswiner,
Thank you for your help. I pulled up NCS, SCL and SDA with 3.3v. Then It gave below output.
MPU9250 I AM 73 I should be 71
Could not connect to MPU9250: 0x73

After that I change   ->  if (c == 0x71) // WHO_AM_I should always be 0x68 
in to                          ->  if (c == 0x73) // WHO_AM_I should always be 0x68

Then it works but YPR data are not accurate(means it change YPR even not moving) also data rate is very low like below.

            " rate = 159.21 Hz
              Yaw, Pitch, Roll: 2.42, -0.01, 0.15
              rate = 159.50 Hz
              Yaw, Pitch, Roll: 1.67, -0.22, 0.66
              rate = 159.01 Hz
              Yaw, Pitch, Roll: 2.01, -0.19, 0.64
              rate = 159.46 Hz
              Yaw, Pitch, Roll: 2.67, -0.28, 0.13
              rate = 159.44 Hz
              Yaw, Pitch, Roll: 1.89, -0.16, 0.46
              rate = 158.94 Hz
              Yaw, Pitch, Roll: 2.26, 0.06, -0.30
              rate = 159.62 Hz
              Yaw, Pitch, Roll: 3.43, -0.03, 0.07
              rate = 159.12 Hz
              Yaw, Pitch, Roll: 2.70, 0.01, -0.23"

Could you please help me to solve this. Also I tried to use https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino for the sensor calibration. But I couldn't to compile it properly. It gave more errors like
 
error: 'I2C_MASTER' was not declared in this scope
Wire.begin(I2C_MASTER, 0x00, I2C_PINS_16_17, I2C_PULLUP_EXT, I2C_RATE_400);

How I solved this?
Thank you very much for your help.

kriswiner, 2017-12-13:
> MS5637_AHRS_t3.ino for the sensor calibration. But I couldn't to compile

What MCU are you using? I suspect you need one of these:

https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/

The yaw looks fine, I would expect you just need some calibration and a
faster MCU.

You will have to rewrite the Wire calls since that sketch uses i2c_t3.h
which is specific to the Teensy.

Sandaruwanims, 2017-12-14:
Thanks for your quick reply. 
I'm using Arduino nano ATmega328 board.

kriswiner, 2017-12-14:
16 MHz AVR will not be able to fuse the data fast enough for any kind of
accuracy.

Mustafakyuz, 2021-06-16:
> Hi kriswiner,
> Thank you for your help. I pulled up NCS, SCL and SDA with 3.3v. Then It gave below output.
> MPU9250 I AM 73 I should be 71
> Could not connect to MPU9250: 0x73
>
> After that I change -> if [...]

Hi Kris, thanks for your informations. I implemented Sandaruwanims's steps. But my Data is not changing when I move sensor. I dont know why not changing. Please look my data and help me. Thanks for everything.
Serial monitor:

AK8963 I AM FF I should be 48
AK8963 initialized for active data mode....
X-Axis sensitivity adjustment value 0.50
Y-Axis sensitivity adjustment value 0.50
Z-Axis sensitivity adjustment value 0.50
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 0.15 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 534.67 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 533.31 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 533.33 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 533.40 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -13.80, 0.00, 0.00
rate = 532.98 Hz

-------------------------------------------------------------------------------------------------------------------------------
My schematic of sensor and arduino:
https://www.google.com/imgres?imgurl=https%3A%2F%2Flucidar.me%2Fen%2Finertial-measurement-unit%2Ffiles%2Fwiring-mpu-9250-arduino-mega.png&imgrefurl=https%3A%2F%2Flucidar.me%2Fen%2Finertial-measurement-unit%2Fmpu-9250-and-arduino-9-axis-imu%2F&tbnid=XbDOOtWE_R26EM&vet=12ahUKEwiyv7na8prxAhWHnqQKHQp3BrUQMygAegUIARCrAQ..i&docid=Q56smhwHlIBmgM&w=1024&h=637&q=Schematic%20of%20mpu-9250%20and%20Arduino%20Mega&ved=2ahUKEwiyv7na8prxAhWHnqQKHQp3BrUQMygAegUIARCrAQ

kriswiner, 2021-06-16:
Looks like you are not connect with the device via I2C:

*AK8963 I AM FF I should be 48*

Mustafakyuz, 2021-06-16:
> Looks like you are not connect with the device via I2C: *AK8963 I AM FF I should be 48*
> […](#)

Thanks for your quick reply. 
What can I do for connecting with the device via I2C?

kriswiner, 2021-06-16:
Connect 3V3 and GND, then SDA and SCL. What devices do you see in an I2C
scan?

Mustafakyuz, 2021-06-16:
I connected 3V3 and GND, then SDA and SCL. I checked with I2C scan and it gave below lines :
Scanning...
I2C device found at address 0x68 !
done

Mustafakyuz, 2021-06-16:
> […](#)

I connected 3V3 and GND, then SDA and SCL. I checked with I2C scan and it gave below lines :
Scanning...
I2C device found at address 0x68 !
done

kriswiner, 2021-06-16:
Good. Now enable I2C pass through so you can see the AK8963C also.

Mustafakyuz, 2021-06-16:
> Good. Now enable I2C pass through so you can see the AK8963C also.
> […](#)

How can I enable I2C pass?

kriswiner, 2021-06-16:
This is in the init function:

 // Configure Interrupts and Bypass Enable
  // Set interrupt pin active high, push-pull, hold interrupt pin level
HIGH until interrupt cleared,
  // clear on read of INT_STATUS, and *enable I2C_BYPASS_EN* so additional
chips
  // can join the I2C bus and all can be controlled by the Arduino as master
//   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x22);
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecond
pulse and any read to clear
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit
0) interrupt

Check the data sheet.

kriswiner, 2021-06-16:
This should all be done properly if you are using any of my sketches...