kriswiner/MPU9250 #207: Should I trust the device? Low self test result

aripod, 2017-11-17:
Hello Kris,

First of all thank you very much for such big effort and in advanced for taking the time to answer this.

I have ported your code to a Zynq FPGA:

```
void I2C_Write(XIicPs *Iic, unsigned char Address, unsigned char Register, unsigned char Value)
{
	unsigned char u8TxData[2];
	u8TxData[0] = Register;
	u8TxData[1] = Value;
	XIicPs_MasterSendPolled(Iic, u8TxData, 2, Address);
}

void I2C_read(XIicPs *Iic, unsigned char Address, unsigned char Register, unsigned char Bytes, unsigned char *Data)
{
	unsigned char u8TxData[1];
	u8TxData[0] = Register;
	XIicPs_MasterSendPolled(Iic, u8TxData, 1, Address);
	while(XIicPs_BusIsBusy(Iic));
	usleep(50000);
	XIicPs_MasterRecvPolled(Iic, Data, Bytes, Address);
}

void MPU9250_SelfTest(XIicPs *Iic, float * result)
{ // Should return percent deviation from factory trim values, +/- 14 or less deviation is a pass
	int i;
	uint8_t rawData[6] = {0, 0, 0, 0, 0, 0};
	uint8_t selfTest[6];
	int32_t gAvg[3] = {0}, aAvg[3] = {0}, aSTAvg[3] = {0}, gSTAvg[3] = {0};
	float factoryTrim[6];
	uint8_t FS = 0;

	I2C_Write(Iic, MPU9250_ADDR, SMPLRT_DIV, 0x00);		// Set gyro sample rate to 1 kHz.
	I2C_Write(Iic, MPU9250_ADDR, CONFIG, 0x02);			// Set gyro sample rate to 1 kHz and DLPF to 92 Hz.
	I2C_Write(Iic, MPU9250_ADDR, GYRO_CONFIG, FS<<3);	// Set full scale range for the gyro to 250 dps.
	I2C_Write(Iic, MPU9250_ADDR, ACCEL_CONFIG2, 0x02);	// Set accelerometer rate to 1 kHz and bandwidth to 92 Hz.
	I2C_Write(Iic, MPU9250_ADDR, ACCEL_CONFIG, 0x00);	// Set full scale range for the accelerometer to 2 g.

	// Get average current values of Acclerometer and Gyroscope.
	for(i=0; i<200; i++)
	{
		I2C_read(Iic, MPU9250_ADDR, ACCEL_XOUT_H, 6, rawData);				// Read the six Acc raw data registers into data array.
		aAvg[0] += (int16_t)(((int16_t)rawData[0] << 8) | rawData[1]) ;  	// Turn the MSB and LSB into a signed 16-bit value
		aAvg[1] += (int16_t)(((int16_t)rawData[2] << 8) | rawData[3]) ;
		aAvg[2] += (int16_t)(((int16_t)rawData[4] << 8) | rawData[5]) ;

		I2C_read(Iic, MPU9250_ADDR, GYRO_XOUT_H, 6, rawData);				// Read the six Gyro raw data registers into data array.
		gAvg[0] += (int16_t)(((int16_t)rawData[0] << 8) | rawData[1]) ;  	// Turn the MSB and LSB into a signed 16-bit value
		gAvg[1] += (int16_t)(((int16_t)rawData[2] << 8) | rawData[3]) ;
		gAvg[2] += (int16_t)(((int16_t)rawData[4] << 8) | rawData[5]) ;
	}
	// Get average of 200 values and store as average current readings.
	for(i=0; i<3; i++)
	{
		aAvg[i] /= 200;
		gAvg[i] /= 200;
	}

	// Configure the Accelerometer and Gyroscope for self-test.
	I2C_Write(Iic, MPU9250_ADDR, ACCEL_CONFIG, 0xE0);		// Enable self test on all three axes and set accelerometer range to +/- 2 g.
	I2C_Write(Iic, MPU9250_ADDR, GYRO_CONFIG, 0xE0);		// Enable self test on all three axes and set gyro range to +/- 250 degrees/s.
	usleep(25000);											// Delay a while to let the device stabilize.
	// Get average current values of Accelerometer and Gyroscope.
	for(i=0; i<200; i++)
	{
		I2C_read(Iic, MPU9250_ADDR, ACCEL_XOUT_H, 6, rawData);				// Read the six Acc raw data registers into data array.
		aSTAvg[0] += (int16_t)(((int16_t)rawData[0] << 8) | rawData[1]) ;  	// Turn the MSB and LSB into a signed 16-bit value
		aSTAvg[1] += (int16_t)(((int16_t)rawData[2] << 8) | rawData[3]) ;
		aSTAvg[2] += (int16_t)(((int16_t)rawData[4] << 8) | rawData[5]) ;

		I2C_read(Iic, MPU9250_ADDR, GYRO_XOUT_H, 6, rawData);				// Read the six Gyro raw data registers into data array.
		gSTAvg[0] += (int16_t)(((int16_t)rawData[0] << 8) | rawData[1]) ;  	// Turn the MSB and LSB into a signed 16-bit value
		gSTAvg[1] += (int16_t)(((int16_t)rawData[2] << 8) | rawData[3]) ;
		gSTAvg[2] += (int16_t)(((int16_t)rawData[4] << 8) | rawData[5]) ;
	}
	// Get average of 200 values and store as average current readings.
	for(i=0; i<3; i++)
	{
		aSTAvg[i] /= 200;
		gSTAvg[i] /= 200;
	}

	// Configure the Accelerometer and Gyroscope for normal operation.
	I2C_Write(Iic, MPU9250_ADDR, ACCEL_CONFIG, 0x00);
	I2C_Write(Iic, MPU9250_ADDR, GYRO_CONFIG, 0x00);
	usleep(25000);								// Delay a while to let the device stabilize.

	// Retrieve accelerometer and gyroscope factory Self-Test Code from USR_Reg.
	I2C_read(Iic, MPU9250_ADDR, SELF_TEST_X_ACCEL, 1, &selfTest[0]);	// X-axis accel self-test results.
	I2C_read(Iic, MPU9250_ADDR, SELF_TEST_Y_ACCEL, 1, &selfTest[1]);	// Y-axis accel self-test results.
	I2C_read(Iic, MPU9250_ADDR, SELF_TEST_Z_ACCEL, 1, &selfTest[2]);	// Z-axis accel self-test results.
	I2C_read(Iic, MPU9250_ADDR, SELF_TEST_X_GYRO,  1, &selfTest[3]);	// X-axis gyro self-test results.
	I2C_read(Iic, MPU9250_ADDR, SELF_TEST_Y_GYRO,  1, &selfTest[4]);	// Y-axis gyro self-test results.
	I2C_read(Iic, MPU9250_ADDR, SELF_TEST_Z_GYRO,  1, &selfTest[5]);	// Z-axis gyro self-test results.

	// Retrieve factory self-test value from self-test code reads.
	factoryTrim[0] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[0] - 1.0) )); // FT[Xa] factory trim calculation
	factoryTrim[1] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[1] - 1.0) )); // FT[Ya] factory trim calculation
	factoryTrim[2] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[2] - 1.0) )); // FT[Za] factory trim calculation
	factoryTrim[3] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[3] - 1.0) )); // FT[Xg] factory trim calculation
	factoryTrim[4] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[4] - 1.0) )); // FT[Yg] factory trim calculation
	factoryTrim[5] = (float)(2620/1<<FS)*(pow( 1.01 , ((float)selfTest[5] - 1.0) )); // FT[Zg] factory trim calculation

	 // Report results as a ratio of (STR - FT)/FT; the change from Factory Trim of the Self-Test Response
	 // To get percent, must multiply by 100
	for (i = 0; i < 3; i++)
	{
		result[i]   = 100.0*((float)(aSTAvg[i] - aAvg[i]))/factoryTrim[i] - 100.;   // Report percent differences
		result[i+3] = 100.0*((float)(gSTAvg[i] - gAvg[i]))/factoryTrim[i+3] - 100.; // Report percent differences
	}
}
```

I tested the communication (like you) and I get the values that I should receive. The result of the self test is the following:

[MPU9250] Read value in reg 0x75 is 0x71. It should be 0x71.
selfTest: [Ax, Ay, Az] = [-79.-388%, -98.-27%, -58.-327%] - [Gx, Gy, Gz] = [-77.-833%, -60.-471%, -82.-575%]

Should I trust the device? Or this would be fixed after the calibration? I have double checked the code and it is like yours...I believe I am getting the proper values from the chip. What do you think?

Thank you very much for your help.

kriswiner, 2017-11-17:
Self test is just a diagnostic, uses only LSbyte see attached.

aripod, 2017-11-18:
Thank you Kris for your quick answer. I cannot see what you attached but it is clear what could be the issue for the low % with the LSB. I will proceed with the rest.

kriswiner, 2017-11-18:
send me an e-mail at tleracorp@gmail.com if you want the self test
application note.

aripod, 2017-11-27:
Hello Kris,

Rather than keep asking you via a private e-mail, I thought it might help someone else who is facing similar issues if I post here.

I adapted the code so it is like [https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino](https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino). I gathered some samples and plot them to see if data coming from the sensor is correct. The sensor is flat on a table, motionless.

# Accelerometer:
These are the plots according to all 3 different phases (on startup, after calibration and after initialization), taking 200 samples on each stage.

[image]
These are **raw** values, without subtracting accelBias. (only multiplied by the aRes) They seem to be correct as the only acceleration is gravity on the Z axis. If I rotate the sensor I can see gravity on the other axis. Therefore, _I could consider the accelerometer working properly, right?_

# Gyroscope
Same as with the accelerometer. It is divided in 3 stages with 200 samples each.
[image]
Here as the sensor is motionless, there seems to be something wrong on the Z axis. Could this already be because of the drift as the sensor has been on a long time already? I am also plotting the raw data (just multiplied by gRes).

# Magnetometer
I am using the calibration function that you use to with the hard and soft iron. The output values are: magBias = {0,0,0} and magScale ={0.65555, 0.6781609, +Infinity}. And these are the plots from the raw values:

Initialization:
[image]
Calibration:
[image]

The X/Y, X/Z, Y/X, Y/Z, Z/X and Z/Y should be similar to a circumference for what I've seen...but they do not look that good.....

I hope this helps a bit to figure out my issue.
Thank you again for all your time and help.

kriswiner, 2017-11-27:
> MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino. I gathered some samples

I am not sure what your issue is but the gyro data is seriously out of
whack. Are you sure you are scaling it properly? You should be seeing tens
of milli dps jitter not tens of dps.

What kind of MPU9250 breakout are you using?

aripod, 2017-11-29:
I am following your code so the gyro is set to 250 dps. Therefore, I am multiplying the raw data by (250.0/32768.0) all three axis...so it is strange that Gx and Gy **seems** fine and Gz has that variation....

I am using a cheap (Chinese I would say) [MPU9250 from ebay](https://www.ebay.com/itm/SPI-IIC-MPU-9250-9-Axis-Attitude-Gyro-Accelerator-Magnetometer-Sensor-Module-/281866691541). Could it be that it is a cheap sensor....? Same could go for the _nmagnetometer_?

Thanks!

kriswiner, 2017-11-29:
Yes, looks like a crappy design. Try the ones at Tindie.

aripod, 2017-11-29:
I am not sure I can get that as I am part of a university and it should be a company where I can have the product and then pay....plus shipping to Europe will cost as much as the sensor.... What about the one from [sparkfun](https://www.sparkfun.com/products/13762)? There is a distributor 1 hour from where I live....

kriswiner, 2017-12-05:
> I found a few bugs on the code and discovered that if I am going to make
> two consecutive reads on the sensor, I need to put a delay (~20us) so the
> second read does not give me trash. As far as the accelerometer and
> gyroscope I am getting what I should. When the sensor is flat and
> motionless on a table, I get 0g on Ax and Ay and 1g on Az. For Gy, Gy and
> Gz is 0 and when I rotate the sensor around each axis, the values change
> accordingly to the direction of my movement.
>
> As far as the magnetometer, I run the calibration as your function,
> followed the 8 pattern to collect data and I got these graphs. One is with
> the sensor flat and motionless on the table and the second one I am
> following the same 8 pattern.
>
> [image: magnetometer_flat] <https://ibb.co/gGUsvG>
> [image: magnetometer_8_pattern] <https://ibb.co/jrZQFG>
>
> For the second graph, it does not look like it would make a 3d sphere but
> at least they are centered around the origin. Can I consider my
> magnetometer calibrated or I still should not rely on these values?
>
> Thanks again for the help.

The mag does not look calibrated to me. Might need to increase the time for
collecting data...

aripod, 2017-12-06:
I am getting 1500 samples as you do in your code....I will try doubling.... Should I increase also the period of sampling? I assumed that you meant get more samples when you said increase the time for collecting data. 

One silly question....I am following an 8 pattern on all 3 axis....and I am also rotating the sensor while I do this....I mean, it is not like the sensor is orthogonal to the axis while following the 8 pattern.

aripod, 2017-12-06:
Second post today.... I believe I should not stay orthogonal to the axis but rotate the sensor randomly.....If I do that, I get something similar to your figures from [your wiki](https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration). They do not look like a circumference but is getting better than previous graphs....right?

[image]
[image]

kriswiner, 2017-12-06:
You need to sample the entire response surface, the entire 3D space as much
as possible.

You can test the calibration by noting the field in the z-axis whent he
sensor is flat and motionless on a table. It should read somewhere around
400 milliG depending on your area of the Earth. When you turn it 180
degrees so the sensor if facing bottom it should read -400 something. You
can test the other two sensor axes the same way. If the sensor response is
the same magnitude but + and - then the mag response surface is centered on
the origin. This doesn;t guarantee there is not elliptical distortion still
present, but at least you would have gotten rind of tany large offsets.

aripod, 2017-12-07:
That seems to be the hardest thing for me to achieve...cover the entire 3D space....If I change the distance from the sensor to the table when I am rotating it, the raw values change, as you can see on the top 3 images (raw values). I end up with one closer to a circle and two distinct ellipses...Therefore, in this case, I have around 400mG in Z and -400mG when I rotate 180 degrees.....but on the X and Y axis the values are greater than 400...closer to 800 I would say....

[image]<br /><a target='_blank' href='https://imgbb.com/'>free img</a><br />

I am now taking samples and ploting them in real time...should my gold then be to keep capturing samples until I have 3 circumferences on the top 3 figures?

kriswiner, 2017-12-07:
I hope you understand that you need to move the device in all directions.
Imagine an arrow pointing from  the + z-axis of the magnetometer. You have
to move the device during calibration such that the arrow tip touches every
point on the 3D spherical surface surrounding the centeroid of the sensor.
Like make sure the arrow touches every point on the interior of a balloon
with the sensor at the center of the balloon. Get it?

shomedas, 2019-03-03:
Even I am getting a jitter of upto 0.3 dps in the gyro readings when the gyro is stationary on the table.
@aripod : How did you add a delay between two consecutive reads of the sensor to avoid reading junk values

My readings for stationary IMU on table with x pointing towards north.

ax=49.0 ay=-6.71 az=1014 mg
gx=-0.11 gy=0.23 gz=-0.17 deg/s
mx=-7, my = 402, mz =114

Are these ok? Why is the my max when x points north? shouldnt it be mx that should be nearly 400 when the x axis points north

When I point any axis towards north and then towards south

My device passed the self test.

kriswiner, 2019-03-03:
Magnetic field varies a lot over the Earth. The only constant is that the
magnitude should be ~500 milliGauss. In my part of northern California the
field is mostly down so Mz reads ~420 milliGauss. x and y are ~50 and ~200
depending on orientation. Check with you local geologic survey office to
see what you should expect in your area.

How are you calibrating your sensors?

If you are removing the offset bias your gyro readings should be 0, 0, 0
and accel 0, 0 1, with maybe 10 millig jitter in the accel when the sensor
is flat and motionless on a table top.

shomedas, 2019-03-03:
I have used your code MPU9250BasicAHRS.ino for calibration and for reading the data. I have done the magnetic calibration also using your code. the gyro calibration is with the mpu flat & stationary on  a table with z axis pointing up

According to you the acceleration and the gyroscope gitter should be with tens of mg and milli dps resp.
I am getting the jitter in 100s of milli dps in the gyro reading.

aripod  says that adding a delay between two consecutive reads of the sensor helps to avoid reading junk values. Kindly suggest how do I add a delay in the "calibrateMPU9250" function.

kriswiner, 2019-03-03:
Why aren't you using the data ready interrupt? Then no delay is needed.

shomedas, 2019-03-03:
the INT_ENABLE interrupt is enabled that is set to 0x01 the initMPU function. Is there any physical connection I have to do? There is a INT pin in the board. Do I have to connect it to Ground or VCC?

My current connection
Vcc- 3.3V,
Gnd-0V :)
SDA-A4
SCL-A5
FSYNC- GND

kriswiner, 2019-03-03:
Connect INT to the MCU interrupt input pin...See Arduino.cc for how
interrupts work.

shomedas, 2019-03-04:
what about AD0 & NCS. I read in some of your other posts that they too need to be connected.

I am using arduino uno and MPU9255

kriswiner, 2019-03-04:
Ncs needs to be pulled HIGH, ado high or low as you like.