kriswiner/MPU9250 #208: dmp setup

aster94, 2017-11-17:
hello kriswiner,

I tried your code and wow... it works very well! I could see that you are having busy days so i hope you could help me
I am trying to use the mpu9250 with spi and i reached good results. right now i am reading just raw data.
i would like to ask you if you could point to me which lines of your code set up the DMP so i could finally start to use this imu

thanks is advance

kriswiner, 2017-11-17:
Don't use the DMP...

aster94, 2017-11-18:
Oh i see, so you feed the quaternion filters with raw data? Or do you do any manipulation before?

aster94, 2017-11-18:
ok finally i managed to have the SPI interface working :D 
https://github.com/aster94/MPU9250-SPI-quaternionFilters