kriswiner/MPU9250 #209: Pitch behaviour funny

NickMortimer, 2017-11-18:
Hi Kris

Sorry to bother you again. I'm having some funny stuff happening. I have my system calibrated and outputting yaw pitch roll. The funny thing is that pitch and yaw work perfectly but roll is funny. starts at 0 then goes up to 90 and then reduces to 0 and then to -90 and then again back  0 as I turn it. so I get ambiguity eg. I have 45 points on the circle that have a value of 45 degrees and they are 90 apart. Its giving me a real headache!

Is it  a faulty chip?

Thanks

Nick

kriswiner, 2017-11-18:
More likely you are not feeding the Madgwik filter correctly.

Which MPU9250 breakout are you using?

NickMortimer, 2017-11-18:
I've tried swapping and sign changing on most of the inputs!

https://www.sparkfun.com/products/13762

NickMortimer, 2017-11-21:
Ok I have finally worked this out! My bad pitch only has a valid range +/-90 degrees and that in my application of 360 rotation in all directions is not good thus pitch should be calculated using:

 pitch = atan2f(2*q[1]*q[0] - 2*q[1]*q[3], 1 - 2*q[2]*q[2] - 2*q[3]*q[3]);

Nick