kriswiner/MPU9250 #210: MPU9250 extreme strange behaviours

fschramka, 2017-11-24:
Hi Kris

I build my own uP board with bluetooth (rn42), mpu (9250), a atmega32u4 @ 8MHz and a battery charging chip. My goal is to have a sensor for rotations for a research project. Here a small movie of what we have:

https://www.dropbox.com/s/dnt5pyxb6nzgmva/Bike%20Simulator.mp4?dl=0

The sensors we have so far are built from a Adafruit 32u4 Proto with a MPU6050 and a EZ-link for bluetooth. 
For the steering we still use a HTC Vive controller and therefore the whole HTC Lighthouse system is needed. The new sensor should replace it to overcome this issue. 

Now my problem. When I use your lib for the MPU9250 I dont get any nice rotations. I receive data and if I rotate the board I can see that something happens (I wrote a small Unity visualisation). But it's far away from a nice rotation and it has a huge drift. This happens if I dont move the sensor:

https://www.dropbox.com/s/1p0nfurofmsrun5/Unity%202017-11-24%2018-50-51-84.avi?dl=0

This when I rotate the sensor around its Y axis:

https://www.dropbox.com/s/9g1sxipmhgil5zu/Unity%202017-11-24%2018-53-20-16.avi?dl=0

I dont need any EulerAngles / roll-pitch-yaw angles, so I don't go for the last conversion in your code (from if (!AHRS) on). I just update the Time, and go for the MadgwickQuaternionUpdate. Then I send it over bluetooth to the computer (x,z,y,w). xzy because of the directX coordinate system. 

I also use your function  magcalMPU9250(MPU9250magBias) to get the magBias which I can later use to correct the error. I rly dont know what i'm doing wrong...

Hope you can help me :)

P.s. If you have time to check the code 👍 
[MPU9250Board.txt](https://github.com/kriswiner/MPU9250/files/1502320/MPU9250Board.txt)

kriswiner, 2017-11-24:
> 11-24%2018-50-51-84.avi?dl=0
> 11-24%2018-53-20-16.avi?dl=0

Maybe many things, but you are not feeding the Madgwick filter the sensor
data in the correct order. It expects NED order.

fschramka, 2017-11-27:
Well okay, according to page 38 of the datasheet the Acc & Gyro are ENU, the Magnetometer is, like you said, NED. 

So I call the filter (a.x, a.y, a.z, g.x, g.y, g.z, M.Y, M.X, -M.Z).

Or do you mean the filter itself needs NED for all inputs? So it would be:

filter(a.y, a.x, -a.z, g.y, g.x, -g.z, m.x, m.y, m.z) ?

I guess I still fail somewhere, both ways still show strange results...

https://www.invensense.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf

kriswiner, 2017-11-27:
> MPU-9250A-01-v1.1.pdf

Either of these should work, so maybe the sensors aren;t fully calibrated?
Maybe there is an error in the hardware design?

fschramka, 2017-11-28:
After long searching I found a solution for my problem, but I had to use another library:

https://github.com/jrowberg/i2cdevlib

I used the code of the MPU9150, the HW registers are the same, hence it worked. The chip is in FIFO mode and I get directly a Quaternion from the MPU9250. It is extremely stable, I guess there is some filter algorithm implemented in hardware.

closed 2017-11-28 by fschramka