kriswiner/MPU9250 #211: MPU9250 I AM 0x0 I should be 0x71. Could not connect to MPU9250

antpgomes, 2017-11-28:
Hi Everyone

I read through all the posts and for the life of me I can't figure this out.

I have a sparkfun mpu9250, i see it as i2c address 0x68 on the scanner but i can't seem to get the example code to work.

I changed the following line:
byte c = myIMU.readByte(0x68, WHO_AM_I_MPU9250); to reflect the address but still no luck.

Any help would be appreciated

lukecaj, 2018-03-16:
I'm having the same sort of issue on my Arduino Zero, except I get a value of 0x1.  I see 0x68 returned in the I2C scan list, but not 0x71.

kriswiner, 2018-03-16:
0x68 is the MPU9250 I2C addresYou have to read the WHO_AM_I register of the
MPU9250 to get 0x71.

yNosGR, 2018-05-10:
I am having the same problem with an amazon MPU9250 board on an 8266.
@kriswiner, the line @antpgomes references above:
byte c = myIMU.readByte(0x68, WHO_AM_I_MPU9250);
reads i2c address 0x68 and the WHO_AM_I_MPU9250 register (0x75) into byte c.

kriswiner, 2018-05-10:
Are there pullups on the board? Are they working? If you scan for I2C
devices what do you see?

yNosGR, 2018-05-10:
When I scan for devices, I see the bmp180 and the MPU9250; I can poll the bmp180 w/o issue - maybe the 9250 wants to be the only one on the bus?
-------8<--snip-->8-------
Scanning...
I2C device found at address 0x68  !
I2C device found at address 0x77  !
done
-------8<--snip-->8-------

kriswiner, 2018-05-10:
No, multiple devices can be on the bus, that is the whole point of I2C.

And when you read the WHO_AM_I register, what do you get?

yNosGR, 2018-05-10:
It replies with a 0x0, like OP.

kriswiner, 2018-05-10:
Maybe a crappy sensor, or bad I2C API

yNosGR, 2018-05-11:
Since I am using the arduino IDE, I'll blame it on the sensor.

kriswiner, 2018-05-11:
Can you try another sensor then. it could be something wrong with your
Wire.h calls.

yNosGR, 2018-05-16:
I just got a sparkfun board in, wired it up with the same results. The puzzling part is that I am able to communicate with both the bmp180 and bmp280 sensors have here w/o issue. I am not sure what is different about the mpu9250.

kriswiner, 2018-05-16:
Do you have pullups enabled on the Sparkfun board?