kriswiner/MPU9250 #212: MPU9250 Self Test failed

ssarathkumar916, 2017-12-02:
Hello,

We are using MPU9250 in our board for motion processing. Recently we bought some MPU9250 sensors  for the latest versions of our board. In that we are reading some wrong values from accelerometer and gyrometer. so we ran self test. From the self test report, recently fabricated boards (with latest sensors)  (2v0 and 3v0) got failed in self test. I have Attached the selftest report below for reference.

please help us to find the root cause of this issue. And please let us know what calibration technique can be used to overcome this issue.

Thanks,
Sarathkumar S

[pedometer_self_test_report_(1).XLS - Sheet1.pdf](https://github.com/kriswiner/MPU9250/files/1523977/pedometer_self_test_report_.1.XLS.-.Sheet1.pdf)

kriswiner, 2017-12-02:
Looks like you are not doing the self test correctly. The self tests uses
only the LS Byte for comparison.E-mail me at tleracorp@gmail.com for the
self test AN.

ssarathkumar916, 2017-12-06:
Hi Kris,

According to selftest report i have attached , only 2V0 and 3V0 were failing in self test. How did 1v0 board alone get pass in self test ,if i did the self test wrong ?. 
Please clarify.

Thanks,
Sarathkumar S.
[pedometer_self_test_report_.1.XLS.-.Sheet1.pdf](https://github.com/kriswiner/MPU9250/files/1534825/pedometer_self_test_report_.1.XLS.-.Sheet1.pdf)

kriswiner, 2017-12-06:
Did you implement the self test as described in the application note?
Invensense self test only uses the least significant byte in the test.