kriswiner/MPU9250 #213: AK8963 WHO I AM is 0xFF instead of 0x48, but can read data without problem

abhayghatpande, 2017-12-08:
Hi Kris,

This is a strange one, I'm not able to figure out. Using a 9250 and AK8963 I can access the Fuse ROM, the default calibration data is 1.5 for all axes. I can also read the Magnetometer values from the other registers, which appear to be correct ... 

But the Who I Am register always returns 0xFF instead of 0x48. Is this normal? Also the INFO register returns 0xFF.

How can I verify if the magnetometer data being returned is correct? Thanks.

kriswiner, 2017-12-08:
Usually by reading the WHO_AM_I register. The fuse ROM data should be
closer to 1, like 1.01 or 1.05, etc, not 1.5 and not all 1.5.

Are you sue you set the MPU9250 in bypass mode?

Can you read any other AK8963C WHO_AM_I?

abhayghatpande, 2017-12-09:
AFAIK, the initMPU function sets the bypass mode using:
```
writeByte(INT_PIN_CFG, 0x22);
writeByte(INT_ENABLE, 0x01);
```
Is that correct?

The I2C Scan function does not find a device at address 0x0C. Only a single device at 0x68 (the 9250).

kriswiner, 2017-12-09:
That should do it, maybe a poorly-designed breakout board?

abhayghatpande, 2017-12-09:
Not sure how to verify, the device we're using is this: http://freematics.com/products/freematics-one-plus/

So if no device is found at 0x0C (error code is 2), then what are all those values being read? Are they completely random?
```
ACC: 0.01, 0.05, 1.07,  -- GYR: 1.52, -0.96, 3.72,  -- MAG: -444.51, -162.73, 164.38,
ACC: -0.00, -0.01, 1.00,  -- GYR: -0.53, 0.14, 0.40,  -- MAG: -636.43, 19.44, -110.76,
ACC: -0.00, -0.01, 1.00,  -- GYR: -0.30, 0.10, 0.33,  -- MAG: -636.43, 19.44, -110.76,
ACC: -0.00, -0.00, 1.00,  -- GYR: -0.28, 0.05, 0.28,  -- MAG: -636.43, 44.18, -119.75,
```

kriswiner, 2017-12-09:
First of all, I am not sure why you are asking me and not the freematics
people...

If you know the local magnetic field you can easily check to see if you are
getting reasonable values. For example, most places on this planet have a
total magnetic field of ~0.5 Gauss; is this what you find?

Also, place the device flat and motionless on a table top. What is the
z-axis mag field? Nor turn it upside down and repeat; is the new z value
the same as the old value times -1? It should be close.

If you rotate the sensor about the compas points do you get reasonable
values?

Maybe you have an error in your I2C read and write functions? Or maybe you
have the wrong register addresses, I don't know...

What code are you using?

If you can''t verify the WHO_AM_I register value I wouldn't trust the
values you "read" from the sensor....

kriswiner, 2017-12-09:
Looks like the freematics people have simply copied one of my sketches but
they don;t know what they are doing since they have kept the hard-wired mag
biases I measured for one of my sensors three years ago!

From
https://github.com/stanleyhuangyc/Freematics/blob/master/libraries/FreematicsONE/FreematicsMEMS.cpp
:

if (mag) {
float magbias[3];
magbias[0] = +470.; // User environmental x-axis correction in milliGauss,
should be automatically calculated
magbias[1] = +120.; // User environmental x-axis correction in milliGauss
magbias[2] = +125.; // User environmental x-axis correction in milliGauss

// Calculate the magnetometer values in milliGauss
// Include factory calibration per data sheet and user environmental
corrections
mag[0] = (float)magCount[0]*mRes*magCalibration[0] - magbias[0]; // get
actual magnetometer value, this depends on scale being set
mag[1] = (float)magCount[1]*mRes*magCalibration[1] - magbias[1];
mag[2] = (float)magCount[2]*mRes*magCalibration[2] - magbias[2];
}
and, of course, no attribution.

abhayghatpande, 2017-12-09:
Hi Kris,
Everyone flocks to you because you are helpful and respond to questions right away. :)
Not to mention the fact that 90% of the planet's 9250 code derives from your repositories!
I've asked the Freematics folks too. Waiting for a response.
I agree that the values are completely unreliable considering that a device is not even found.

kriswiner, 2017-12-09:
You really must be able to read the WHO_AM_I register. Can you try a simple
program to set the MPU9250 in bypass mode and repeat the I2C scan, you must
see 0x0C on the I2C bus.

abhayghatpande, 2017-12-09:
OK. Will do and report.

abhayghatpande, 2017-12-09:
Nope ... same result. The magnetometer is not detected on the I2C bus. After 1 AM here now ... will continue tomorrow. Hopefully will receive a response from Freematics by then.

abhayghatpande, 2017-12-09:
Mystery solved: "For current version of ONE+, AK9863 is not configured to be accessed directly by I2C. Instead, it can only be accessed through DMP. This will be changed in next batch of hardware."

Thank you for your help, Kris. I'm sure I'll be back again with more questions after trying the DMP.

kriswiner, 2017-12-09:
Odd, since the MPU9250 DMP can only do 6 DoF fusion, and the DMP cannot use
the magnetometer data.

abhayghatpande, 2017-12-09:
Thanks for confirming my suspicions.
I kind of figured that when I looked at the code. So looks like there is no mag data for us at the moment then ...

closed 2017-12-09 by abhayghatpande

ypatidisioannis, 2018-01-15:
Hi Kriswiner,

I am having an issue with the MPU9250BasicAHRS_I2C example. After downloading to a Genuino Zero , the IDE has stopped recognizing the ATMEL . 

Do you have any recommendation on how to solve this?

Regards
Ioannis

ypatidisioannis, 2018-01-15:
ohh sorry , just found it, my mistake.

ypatidisioannis, 2018-01-15:
Hi again, also my MPU address is 0x68 , is this correct?

kriswiner, 2018-01-15:
MPU9250 address can be 0x68 if ADO is LOW or 0x69 if ADO is HIGH.

mynameismyname5, 2018-01-17:
connect the pins scl and sda to (scl_pin, sda_pin) on the arduino not to (A4 , A5), hope that help :)

kriswiner, 2018-07-01:
> hie @kriswiner <https://github.com/kriswiner>...... I am using your
> library for MPU9250 and programming it using Arduino nano. I am getting the
> same error "AK8963 I AM FF I should be 48" and then I get 13 for yaw and 0
> for both pitch and roll. In your code the bypass is already there, so now I
> don't know what to do. Your help will be deeply appreciated.

Are you able to get the correct who_am_i reading from the MPU9250?

Sanjanawinchester, 2018-07-02:
No, I am getting 0xFF.

kriswiner, 2018-07-02:
Your I2C connections are wrong or your MCU I2C pins are wrong, etc.

Sanjanawinchester, 2018-07-02:
I connected as follows:
IMU         Arduino
SCL--------A5
SDA-------A4
VCC-------5V
GND------GND

kriswiner, 2018-07-02:
The MPU9250 is not 5 V tolerant. Do you have pullups on the I2C bus?

Sanjanawinchester, 2018-07-03:
Nope, I don't have any pullups. Should I connect 10K resistances?

kriswiner, 2018-07-03:
What MPU9250 breakout board are you using?

camerart1, 2019-01-13:
Hi,
Can I join in and ask about the AK8963 on it's own, not behind a MPU9250?
Camerart.
[image]

kriswiner, 2019-01-13:
Yes, I have used this IC before, what is it you want to know?

camerart1, 2019-04-12:
> Yes, I have used this IC before, what is it you want to know?
> […](#)

Hi K,
I've only just seen your reply, is it still live?

I use the attached board with the AK8963C.  I write in BASIC, but do you have any code at all please?

C.

kriswiner, 2019-04-12:
Code for what, the AK8963C by itself?

camerart1, 2019-04-12:
> 
> Code for what, the AK8963C by itself?
> […](#)

Hi Kris,
Yes, for AK8963C by itself.

In my BASIC code, I didn't use SELF TEST MODE--------------SINGLE MEASUREMENT MODE----------HOFL---------- OR DRDY, but as I'm getting very varied results, perhaps I should be.
C.

kriswiner, 2019-04-12:
> ask about the AK8963 on it's own, not behind a MPU9250? Camerart. [image:
> ak8963c break out board]

I have a sketch that uses the AK8963N. Send me an e-mail at
tleracorp@gmail.com and i will send it to you.

quichedood, 2019-11-27:
Hi,

Just encountered the same error and would like to share my personnal experience.
As an hobbist, I'm ordering some components on Al***press. Is it possible to get fake MPU9250 (MPU-6050 for example) ?
I resoldered an other one which I get from official and know reseller and everything is now working smoothly !
/my2cts

Amazing job Kris by the way, your lib is now running on a SAMD51G19A @ ~6200Hz

kriswiner, 2019-11-27:
Is it possible to get fake MPU9250 (MPU-6050 for example) ?

Oh yes......I have many!

camerart1, 2019-11-28:
> I have a sketch that uses the AK8963N. Send me an e-mail at tleracorp@gmail.com and i will send it to you.
> […](#)

Hi K,
Thanks for the CODE.
I am using 18F4620 with the AK8963C and am getting intermittent results.

After months of forums and tests, then adding a level shifter to the MISO, which was intermittently triggering the PIC.  I recalled a message that was by-passed, that the 18F4620 will run as low as 4.2V (Not low enough)  I've now ordered 18LF4620 PIC, which will run at the required 3.3V.  Let's hope this solves the problem, I'm sure it will.
Thanks.
C