kriswiner/MPU9250 #214: Altitute / lifting affects Yaw in AHRS not in IMU

ben5en, 2017-12-13:
Hello!

great thanks to Kris for that fantastic source and all your work!

My question deals more with the quaternion filter:
During the implementation of the mpu9250AHRS I noticed a strange behavior:
Lifting the board (I am using an evaluation-board with a mpu9250-evaluation board on top) horizontal with no rotation at all affects the yaw value clearly. This only happens when using the AHRS function, while using only the IMU function this effect is not visible.

`MahonyAHRS( ax, -ay, -az, \
                        gx, -gy, -gz, \
                        my, -mx, mz);

  MahonyIMU(ax,  -ay, -az, \
                      gx, -gy,  -gz);`

How is this phenomenon explained? All sensors are calibrated.
-> The IMU function is related to Paul Stoffregens source

kriswiner, 2017-12-13:
No idea, in general I would expect the 9 dof solution to be better, more
stanle. Sounds like cross axis effect. Are you sure the sensor data is in
the correct order?

ben5en, 2017-12-13:
I tried a lot of variants. I started with your from the MPU9250_M5637_HARS_t3 sketch. I also tripple checked my code. I can not find the mistake and will now build up another board with the MPU9250. Maybe the chip really has a bug?!

ben5en, 2017-12-13:
SOLVED: After a **lot** of code testing it seems like something was disturbing the magnetic field. Using a long USB-Kabel and testing it far away from the computer / monitor works!

Have you ever having trouble with magnetic distortion?
Thanks!

kriswiner, 2017-12-13:
Yes