kriswiner/MPU9250 #218: Poor calibration, overshoot and settling time

JonAdams66, 2017-12-18:
Hello Kris -

Many thanks for your excellent code for the Teensy and MPU9250.
I am trying to use this as the basis of an antenna rotator controller, but am running into a few problems with compass accuracy, linearity and overshoot/settling time.

I have used the basic 'figure of 8' calibration routine (storing the [magbias] and [magscale] values as statics.  Have also set the correct Declination for my location.

I am getting an incorrect North heading (by about 15 deg), and South is not 180 deg opposite (i.e. no a linear change in angle when rotated).  There is quite a bit of overshoot/slow settling time when turning the sensor (I can see that the filter has a settling time - am running Mmode at 100Hz and have tried adjusting GyroMeasError up and down with some effect on overshoot/settling time).
 I suspect that the calibration is part (or most of) the problem.

Is there a better methods of calibration (I see you have produced plots - how do you get these data points and plots - from the serial port??)

Any help much appreciated!

All the best

Jon

kriswiner, 2017-12-18:
Calibration could be the problem. I usually plot Mx vs My, Mx vs. Mz and My
vs Mz to check the response surface. Of course, you need to remove the
offset biases fro the accel and gyro as well.

The other thing to do is to make usre to run the fusion filter at fast
enough rate, typically 1 kHz or higher.

Lastly, make sure you are feeding the data into the Madgwick or Mahony
filter properly. I usually Use a NED convention.

JonAdams66, 2017-12-18:
Hi Kris -

Many thanks for your speedy suggestions.

I am running your code virtually 'as is' from the site on a Teensy 3.2 - does this assume a 1 kHz fusion rate (writeByte(MPU9250_ADDRESS, CONFIG, 0x03); ) , and the data in NED convention (it appears to be?)

It's Beta I have been tweaking.

Do I plot Mx, My and Mz from the serial monitor?

kriswiner, 2017-12-18:
Beta does nothing.

Yes, plot using the serial monitor.

The fusion rate is determined by how often you call the filter per new data
sample, which is determined by the processor speed, use of interrupt vs.
polling, etc. It is partly user configurable but not by changing a register
value.

This is usually how I manage the Madgwick filter for the MPU9250:

    MadgwickQuaternionUpdate(-ay, -ax, az, gy*pi/180.0f, gx*pi/180.0f,
-gz*pi/180.0f,  mx,  my, mz);

It will depend on which direction you choose for North.