kriswiner/MPU9250 #220: Strange AHRS results

nitrosalat, 2017-12-30:
Hi, i'm sorry if this question is asked before.

When mpu9250 module accelerates along x-axis and not rotates around y-axes pitch value shouldn't change. But in my case pitch value changes. What am I doing wrong?

kriswiner, 2017-12-30:
Did you calibrate all thre sensors?

nitrosalat, 2017-12-30:
Yes, i did

kriswiner, 2017-12-30:
Are you passing sensor data to the Madgwick filter in the proper order?

nitrosalat, 2017-12-30:
Yes, i tried Madgwick and Mahony filters and i'm passing sensor data in proper order. The acceleration still affects the Euler angles.

kriswiner, 2017-12-30:
What is the order?

If it correct, and I assume your sensors are properly calbrated, only other
possiblity apart from a faulty sensor breakout would be man-made magnetic
fields.

nitrosalat, 2017-12-30:
ax, ay, az, gx, gy, gz, mx, my, mz, deltat

kriswiner, 2017-12-30:
No, needs to be NED.

nitrosalat, 2017-12-30:
In static position(without accelerations) pitch, roll angles are correct but if i accelerate the module along one of axes WITHOUT rotation pitch and roll are changing. I think it is not a problem with sensor data pass order to filter.

kriswiner, 2017-12-30:
" I think it is not a problem with sensor data pass order to filter."

You are wrong.

Choose which accel axis you want to point North. Figure out which gyro and
mag axes point in the same direction. Then figgure out which oint East, and
which point Down. If you look at the data sheet you should be able to see
immediately why this is necessary; i..e.,Az = -Mz in terms of direction.

Feed the Madgwick filter as so:

(AN, AE, AD, GN, GE, GD, MN, ME, MD)

It cannot be as you had it above.

nitrosalat, 2017-12-30:
Ok, i'm feeding the madgwick filter in order ax, ay, az, gx, gy, gz, my, mx, -mz but pitch and roll values are still changing while the module accelerates along x-axis or along y-axis. 
Btw, when i use madgwick filter, pitch and roll less react on acceleration than mahony filter.

kriswiner, 2017-12-30:
What kind of MCU are you using? What is the fusion rate?

nitrosalat, 2017-12-30:
ATmega328P(arduino nano)

kriswiner, 2017-12-30:
Yeah, 8 or 16 MHz AVR is not going to give you very good results. You might
get 100 Hz fusion rates, you really need 1 kHz or better.

See:
https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion

Teensy 3.2 or, better, this
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
would be more appropriate.

nitrosalat, 2017-12-30:
Thanks, Kris :)

kriswiner, 2017-12-30:
Sorry there is no quick fix.

Accurate absolute orientation is hard and there are no short cuts,
unfortunately ;<