kriswiner/MPU9250 #221: Observing magnetometer calibration

mrrobles09, 2018-01-02:
I am currently puzzled right now with this outputs!  I am conducting two scenario where one is being calibrated by a figure of 8 pattern and the other is just letting it still like upward flat in the surface and as the result, the sensor that is set up in "upward flat in surface" is much similar to the checking of the magnetometer bias compared to the figure 8 pattern that is normally applied for the magnetometer. I've followed your https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration which helps me to know that the scattered plot of x and y for magnetometer should be forming a spherical plots around the 0 point of the x and y. Is this readings for "upward flat in surface" okay to use for the magBias or the figure 8 pattern? I've checked my surroundings also if there are nearby magnetic object that may affect my calibration process and have found nothing.

                                       **mX (upward in flat surface)**
[image]

                                       **mY (upward in flat surface)**
[image]

                                        **mZ (upward in flat surface)**
[image]

                                         **mX (figure 8 pattern)**
[image]

                                         **mY (figure 8 pattern)**
[image]

                                         **mZ (figure 8 pattern)**
[image]

P.S The following pictures are based on the output magnetometer of the sensor MPU9250 using the code in MPU9250_MS5637_AHRS_t3.ino script to also get the calibrated offsets of accel and gyro. Also, the images here are from the format arrange of the following:
 mx = (mx,my), my = (my, mz), and mz = (mx,mz).

kriswiner, 2018-01-03:
> Effective-Magnetometer-Calibration which helps me to know that the

OK, I am assuming you are plotting MX vs MY, etc.

The idea here is to move the magnetometer around to sample all of three
dimensional space while accumulating calibrating data. keeping it lying on
a flat surface is not useful. Figure eight simply means move it so the +z
magnetometer axis touches every pint on the inside of an imaginary sphere
many times.

When calibrated you should be able to check that when facing up the mag
field is the same magnitude as when the sensor is facing down, i.e., Mup =
-Mdown. The plots will give you information on the remaining elliptical
skew.

mrrobles09, 2018-01-12:
Thank you sir for your insight about this thread. I will get into it in fixing this and I'll get back to you with the results ^^

29676, 2020-07-21:
hi
I do not know what Figure 8 means and how to calibrate the compass
Please help me;

kriswiner, 2020-07-21:
Move the sensor all around sampling every possible position in 3 dimensions.

29676, 2020-07-21:
This means rotating the sensor around each axis
(x, y, z), in the form of 8 ?

Is the accelerometer and gyroscope also calibrated in this way?

kriswiner, 2020-07-21:
Accel/gyro are calibrated with sensor flat and motionless.

Mag with slow motion in all possible directions.

29676, 2020-07-21:
What are the uses of the buttons on the stm 32 board for this job? Are these used for calibration?
Or when calibrating the magnet, only the calibration function should be called and the sensor moved، correct?