kriswiner/MPU9250 #222: Magnetometer Sensitivity Adjustment Values (ASAX,ASAY,ASAZ)

juanibarracantavella, 2018-01-04:
Hello Kris,

I'm programming on arduino nano to get data from MPU 9250. The goal is to obtain attitude angles(roll, pitch, yaw) in a sailboat.

I came with some doubts:
1-The ASAX,ASAY,ASAZ are read only registers so I guess they come calibrated from factory, right?
2-The measurement data registers(0x03-0x08) are already affected by the ASA or they are plain data from the ADC? If they are not affected I guess it's my decission to use the factory values of ASA or use other ones calculated by me.
Thanks a lot in advance! Bye

kriswiner, 2018-01-04:
Hi Juan,

The ASA are factory calibrations of the axial sensitivity; they scale the
measured data. They are not included automatically in the data, you have to
apply them (by multiplication) manually. You can ignore them but I usually
(and recommend you) read them and then apply them for both the measurements
and calibration. The calibration of the magnetometer needs to be done in
any case.

"If they are not affected I guess it's my decission to use the factory
values of ASA or use other ones calculated by me."

 I recommend you use both for best results.

juanibarracantavella, 2018-01-06:
Hello Kris, thanks for your fast response. 

I'm having a little trouble getting a right yaw value when roll or pitch aren't close to zero. I've calibrated the magnetometer exposing every face of the die to all orientations. After that I've determined the biases by forcing the mean value to 0 (bias=(max-min)/2). 

After this calibrating I get a stable yaw angle as long as I keep the MPU board flat(parallel to ground). The problem begins when I start to tilt the board(pitch or roll). For roll/pitch angles of 20° the yaw increases or decreases by more than 90°. 

What I've observed from the magnetic field readings is that Z component is much higher that X and Y when the board is flat. As if the earth magnetic field is poining mainly down. Does that make sense?? I would think of it more like a vector parallel to the ground with a small vertical component. I live in Spain (39° N of latitude).

I attach two plots. Both have the same bias values. One is moving the mpu in every orientation to check calibration and the second one consists of various rotations of the mpu board being flat(parallel to ground). You can see in this one a vertical component of around 250 as you rotate it and horizontal components always below 180.

My questions are: is it normal to have such a big vertical component in the magnetic field? is it normal to get wrong yaw values when roll or pitch are bigger than 5°?

Hope you can help me a little bit. It would be great to have stable yaw angles and under 3-4° of error for roll/pitch angles up to 30°. 

Bye
[image]

[image]

juanibarracantavella, 2018-01-06:
I did some reseach and looks like the actual earth magnetic field where I live is more vertical than I thought: about 54° pointing down as you look to the north. This is the table I got from https://www.ngdc.noaa.gov/geomag-web/#igrfwmm :
[image]

kriswiner, 2018-01-06:
> geomag-web/#igrfwmm :

"My questions are: is it normal to have such a big vertical component in
the magnetic field? "

Yes, indeed.

Once the mag is calibrated, next are you feeding the Madgwick filter the
sensor data in the correct order?

Should be North, East, Down.

juanibarracantavella, 2018-01-07:
Hello,

I realized I'm using a using your sketch it was modified afterwards:

> _/* MPU9250 Basic Example Code
>  by: Kris Winer
>  date: April 1, 2014
>  license: Beerware - Use this code however you'd like. If you
>  find it useful you can buy me a beer some time.
>  Modified by Brent Wilkins July 19, 2016_

I'm checking your repository, can I use any of your sketches on arduino nano?

The filter I'm feeding now is the Mahony:
  MahonyQuaternionUpdate(myIMU.ax, myIMU.ay, myIMU.az, myIMU.gx*DEG_TO_RAD,
                         myIMU.gy*DEG_TO_RAD, myIMU.gz*DEG_TO_RAD, myIMU.my,
                         myIMU.mx, myIMU.mz, myIMU.deltat);

Should I use the Madwick??

I realized now that acceletation is not well calibrated. I guess it should measure around 1 in the z component and around 0 in the x and y components when the board is flat. I've tried to calibrate it the same way as with magnetic field but the problem is that accelerations are different depending on the orientation of the MPU when you turn it on. Is that normal??

kriswiner, 2018-01-07:
accel and gyro calibrations need to be done with the sensor flat and
motionless.

The accel and gyro +z axes don;t point in the same direction as the mag
z-axis, do they? You need to change your data input to the filter.

juanibarracantavella, 2018-01-07:
Good news. Fliping mag z axis was the solution to get stable yaw angles as I roll/pitch! Thank you very much for the solution!

I'm assigning accel bias at start by forcing accel to be (0,0,1). The same with gyro data to be (0,0,0).

The problem is that I would like to calibrate accel and gyros just once and store them in arduino EEPROM, and not everytime I turn the unit on. The idea is to calibrate it once and then being able to turn the unit on even if it is tilted. The unit goes in a sailboat and it's kind of hard to get repeatability in roll/pitch angles if calibration is done everytime, being the sailboat at slightly different angles (not perfectly flat) when turning the unit on.

I read in MPU9250 datasheet that there are registers for accel and gyros offsets(bias), which are applied to data before served by the unit. In your library I see that these offset registers are written at the calibrateMPU9250 function in the setup.

Is it possible to get repeatability by calibrating accel and gyro offsets just once and not everytime I turn the unit on? I'll try this and let you know but it would be interesting for me to know your opinion/experience.

juanibarracantavella, 2018-01-12:
I'm able to store calibrating values in arduino eeprom by now and seems to work fine when I reset the MPU.

Accel an magnetometer biased are calculated in the arduino while gyro bias are introduced in the mpu9250 appropriate register so it gives calibrated values. I did this following your sketch. 

I wasn't able to make the mpu9250 to make use of the accel bias registers. I wonder why, but I can solve this by loading the arduino with this task.

kriswiner, 2018-01-12:
This is the right approach. I usually do use the gyro offset bias registers
but the accel offset bias registers have the LSB reserved as a temperature
correction bit and I just don;t understand how to use this properly. Some
of the times I can get good results and sometimes not, so I just correct
the offset manuall or, like you, store to memory for subsequent use. There
are no offset bias registers for the magnetometer.