kriswiner/MPU9250 #223: Questions about MPU9250_MS5637_AHRS_t3.ino script...

mrrobles09, 2018-01-12:
Good day sir Kriswiner!

As I was observing this script and your other related MPU9250 codes I've notice the following codes here where my questions are as follows:
 - MPU9250_MS5637_AHRS_t3.ino script after calibrating the accel and gyro offsets you subtracted the offsets of the accel and magnetometer but not the gyroscope, why is that?

- In getting the gravity vector for the linear acceleration I've noticed some similarities between this script and in MPU6050_DMP6 of Jeff Roberg which is also similar to what Fabio Verasano is discussing about removing the gravity from the raw acceleration of the sensor to get the pure acceleration (RIP fabio verasano) link : http://www.varesano.net/blog/fabio/simple-gravity-compensation-9-dom-imus#comment-23390. 

 While observing these three scripts, it's gotten to my attention that yours are quite different compared to them specifically in getting the linear acceleration or the pure acceleration of the sensor. The following image are as follows:                                               

> From MPU6050_DMP
[image]

[image]
                                     

> From fabio's blog
[image]

                                                                         

>  Your script 
[image]

As you can see inside the yellow rectangles that your approach is quite different from them. Checking both of your outputs for linear acceleration, (MPU6050_DMP vs. MPU9250_MS5637_AHRS_t3) yours are coming from "-X " and "+Y" whereas the other is "+X" and "-Y" prior to the direction of my MPU9250 which is in flat steady surface with no motion or other unnecessary movement involve. With this all in mind what are your basis in coming up with gravity compensation to get the pure acceleration? 

- Your "a31" and "a32" equation in getting the gravity compensation of X and Y is different from the other two, why is that?

- When the other two subtracts the gravity compensated from the raw accel ( X, Y, and, Z), why are your
X and Y are adding the gravity compensated? Also,

- Is it because you all use different sensors that this approach also changes depending of what sensor's orientation is?

Regards, 
mrrobles09

P.S SO sorry for the long post sir Kriswiner its just that this questions make me want to dive more on this technology and maybe get some great success out of it . ^_^

mrrobles09, 2018-01-19:
How about the difference between your gravity compensation approach ? Can you show me how did you come up with your approach there?

kriswiner, 2018-01-19:
What ate you referring to?

kriswiner, 2018-01-19:
I didn't understand what you are asking.