kriswiner/MPU9250 #224: Yaw depends on pitch angle

kephaII, 2018-01-12:
Thanks for a nice library Kris!
I have tried two  IMU9250 breakouts    [https://www.ebay.com/itm/401062812000] hooked up to an Arduino Mega via an logic level shifter. 
Each sensor work OK with your calibration and realtime program ( MPU9250_MS5637_AHRS_t3.ino), except for that the yaw is dependent on the pitch, also after calibration has been applied.
However, the influence on the yaw changes a lot when i rotate the sensor 180 degrees, in the yaw plane.

I call the fusion filter this way, then the X axis on the breakout points north:
`  MadgwickQuaternionUpdate(-ax, ay, az, gx * PI / 180.0f, -gy * PI / 180.0f, -gz * PI / 180.0f, my, -mx, mz);`

[image]

[image]

What am i doing wrong?

best regards
Hans Otto Johansen

kephaII, 2018-01-12:
A bit more info:
MPU9250 I AM 71 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : 1.4% of factory value
y-axis self test: acceleration trim within : 0.1% of factory value
z-axis self test: acceleration trim within : 2.7% of factory value
x-axis self test: gyration trim within : -4.8% of factory value
y-axis self test: gyration trim within : -1.4% of factory value
z-axis self test: gyration trim within : -0.2% of factory value
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
X-Axis sensitivity adjustment value 1.21
Y-Axis sensitivity adjustment value 1.21
Z-Axis sensitivity adjustment value 1.16

closed 2018-01-12 by kephaII

kriswiner, 2018-01-12:
As far as I can tell you have done everything correctly. I assume you have
corrected for accel and gyro offset as well.

Only things left are possible magnetic interference (unlikely but possible,
would affect yaw more generally) or that your fusion rate is too low so
there is latency (symptoms not consistent with latency).

This problem ony seems to ccur sometimes, for example it doesn't show up at
the 310 mark.

Maybe there is jitter in your fixture?

kephaII, 2018-02-01:
Thanks Kris!

Sorry for late feedback.
It took some time to get hold of a 3.3 volt unit for test.

When i run the same code on my new  ESP32 devkit C, then there is no error.

On the Mega there was some instability with the INT pin. some times the Mega froze.
When i disconnected INT, it worked stable. but with the same error.
Maybe the logic level shifter added jitter, i tried with and without 4.7 k pullups.  No diffrence.

Anyhow, my IMU is 9 DOF, not 10. so i should not have used a program which reads the pressure sensor.
May be that also complicated?

The ESP32 were that cool, so i continue with those from now on!

Thanks and best regards

Hans