kriswiner/MPU9250 #225: Local or Global Frame

Mendes11, 2018-01-14:
As the Title says, is the ypr output given from a local or global reference frame?

I mean... for example is pitch always the same direction or is it fixed to a plane wings?
And finally if it is global, how can I make it locally fixed to the MPU?

Thanks! :)

kriswiner, 2018-01-14:
absolute orientation is global wrt True North, not sure how to turn it into
a local measure...

Mendes11, 2018-01-14:
for absolute you mean when using magnetometers? If so a MPU6050 would make that local reference possible? Or even ignoring the mpu9250 mag?

kriswiner, 2018-01-14:
nope...

kriswiner, 2018-01-14:
Perhaps you can take ratios of magnetic field ATan(Mx/Mz), for example,
this might give you local pitch...