kriswiner/MPU9250 #227: no use of zeta parameter (Madgwick)

neroner, 2018-01-14:
First of all great work, thanks!

I would like to notice that you define the zeta parameter in the main program but there is no use of zeta parameter in your Madgwick update filter. 

In a Magwick's paper entitled 'An efficient orientation filter for inertial and inertial/magnetic sensor arrays,' there is the version with the zeta parameter. You use another version which is available from x-io.co (again from Madgwick). 

I have implemented both of versions and I take different results. Can you inform me why you used this version and what is the difference?

Thank you in advanced!

kriswiner, 2018-01-14:
Didn;t notice any effect of using zeta. The filter is computationally
efficient and produces good results as it is implemented here.

neroner, 2018-01-14:
True. I take very good results. I just informed you that there is no use of zeta parameter... 

Thank you for your prompt reply

kriswiner, 2018-01-14:
Thanks for the info...