kriswiner/MPU9250 #228: I am 0xFF. I should be 71

rachit2jain, 2018-01-15:
I don't have Vddi on my breakout board so my wiring is something like this:

VCC --> 3.3 V
GND --> GND
SCL --> A5
SDA --> A4
ADO --> GND

After reading through a lot of issues already raised, I did an I2CScan() and the output was:
Scanning...
I2C device found at address 0x68!

But it shows 
MPU9250 I AM FF I should be 71.

These are the 4 basic lines of my code.
```
#define    MPU9250_ADDRESS            0x68
#define    WHO_AM_I_MPU9250           0x75
byte c = readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250);
Serial.print("MPU9250 "); Serial.print("I AM "); Serial.print(c, HEX); Serial.print(" I should be "); Serial.println(0x71, HEX);
```

Any idea on why is it happening? Thanks a ton in advance!

kriswiner, 2018-01-15:
No idea, what kind f MCU and Arduino Wire library are you using?

rachit2jain, 2018-01-15:
Hi Kris, 

Thank you so much for your reply. I am using Arduino Uno R3 and using Wire.h

```
#include <Wire.h>
#define    GYRO_FULL_SCALE_250_DPS    0x00
#define    GYRO_FULL_SCALE_500_DPS    0x08
#define    GYRO_FULL_SCALE_1000_DPS   0x10
#define    GYRO_FULL_SCALE_2000_DPS   0x18
#define    MPU9250_ADDRESS            0x68
#define    WHO_AM_I_MPU9250           0x75

void setup() {
  Wire.begin();
  Serial.begin(9600);
  I2Cscan();
  delay(1000);
  I2CwriteByte(MPU9250_ADDRESS,27,GYRO_FULL_SCALE_2000_DPS);
  byte c = readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250);
  Serial.print("MPU9250 "); Serial.print("I AM "); Serial.print(c, HEX); Serial.print(" I should be "); Serial.println(0x71, HEX);
  if (c == 0x71) 
  {  
    Serial.println("MPU9250 is online...");
  }
  else{
    Serial.println(c);
  }
  
}

void loop() {
  uint8_t Buf[14];
  I2Cread(MPU9250_ADDRESS,0x3B,14,Buf);
  int16_t gx=-(Buf[8]<<8 | Buf[9]);
  int16_t gy=-(Buf[10]<<8 | Buf[11]);
  int16_t gz=Buf[12]<<8 | Buf[13];

  Serial.print (gx,DEC); 
  Serial.print ("\t");
  Serial.print (gy,DEC);
  Serial.print ("\t");
  Serial.print (gz,DEC);  
  Serial.print ("\t");
  delay(100);
  Serial.println("");

}

void I2Cread(uint8_t Address, uint8_t Register, uint8_t Nbytes, uint8_t* Data)
{
  // Set register address
  Wire.beginTransmission(Address);
  Wire.write(Register);
  Wire.endTransmission();
  
  // Read Nbytes
  Wire.requestFrom(Address, Nbytes); 
  uint8_t index=0;
  while (Wire.available())
    Data[index++]=Wire.read();
}
void I2CwriteByte(uint8_t Address, uint8_t Register, uint8_t Data)
{
  // Set register address
  Wire.beginTransmission(Address);
  Wire.write(Register);
  Wire.write(Data);
  Wire.endTransmission();
}
void I2Cscan()
{
// scan for i2c devices
  byte error, address;
  int nDevices;

  Serial.println("Scanning...");

  nDevices = 0;
  for(address = 1; address < 127; address++ )
  {
    // The i2c_scanner uses the return value of
    // the Write.endTransmisstion to see if
    // a device did acknowledge to the address.
    Wire.beginTransmission(address);
    error = Wire.endTransmission();

    if (error == 0)
    {
      Serial.print("I2C device found at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.print(address,HEX);
      Serial.println("  !");

      nDevices++;
    }
    else if (error==4)
    {
      Serial.print("Unknow error at address 0x");
      if (address<16)
        Serial.print("0");
      Serial.println(address,HEX);
    }
  }
  if (nDevices == 0)
    Serial.println("No I2C devices found\n");
  else
    Serial.println("done\n");

}
uint8_t readByte(uint8_t address, uint8_t subAddress)
{
  uint8_t data; // `data` will store the register data   
  Wire.beginTransmission(address);         // Initialize the Tx buffer
  Wire.write(subAddress);                  // Put slave register address in Tx buffer
  Wire.endTransmission(false);             // Send the Tx buffer, but send a restart to keep connection alive
  Wire.requestFrom(address, (uint8_t) 1);  // Read one byte from slave register address 
  data = Wire.read();                      // Fill Rx buffer with result
  return data;                             // Return data read from slave register
}
```

This is my whole code.

kriswiner, 2018-01-15:
No idea, do you have pullups on your breakout board?

rachit2jain, 2018-01-16:
https://www.makerlab-electronics.com/product/mpu9250-9-dof-3-axis-accelerometer-gyro-magnetometer/

I have this breakout board. It says the pull ups are on SDA/SCL

rachit2jain, 2018-01-16:
I measured the voltage across SCL and SDA and they are around 2V. Across VCC is around 1.5V.

kriswiner, 2018-01-16:
This is too low. The MPU9250 minimum voltage is 2.4 V. Something is wrong
with your set up. Use 3.30 V for VCC.

xabixab, 2018-04-02:
Hi,

I have the same breakout board and i am having the same issue. I am using Teensy 3.2 and i am connecting the MPU9250 to 3.3v

Any ideas?

Regards,
Xabier

xabixab, 2018-04-02:
Sorry, it was my fault. I hadn't changed the i2c pins to the 18/19 of the teensy.

Working great! Amazing!

VishuMangla, 2019-07-01:
Hi Kris,
I am using GY-87 MPU9250 
https://robu.in/product/mpu9250-9-axis-attitude-gyro-accelerator-magnetometer-sensor-module/?gclid=Cj0KCQjw3uboBRDCARIsAO2XcYBY9YTLmjqps-GraML8phRKSW5r56VKu8F4qGRRa5r1Z4hsQnKMl1caAo5XEALw_wcB
 I tried to connect it to arduino board and after running code .
After reading through a lot of issues already raised, I did an I2CScan() and the output was:
Scanning...
I2C device found at address 0x68!

But it shows
MPU9250 I AM FF I should be 71.

kriswiner, 2019-07-01:
What address are you using to read registers? 0x68? 0x69?

iFrostizz, 2020-04-12:
Hi, same issue... I get 0x68 I2C adress on the scanner, and I have the same error message.
Using an Arduino Uno with a MPU9250 bought here: https://www.ebay.fr/itm/MPU-9250-9-axes-Gyroscope-Accelerateur-magnetometre-MPU9250-1300Z/293106686659?_trkparms=ispr%3D1&hash=item443e851ac3:g:8~UAAOSwkLBc77qH&enc=AQAEAAACQBPxNw%2BVj6nta7CKEs3N0qXGRBquWmwR0g8541HJZFA%2BTerOzTshkq1bInEEMT9xhtbc5AUr%2FhlYg9S7jriVujLpTA1K%2BmVRripX6fNJT%2F45%2Bo9rLK%2By%2BFQ1IuNNwULO29wy0WMBqaKm0OePjqElXfEGQx%2B2vUmYP%2Fk%2BBsYVyq%2F7%2BoWxpnmtN6Ln0xt%2BBiS1CBTDy6eD11xTFOHHVslMR%2BjT3itIqIffvujPZpwBdHnLLxVypB1lj%2F8ijgkDVZFGXn%2BtJIOD2hkV9Aifql7MKks72m48pMBnpGYa7IHPEJG%2F%2FIMNp79JQtovN0govsggNauwYCZZE2eiMCObitW2FHHEhZUeoeZptPy93wP%2BI4eztAfzlGtOF5OyrATjQuovxBcEC4tisQphVcDkZ1LAABlpqNBTMxvaLKZtDVlGYHM8j4%2FIsndvxZ0u0oihcHipChwA7DcvUsa4uri9avIZJS3mgynn8BA5zz1xV4OpR11AnHll60ySOI0sB9mY0zcdXEybxK6YO%2B6ahbMfmUcg5tlR19z6YUWGV7Km8KaJLP8mU4I0bdsFuVJhfP3cJJAQWKSnEaYdEtQ0ewWq27cPRzlaqbm6Wf17IpteOf5hNth0SO9GEGtaZ67FyqWFE5t0uutt7kCQCO661wZaEjVdQG2VzmPXN4wyyoBXeS6QT5NPk1ooZ6ojC23E1cGEzOmn6fuL9n50Bha7hKrau36dLzzvWzyhRDRuPqFLrvz9dLxX113aiQf1FgW5OminbdjjmA%3D%3D&checksum=2931066866594e2c1bac1d834f969ab55527943c2d84

kriswiner, 2020-04-12:
Are you pulling up nCS also?

fabiopdm, 2020-12-24:
Same behavior here...

Using Arduino Uno R3 and this board [GY-91](https://produto.mercadolivre.com.br/MLB-1436250599-acelermetro-e-giroscopio-9-eixos-10-dof-mpu-9250-com-bmp280-_JM). 

GY91-------------------------Arduino Uno
VIn---------------------------5v
3v3---------------------------X
GND------------------------- Ground
SCL---------------------------A5
SDA--------------------------A4
SD0/SA0--------------------- 3.3V
NCS--------------------------X
CSB--------------------------X

Tried also without Ground on NCS, and also without 3.3V on SD0/SA0 (changing the address on the library, like 0x76 for BPM280).

I can reach the BPM280 and obtain the measures. Using a little example from the Adafruit for BPM280. 

Using you MPU9250_BPM280_MS5637_t3 code (changing the comments to the Wire.h for Arduino, I have the follow output from serial monitor:

```
Scanning...
I2C device found at address 0x77  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM FF I should be 71
Could not connect to MPU9250: 0xFF

```

Maybe I have a problem in the MPU9250 (burned or malfunctioning) or maybe this is in the sleeping mode. Can you give me some clue? Maybe something in the code to write in some register, to put online the MPU. I am losing hope of finding a solution.

kriswiner, 2020-12-24:
No idea. Maybe the GY-91 is broken? nCS should be HIGH if using I2C. Try
connecting 3V3 and nCS...

fabiopdm, 2020-12-24:
> No idea. Maybe the GY-91 is broken? nCS should be HIGH if using I2C. Try connecting 3V3 and nCS...
> […](#)

Connecting NCS on 3.3V, still the same results. Not working. 

I found this [site](http://vi.raptor.ebaydesc.com/ws/eBayISAPI.dll?ViewItemDescV4&item=282817706051&category=92078&pm=1&ds=0&t=1572116359000&ver=0&cspheader=1), and looking for the wiring connection, I believe that there is no need for NCS connection. 

I will ask a refund for the vendor here, if its not possible... maybe buy a new one. #448 

Thanks for you time @kriswiner.

fabiopdm, 2020-12-27:
Hi @kriswiner! I found the problem. 

It was on the chip MPU9250, it is a problem of cold soldering, causing poor contact of the chip. 

I bought a new GY-91, waiting to arrive. 

Only to report the solution/problem to our friends here. 

Best regards.

kriswiner, 2020-12-27:
Hope you have better luck with the new board.

The MPU9250 is out of production for almost a year now.

You might think about switching to something like this
<https://www.tindie.com/products/onehorse/all-st-motion-sensor-breakout-board/>

baraiyadk, 2023-03-09:
i am facing same problem
I am using arduino mega2560 with grove IMU 9 DOF

VCC --> 3.3 V
GND --> GND
SCL --> SCL
SDA --> SDA

error below:
Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
I2C device found at address 0x69  !
done
MPU9250 9-axis motion sensor...
MPU9250_1 I AM 11 I should be 71
MPU9250_2 I AM 11 I should be 71
Could not connect to MPU9250 1: 0x11
Could not connect to MPU9250 2: 0x11

kriswiner, 2023-03-09:
Well, you are seeing two MPU9250's, one at 0x68 and one at 0x69 on the I2C
bus as well as one AK8963C. If you are having trouble reading the WO_AM_I
register it is likely a problem with the I2C readByte API and not the
sensors.