kriswiner/MPU9250 #229: Eagle Schematic Question

ElectroStef, 2018-01-18:
Hi Kris,

I'm developing a black box of sorts for a high altitude glider and I'm using the MPU9250. My biggest question is whether or not I have the VDDIO pin connected correctly. I noticed in your design you have a capacitor going to ground, but I'm thinking if I just tie it in with VDD I should be fine. I also have a few other sensor running on my I2C lines so the pull up resistors aren't in this portion of the schematic (my pull ups are 2.7k which is recommended for 3.3V) I am using an ATmega328P with 3.3V. I think this will work, but I would like a second opinion as I am still confused on what the VDDIO pin is used for. I have read the data sheet but I still seem a little confused. If you could help that would be great, thank you!

-stef

[image]

kriswiner, 2018-01-18:
The VDDIO supplies power to the digital sector, mostly I2C, but it is a
different domain and best practice is to bypass the analog (VDD) and
digital (VDDIO) domains separately. Standard paractice it to connect the
caps directly to the MPU9250 pins and use a via on the other side of the
cap to provide the connection to power. I would also tie ADO to GND or VDD,
one or the other depending on which device address you want. If 0x68 is
fine tied ADO to GND. nCS must be pulled HIGH to VDD, no bypass is
required, for I2C. I would recommend you connect the interrupt to the MCU.
The Pro Mini is one of my favorite MCUs but at best you will be able to get
~100 sensor fusion rate out of it, not really adequate if your intention is
absolute orientation estimation. My recommendation is to use 4K7 pullups if
the number of I2C devices on the bus is six or fewer. If you have as many
as 10 devices then 2K2 would be appropriate.

ElectroStef, 2018-01-20:
Thanks for the clarification! 

Ya the Sparkfun design wasn't all that great. 

-Stef