kriswiner/MPU9250 #230: Help with running code WITHOUT LCD display

ACrouse123, 2018-01-22:
I'm having trouble, I want to upload this code to an Arduino Mini 05 but it won't let me because it calls out display codes for a screen. I will not be using a screen, I just want to be able to monitor the code in the serial monitor.

kriswiner, 2018-01-22:
So comment out the lines you don't want...

kriswiner, 2018-01-22:
> Oh okay, sorry. So after I commented all the display commands out, I
> uploaded it to the arduino and I get this error
>
> MPU9250 I AM FF I should be 71
> Could not connect to MPU9250: 0xFF

What do you get when you run an I2C scan? Any devices on the bus?

Do you have pullups on the SDA and SCL?

What is the voltage showing on SDA and SCL?

What is your MCU and what is the voltage input to the MPU board?