kriswiner/MPU9250 #232: MPU9250 generate different values after every RESET, why?

Sterpa, 2018-01-27:
Dear Kris, and All
can anybody explane my, why MPU9250 generate different values after every RESET?
i use a basic [example for M5Stack](https://github.com/m5stack/M5Stack/blob/master/examples/Modules/MPU9250/MPU9250BasicAHRS/MPU9250BasicAHRS.ino) (ESP32) with built-in MPU9250 with no changes, and this example absolutely based on your MPU9250 library and codes.
Every turning RESET button i see a different values, like this:

ax = 0.92 ay = -154.54 az = 1003.05 mg
gx = -0.03 gy = 0.11 gz = -0.01 deg/s
mx = -9445 my = 2018 mz = 538 mG
q0 = 0.03 qx = -0.21 qy = 0.04 qz = 0.98
Yaw, Pitch, Roll: 169.37, 24.01, 4.62
rate = 0.13 Hz

RESET

ax = -1.71 ay = -0.18 az = 844.48 mg
gx = 0.04 gy = 0.08 gz = -0.02 deg/s
mx = -9438 my = 2007 mz = 527 mG
q0 = 0.03 qx = -0.05 qy = 0.05 qz = 1.00
Yaw, Pitch, Roll: 168.68, 6.41, 5.17
rate = 0.13 Hz

RESET

ax = -155.27 ay = -159.12 az = 997.80 mg
gx = -0.11 gy = 0.08 gz = -0.44 deg/s
mx = -9441 my = 2025 mz = 517 mG
q0 = 0.78 qx = -0.11 qy = 0.01 qz = 0.62
Yaw, Pitch, Roll: 67.79, 8.74, -8.84
rate = 900.86 Hz

RESET

ax = -1.77 ay = -157.35 az = 842.96 mg
gx = -0.02 gy = 0.04 gz = -0.01 deg/s
mx = -9425 my = 2013 mz = 491 mG
q0 = 0.77 qx = -0.06 qy = -0.06 qz = 0.64
Yaw, Pitch, Roll: 70.97, -0.74, -10.23
rate = 905.35 Hz

And so on...
The most strange is that Yaw, Pitch, Roll are different every reset, but the device is not moved!
Why accel and magnet are different every reset?
May be i should special calibrate accel and magnet?

P.S. Could you please convert your codes for MPU9250 (MPU9250BasicAHRS_t3.ino and quaternionFilters.ino) to standart library format like *.h and *.cpp, like M5Stack guys done? It's very hard to use *.ino files in other IDE (Visual Studio Code or Eclipce) and hard to verify what you change in your new version - to move changes in to local *.h *.cpp files.
Thx a lot in advance!

kriswiner, 2018-01-27:
Yes, this is uncharacteristically noisy. I was going to suggest calibration
since this would at least give you more or less consistent heading but the
accel values are all over the map and the mag data is just wrong. Looks
like either the MPU9250 is not configured properly or you are using one of
those $3 special boards from aliexpress.

https://github.com/kriswiner/MPU9250/tree/master/MPU9250_BME280_SPIFlash_Ladybug

Sterpa, 2018-02-28:
Dear Kris,
is this example MPU9250_BME280_SPIFlash_Ladybug the most modern and newest for MPU9250?
Why you use only MadgwickQuaternion now? it's better then MahonyQuaternion?

And great thx for mpu9250.h .c style, this is realy great helps!!!

Unfortunatly, i use esp32+mpu9250... and without INP PIN, now with your sketch i have crasy jumping yaw, pitch, rol and i have't any idea what's wrong...
Device stay on a table, but accel (Lin_a) show not zero values, like this:
```
Yaw, Pitch, Roll: 113.52, -0.40, -0.38
Grav_x, Grav_y, Grav_z: 6.61, -7.04, 999.95 mg
Lin_ax, Lin_ay, Lin_az: -8.26, 5.45, -0.44 mg
Gyro temperature is 37.7 degrees C
sumCount = 6050
sum = 0.11
rate = 55238.40 Hz

Yaw, Pitch, Roll: 113.54, -0.38, -0.37
Grav_x, Grav_y, Grav_z: 6.39, -6.69, 999.96 mg
Lin_ax, Lin_ay, Lin_az: -6.02, 6.21, 0.90 mg
Gyro temperature is 37.7 degrees C
sumCount = 6000
sum = 0.11
rate = 54777.34 Hz

Yaw, Pitch, Roll: 113.56, -0.37, -0.35
Grav_x, Grav_y, Grav_z: 6.18, -6.40, 999.96 mg
Lin_ax, Lin_ay, Lin_az: -5.81, 5.91, 0.89 mg
Gyro temperature is 37.7 degrees C
sumCount = 6050
sum = 0.11
rate = 55273.24 Hz
```

kriswiner, 2018-02-28:
This looks pretty good to me, what's the problem?

Sterpa, 2018-02-28:
Well...
I thought, if the sensor is lying motionless on the table, then the acceleration should be 0. And I have them jump. And all the values jump, and a magnetometer and a gyroscope.
Here are more values, more detailed. Are you think that the accelX value can jump from 0.55 to -1.40 for 0.1 second?
```
ax = 0.55ay = 0.24az = 999.08 mg
gx = 0.02gy = 0.13gz = -0.20 deg/s
mx = -5my = 1mz = -12 mG
q0 = 0.96qx = 0.00qy = -0.01qz = -0.27

Yaw, Pitch, Roll: 339.89, -1.18, 0.51
Grav_x, Grav_y, Grav_z: -8.91, -20.55, 999.75 mg
Lin_ax, Lin_ay, Lin_az: 9.46, 20.79, -0.66 mg
Gyro temperature is 40.8 degrees C
sumCount = 1760
sum = 0.12
rate = 14585.02 Hz

ax = -1.40ay = -0.24az = 997.86 mg
gx = -0.05gy = 0.18gz = -0.16 deg/s
mx = -5my = 26mz = -6 mG
q0 = 0.96qx = -0.00qy = -0.01qz = -0.27

Yaw, Pitch, Roll: 340.02, -1.36, 0.33
Grav_x, Grav_y, Grav_z: -5.74, -23.74, 999.70 mg
Lin_ax, Lin_ay, Lin_az: 4.34, 23.49, -1.84 mg
Gyro temperature is 40.8 degrees C
sumCount = 1790
sum = 0.12
rate = 14740.54 Hz

ax = -2.56ay = -1.77az = 1001.89 mg
gx = 0.00gy = 0.14gz = -0.11 deg/s
mx = 3my = -7mz = 1 mG
q0 = 0.96qx = 0.00qy = -0.01qz = -0.27

Yaw, Pitch, Roll: 340.49, -1.39, 0.48
Grav_x, Grav_y, Grav_z: -8.46, -24.20, 999.67 mg
Lin_ax, Lin_ay, Lin_az: 5.89, 22.43, 2.22 mg
Gyro temperature is 40.8 degrees C
sumCount = 1790
sum = 0.12
rate = 14696.36 Hz
```

kriswiner, 2018-02-28:
These are milliGauss/milli g's/ milli dps

These sensors have jitter on the order of 5 - 10 of each of these units.
Nothing can be done about that.

You could try to calibrate the sensors, but no MEMS accelerometer will sho
0 millig when motionless. Unrealistic expectation I am afraid. In fact,
your accel values look very good. Low jitter. Just not zero jitter, never
going to happen.

ifrew, 2018-05-11:
Managed to figure it out. The init routine in the mpu9250.cpp file needs to reset the device to a known set of defaults at startup. I noticed comparing Kris' library to the old one I used to use by Jeff Rowberg that the only difference was in the power management setting. After trying changing it to 0x80 it worked. After reading the MPU9250 spec, it seems 0x80 does a reset of the device to a set of known values.

'void MPU9250::initMPU9250()
{
 // wake up device
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x00); // Clear sleep mode bit (6), enable all sensors 
  delay(100); // Wait for all registers to reset 
' 
Change the 0x00 to 0x80 and you get consistent results after startup, I'm using the M5Stack too and I just verified it.

The sample used by M5stack is an old one from Kris. I checked the latest init routine in his repository but it uses the same 0x00. 

Maybe change that to 0x80 Kris in all your samples for consistency!

Cheers

Iain

ifrew, 2018-05-12:
May have spoke to soon. I see that Kris resets the device in the calibration routine, so the expectation is an Init followed by a Calibration should work. If we change it in reset it would reset to all default register settings. However, I have found that does it does return consistent values each time with the default settings but not after calibration each time...hmmm ..need to look at this a bit more

ifrew, 2018-05-12:
Actually not sure what i was thinking in my last post..fried brain today! My initial assumption is correct. After calibration when you then do an init, it is best to reset the defaults and change the parameters you want to a consistent state. It all is working fine now. So just change the 0x00 to 0x80 and all is well.

kriswiner, 2018-05-12:
These sketches could be done better. Perhaps it would be better to wake up
from sleep, reset, then choose the PLL source in the init function. Then
this wouldn't need to be done in the calibration function. Better still
would be a proper MPU9250 class with wakeup, reset, set PLL functions. I
might augment my MPU9250 class to support these.