kriswiner/MPU9250 #233: 9DoF Razor IMU M0 with MPU9250 lib

ffbboy30, 2018-02-14:
Hi,
I'm trying to use your lib with this board because the code given by the reseller is not working correctly.
I'm just starting arduino for this application and it is not as easy as I thought.
Do you know if someone allready convert your code for this board ? The Chip is a SAMD21.
The current code have problem with the euler angle they are not independant.
Thanks you

kriswiner, 2018-02-14:
Are you calibrating the sensors properly?

Are you feeding the sensor data into the fusion filter properly?

ffbboy30, 2018-02-14:
I've finally make your code working but there is no magnetometer calibration in your files .I think I take the wrong code.
The board don't have an extra magneto.Which code is the best for only use the MPU9250

kriswiner, 2018-02-14:
There is a magnetimeter embedded in the MPU9250.

You can try this one:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

or this one:

https://github.com/kriswiner/MPU9250/tree/master/MPU9250_BME280_SPIFlash_Ladybug

ffbboy30, 2018-02-14:
On both code the magneto is done by a AK8963.
Sorry, I don't find the magneto for the MPU9250

kriswiner, 2018-02-14:
Th AK8963C is the magnetometer embedded in the MPU9250 along with the
MPU6500 accel/gyro.

Look at the data sheet.

ffbboy30, 2018-02-14:
Sorry about the silly question.
Thanks you