kriswiner/MPU9250 #235: Which Biases for the magnetometer, hard coded or from experiment?

yangliu28, 2018-02-17:
Hi Kris, thanks for your contribution, have learnt a lot from it. I've read many issues, and hope this is a new one.

You have tried two ways to set the biases for the magnetometer. One is hard coded values in milliGauss in "MPU9250BasicAHRS_t3.ino".
https://github.com/kriswiner/MPU9250/blob/1594ddd5735e65f35f9d5a5414a067a8d2757a9c/MPU9250BasicAHRS_t3.ino#L414-L422
The other is from experiment that the user needs to rotate the component to all directions to sample the maximum and minimum on all axes, as in all "*_t3.ino" with the following function.
https://github.com/kriswiner/MPU9250/blob/1594ddd5735e65f35f9d5a5414a067a8d2757a9c/MPU9250_MS5637_AHRS_t3.ino#L1064

I am wondering which is better? You said "MPU9250_MS5637_AHRS_t3.ino" is the most modern and correct version, which uses the experiment one. But after a few tried, I found the biases are not consistent, and often varies a lot. I guess it is due to not enough sampling. Personally I found this method kind of tedious, because I need to calibrate it every time, or when move to a new location. And rotating the component definitely needs some skills to cover the extreme values on all axes.

On the other side, the hard coded values are much easier to implement, and the result looks reasonable too! My only concern for this one is, where are these numbers com from? I didn't seem to find them in the datasheet, or are they simply empiric numbers?

Thank you for your time. Your opinion is highly appreciated.

kriswiner, 2018-02-17:
> a8d2757a9c/MPU9250BasicAHRS_t3.ino#L414-L422
> a8d2757a9c/MPU9250_MS5637_AHRS_t3.ino#L1064

The idea is to calibrate the sensor well once then put the offset biases in
the code so you don;t have to calibrate every time.

So the "hard-coded" numbers are for one particular MPU9250 that I
calibrated. These numbers should not be used for any other MPU9250.

Calibrate the magnetometer once empirically and use these numbers in the
code. But remember, the offset biases will depend on location and
temperature, so it is always best to calibrate again before use.

yangliu28, 2018-02-17:
Quick response! Then I should calibrate my sensor at least once. I am actually building a sensor network of 10+ nodes, so got lots of sensors to calibrate, but it's a good chance as well to see how the biases different from one MPU9250 to another. I'll see if I can report this some time later.
Thank you so much for your help!